llvm/circt #888: LowerTypes should lower structures and vectors in memories

darthscsi, 2021-04-05:
firrl.mem allows structs and vectors to escape type lowering and be visible during lower-to-rtl.  These types should be completely eliminated after this pass.

seldridge, 2021-05-07:
This is going to be slightly tricky to do. If memories are "simplified" (what the Scala FIRRTL Compiler calls this) where an aggregate type is combined into a ground type, but the aggregate structure is preserved outside the memory, everything works _except_ for the representation of the write mask.

Consider a memory with a data type of `{a: UInt<3>, b: UInt<5>}`:
```scala
    mem memory:
      data-type => {a: UInt<3>, b: UInt<5>}
      depth => 16
      reader => r
      writer => w
      read-latency => 0
      write-latency => 1
      read-under-write => undefined
```

The write mask here is of type `{a: UInt<1>, b: UInt<1>}` where the leaf `a` is masked independently of the leaf `b`. If we lower this to have a data type of `UInt<8>`, then the mask must become `UInt<1>` and isn't accurately representing how the memory is masked.

We need some way, during type lowering, to capture the fact that the mask has special semantics that the FIRRTL Dialect (and FIRRTL IR) cannot currently express.

Once we get to RTL, then it's the problem of the memory generator to error if it can't handle some weird mask. However, this simplification is way better than the current lowering of splitting memories into one-memory-per-leaf data type.

seldridge, 2023-07-30:
This should be unnecessary now that all memories are correctly simplified. This includes handling the behavior around mask determination.

E.g., in the example given in the comment above, this will lower to a mask size of 1-bit as that is the GCD of the two underlying types:

```verilog
// Generated by CIRCT unknown git version
// VCS coverage exclude_file
module memory_16x8(
  input  [3:0] R0_addr,
  input        R0_en,
               R0_clk,
  input  [3:0] W0_addr,
  input        W0_en,
               W0_clk,
  input  [7:0] W0_data,
               W0_mask,
  output [7:0] R0_data
);

  reg [7:0] Memory[0:15];
  reg       _R0_en_d0;
  reg [3:0] _R0_addr_d0;
  always @(posedge R0_clk) begin
    _R0_en_d0 <= R0_en;
    _R0_addr_d0 <= R0_addr;
  end // always @(posedge)
  always @(posedge W0_clk) begin
    if (W0_en & W0_mask[0])
      Memory[W0_addr][32'h0 +: 1] <= W0_data[0];
    if (W0_en & W0_mask[1])
      Memory[W0_addr][32'h1 +: 1] <= W0_data[1];
    if (W0_en & W0_mask[2])
      Memory[W0_addr][32'h2 +: 1] <= W0_data[2];
    if (W0_en & W0_mask[3])
      Memory[W0_addr][32'h3 +: 1] <= W0_data[3];
    if (W0_en & W0_mask[4])
      Memory[W0_addr][32'h4 +: 1] <= W0_data[4];
    if (W0_en & W0_mask[5])
      Memory[W0_addr][32'h5 +: 1] <= W0_data[5];
    if (W0_en & W0_mask[6])
      Memory[W0_addr][32'h6 +: 1] <= W0_data[6];
    if (W0_en & W0_mask[7])
      Memory[W0_addr][32'h7 +: 1] <= W0_data[7];
  end // always @(posedge)
  assign R0_data = _R0_en_d0 ? Memory[_R0_addr_d0] : 8'bx;
endmodule

module Foo(
  input  [2:0] r_addr,
  input        r_en,
               r_clk,
  input  [2:0] w_addr,
  input        w_en,
               w_clk,
  input  [2:0] w_data_a,
  input  [4:0] w_data_b,
  input        w_mask_a,
               w_mask_b,
  output [2:0] r_data_a,
  output [4:0] r_data_b
);

  wire [7:0] _memory_ext_R0_data;
  memory_16x8 memory_ext (
    .R0_addr ({1'h0, r_addr}),
    .R0_en   (r_en),
    .R0_clk  (r_clk),
    .W0_addr ({1'h0, w_addr}),
    .W0_en   (w_en),
    .W0_clk  (w_clk),
    .W0_data ({w_data_b, w_data_a}),
    .W0_mask ({{5{w_mask_b}}, {3{w_mask_a}}}),
    .R0_data (_memory_ext_R0_data)
  );
  assign r_data_a = _memory_ext_R0_data[2:0];
  assign r_data_b = _memory_ext_R0_data[7:3];
endmodule
```

closed 2023-07-30 by seldridge