llvm/circt #889: [Python] RTL support library broken port names

teqdruid, 2021-04-05:
The recent change to port names https://github.com/llvm/circt/commit/b6b3cb9f9fe7746f5646b66bfa5a2f31812fc6bd broke the RTLModuleOp python wrapper. I've managed to fix it partially, but the autowrapping stuff is still broken.

teqdruid, 2021-04-05:
[rtlmodpy.diff.txt](https://github.com/llvm/circt/files/6260745/rtlmodpy.diff.txt)

Here's my patch. It fixes the building of the port list, if you call RTLModuleOp with all the I/O ports named, it works. The decorator is, however, broken. It seems that python inspect isn't getting the returns.

lattner, 2021-04-05:
I'm sorry for breaking this, thank you for working on a fix!

teqdruid, 2021-04-05:
Seems like it was partially broken before but the port naming enforcement exposed a bug!

teqdruid, 2021-04-06:
@mikeurbach Should I just disable the two decorator tests until we have a chance to drive this to ground? It's led to failed integration tests the last few nights.

mikeurbach, 2021-04-06:
Yep that is fine by me. I can take a look at fixing the decorator.

teqdruid, 2021-04-06:
https://github.com/llvm/circt/commit/3bb6cde8b7d3f3277dddebbb4e90eadfde3c1de3 and re-running the integration tests https://github.com/llvm/circt/actions/runs/724174462

mikeurbach, 2021-04-06:
Thanks. I'm about to add those back fix the issue. I'm also just going to remove _build_ports from there, it is unnecessary now that we have a more intuitive attribute instead of following FunctionLike's (dubious) convention.

The root cause of the issue with the decorator that RTLModuleOp is now verifying that the new argNames/resultNames attributes have as many names in the list as args/resuls (resp.). This isn't an issue with the explicit call to the Python constructor, but in the fancy sugared decorator, it isn't immediately clear how to get the result names. The decorator infers the arg names by introspecting the function's signature, and [using the names of the formal parameters](https://github.com/llvm/circt/blob/ad66d4155b9d65c0303801a581c4aa64d83dc7b3/lib/Bindings/Python/circt/dialects/_rtl_ops_ext.py#L124). I'm not sure if there is something similar for the results though... the quick fix is to just name all the results `""` for now. Patch for that incoming.

teqdruid, 2021-04-06:
I looked into the decorator, but in the 5 mins I spent on it I couldn't even get the # of return values or their types. Or is that what the `i32, i32` decorator is about?

You could always add output port names to the decorator.

mikeurbach, 2021-04-06:
> Or is that what the i32, i32 decorator is about?

That's actually to give hints about the input types. The result types are known after calling the decorated function, here: https://github.com/llvm/circt/blob/ad66d4155b9d65c0303801a581c4aa64d83dc7b3/lib/Bindings/Python/circt/dialects/_rtl_ops_ext.py#L141

> You could always add output port names to the decorator.

I did that in a first draft. In my opinion, naming the results didn't add enough value to justify more typing in what was supposed to be the terse, syntactic sugar approach.

I can take a look and see if there's a neat way to get those names automatically, like with args.

teqdruid, 2021-04-07:
#892 fixed this

closed 2021-04-07 by teqdruid