llvm/circt #899: [exportverilog] Improve location options

lattner, 2021-04-08:
We currently print source location information as a comment with line and column info.  It would be great to expand this (along with a lowering option) to four different options:

1) None
2) Current comment.
3) `line directives (which cover filename + line)
4) `line and comment (which covers filename+line+col)

Something like `veriloglocationstyle={none,comment,line,both}` would be nice.

-Chris

fabianschuiki, 2021-04-12:
See section "22.12 `line" of the [IEEE 1800-2017 Verilog standard](http://ecee.colorado.edu/~mathys/ecen2350/IntelSoftware/pdf/IEEE_Std1800-2017_8299595.pdf) for details on the syntax and semantics of the corresponding Verilog preprocessor directive.

fabianschuiki, 2021-04-12:
One of the challenges will be to translate from the current model where we emit location info in a comment *after* the relevant piece of code to a model for ``` `line``` which should appear *before* it. Consider the following as an example: https://github.com/llvm/circt/blob/ff53056bc574ee297b67c588cf8258bd4e9e1560/lib/Translation/ExportVerilog/ExportVerilog.cpp#L1704-L1715
Here we'd need a hook for location emission both before the `assign` is printed, in addition to the existing `emitLocationInfoAndNewLine`. Other uses are trickier, since they involve us gobbling up emitted operations into a set and then flushing the location info afterwards: https://github.com/llvm/circt/blob/ff53056bc574ee297b67c588cf8258bd4e9e1560/lib/Translation/ExportVerilog/ExportVerilog.cpp#L1698-L1701
This might require us to stash away the current cursor position in the output before we start emitting operations, then going back and inserting the corresponding directive into the output buffer afterwards (similar to spilling into temporaries).

seldridge, 2021-04-12:
Towards simplifying the Verilog Emitter, I'm not against an actual pass/pre-pass that promotes source locators to new `sv.line` operations.

fabianschuiki, 2021-04-12:
Yeah I think that would be an excellent idea!

lattner, 2021-04-13:
I don’t think that will work well, it won’t interact correctly with line breaking, which is inherently syntactic.  It will also bloat out the IR which will cause a signgificant compile time hit