llvm/circt #900: Lack of mux folding causes unreachable code which tools complain about

darthscsi, 2021-04-08:
```
circuit MuxTree :
  module MuxTree :
    input clock : Clock
    input reset : UInt<1>
    input c1 : UInt<1>
    input c2 : UInt<1>
    input d1 : UInt<5>
    input d2 : UInt<5>
    input d3 : UInt<5>
    output foo : UInt<5>
    

    node baz = mux(c1,d2, d3) @[package.scala 32:76]
    node bar = mux(c1, d1, baz) @[Mux.scala 47:69]
    foo <= bar @[IFU.scala 314:11]
```
Gives
```
module MuxTree(
  input        clock, reset, c1, c2,
  input  [4:0] d1, d2, d3,
  output [4:0] foo);

  assign foo = c1 ? d1 : c1 ? d2 : d3;  // Mux.scala:47:69, md_prune2.fir:2:3, package.scala:32:76
endmodule
```
assigning d2 to foo is unreachable and generates warnings.

seldridge, 2021-04-08:
I believe the Scala FIRRTL Compiler solution here is to not inlining things which show up multiple times: https://github.com/chipsalliance/firrtl/pull/1727. Notably, the example you show @darthscsi is almost the exact same in the @jackkoenig comment: https://github.com/chipsalliance/firrtl/pull/1727/files#diff-c6da5ed7e3fd4da61e9f75fd30b40269492a699755bd0752dcd6f264bc541d19R77.

lattner, 2021-04-08:
Seems like we can handle this by canonicalizing `mux(cond1, mux(cond1), x)` and `mux(cond1, x, mux(cond1))`.  I doubt the tools are doing global analysis.

jackkoenig, 2021-04-08:
Thanks for the links @seldridge, I'll just add a little insight from having dealt with this in the past.

One might be tempted (as I was) to solve this via mux folding. In this case you would get: `node bar = mux(c1, d1, d3)`. While I think this is a reasonable `-O2` option, it does actually change the structure of users' code in a way they may find surprising. The feedback I got from users was mixed when I tried folding so I elected to just not do the inlining instead (as in the links).

jackkoenig, 2021-04-08:
All that being said, I think it's fine to target a sensible `-O2` before going back and figuring out how to manage `-O0` since the relationship between generated code and Verilog linting / synthesis is tricky as we all know.

lattner, 2021-04-08:
I don't think there is a reasonable `-O0` model in the Chisel design.

jackkoenig, 2021-04-08:
> I don't think there is a reasonable -O0 model in the Chisel design.

Why not? Even if `-O0` isn't "lint clean" there's still value in it for user understanding.

lattner, 2021-04-08:
-O0 is definable (don't run any elective optimization passes, incl mux folding) but isn't supportable.  As we covered before, a bunch of tools expect certain optimizations otherwise their heuristics break.

jackkoenig, 2021-04-08:
I think it's possible to define a minimal set of optimizations that emits lint-clean and heuristic happy Verilog while only minimally mangling the input. Perhaps this is more of an "`-O1`" than "`-O0`", but this is a perfect example of where one could choose not to do the optimization. You could instead just not inline the mux and you'd get something that more closely matches the input.

closed 2021-04-08 by darthscsi

seldridge, 2021-04-08:
One caveat: there has been push back from users for any non-trivial mux folding, e.g., https://github.com/chipsalliance/firrtl/issues/1049#issuecomment-471234634. If a tool is giving weird QoR on some mux tree, a user may restructure the tree at the Chisel level. Not preserving that intent through to the output Verilog is a change to the current user experience.

_I'm not saying this is the right approach!_ I am saying that there needs to be some fallback to enable front end language design entry with predictable (or exact) output. I think this is what's great about inline assembly for C/C++ and maybe we need a similar abstraction here.

lattner, 2021-04-09:
Yeah, this is a case where the users probably want something that no tool can deliver.

PR #902 (2021-04-08): Canonicalize mux trees
closes #900

Will add other cases