llvm/circt #905: [Python] 'Connect' syntax for Python bindings

teqdruid, 2021-04-08:
I would like to add "late-connect" semantics to the Python bindings. That is, be able to create an arbitrary operation without having `Value`s for each operand. The user would be able to connect them later on using a `connect(a, b)` function. For instance, something like (pseudocode):

```
mod = rtl.RTLModuleOp(
            name='MyWidget',
            input_ports=[('in', i32)],
            output_ports=[('out', i32)],
            ...  )

inst1 = rtl.InstanceOp(mod, "instA")
reg1 = seq.CompRegOp(i32)
inst2 = rtl.InstanceOp(mod, "instB")

connect(inst2.out, inst1.in)
connect(inst1.out, reg1.q)
connect(reg1.d, inst2.in)
```

Would create a feedback loop with a register.

teqdruid, 2021-04-08:
I could obviously do this for specific ops by writing some per-op Python, which is probably what I'll do first. How should we do this in a way that it could be used with any operation?

mikeurbach, 2021-04-09:
At first blush, this `connect` function just looked to me like syntactic sugar around an `sv.connect` operation. Wouldn't it be possible to generically have the proposed `connect` build an `sv.wire` or `sv.reg` and `sv.connect` both of the Values? Maybe I'm missing something, but it seems possible.

However, this gets into `sv.wire` land, and the inout type, which are harder to reason about (not to mention we would be generating superfluous wires). If we wanted to stay away from that, I think it should be possible to thread around the Values and "connect" them appropriately, under certain conditions. I think this would be possible if the Values are coming from rtl.module or rtl.instance.

In the latter case, I'm imagining the rough implementation of connect to:
* look up the OpResult, OpOperand, BlockArgument, etc. for both of the Values
* verify that exactly one Value is an OpOperand
* use the `set` API on that OpOperand to set it to the other value

teqdruid, 2021-04-09:
> At first blush, this `connect` function just looked to me like syntactic sugar around an `sv.connect` operation. Wouldn't it be possible to generically have the proposed `connect` build an `sv.wire` or `sv.reg` and `sv.connect` both of the Values? Maybe I'm missing something, but it seems possible.

It's analogous, but more general since I want it to work for all ops. (My example was probably not demonstrative enough.) Here's another one, though the ops do not yet exist:

```
statusFsm = msft.FsmOp(states=[Init, Running, Stopped], transistion_inputs = [ PipelineDone ])
statusInst = msft.FsmInst(statusFsm, name="status")
...
runDelay1 = seq.CompRegOp(statusInst.Running) # Doesn't use connect syntax
computePipeline = msft.VariableLengthPipelineOp(stages=[...])

connect(runReg.q, actionPipeline.go)
connect(actionPipeline.done, statusFsm.PipelineDone)
```

> In the latter case, I'm imagining the rough implementation of connect to:
> 
> * look up the OpResult, OpOperand, BlockArgument, etc. for both of the Values
> * verify that exactly one Value is an OpOperand
> * use the `set` API on that OpOperand to set it to the other value

Yeah, that's been roughly my thought as well. I'm wondering a few things:

1) Should said API instantiate the ops with known-bad Values (a la `BackedgeBuilder`) when they're created by user code? Or create a "to evaluate later" model wherein the actual ops are created when the full connectivity is known? (I favor the former, though I don't know how feasible it is.)
2) Type inference: I don't want a user to have to specify the types redundantly or when they're obvious. Is it possible to change the type of an operand or result after it is created? Would we need type information from tablegen for better type inference (in the case of 'buildable' types, I think that info would help)?
3) How to enable this on all ops. Is there some pythonic thing we can use? Could we use [this MLIR python binding magic](https://mlir.llvm.org/docs/Bindings/Python/#extending-generated-op-classes) on ops we're interested in? (I don't really understand Python beyond classes and functions or the MLIR Python binding magic.)

mikeurbach, 2021-04-12:
Regarding the last points:

1. I think I prefer the former as well, although that's just my intuition.
2. It should be possible to change the type of both operands and results, but this is generally not recommended. I'm not sure if there is a better way in this case, so if we tread carefully, it should be possible.
3. In general there are two ways I can see to do something for all ops. We could add something to the Python OpView subclass (using the extension mechanism, or some other Python metaprogramming), or we could have a generic function that deals with opaque `Operation *`s and `Value`s. I haven't really thought about this specific use case, but in general you can get pretty far with the latter.

I'm also still thinking about the `sv.wire` approach. Is it not general enough? In your more complex example, wouldn't we still be able to boil this down to generating and connecting wires at the `sv` level? I'm getting hung up on the "want it to work for all ops" part. Don't all ops have `Value`s for their operands and results? And if those `Value`s have a type that can be wrapped in `rtl.inout`, they can be connected to an `sv.wire`.

I think it will be easier to have this discussion after experimenting with a PoC. Is this something you want to prototype @teqdruid? I might be able to experiment later in the week as well.

teqdruid, 2021-04-13:
> I'm also still thinking about the `sv.wire` approach. Is it not general enough? In your more complex example, wouldn't we still be able to boil this down to generating and connecting wires at the [...]

Ah. I think I see the disconnect. While I'm looking for the input to be very structural, I'm looking to end up with an "SSA" or "data flow" model, rather than a FIRRTL-style structural model. And I'd rather not use the `sv.wire` operation since SystemVerilog isn't necessarily the target. Creating `msft.targetValue` and `msft.connectValue` ops (or something like that) to model things structurally, then do a lowering pass to replace the "connect" ops with direct SSA value connections is certainly something to consider.

> I think it will be easier to have this discussion after experimenting with a PoC. Is this something you want to prototype @teqdruid? I might be able to experiment later in the week as well

100% agree. I'm busy the last half of the week ([LATTE workshop](https://capra.cs.cornell.edu/latte21/) + personal time) so I don't know that I'll get to it this week.  Maybe we could continue our newfound trend of staying late at the ODM to discuss/brainstorm specifics?

mikeurbach, 2021-04-13:
> And I'd rather not use the sv.wire operation since SystemVerilog isn't necessarily the target. 

Got it, ok that clears it up for me.

> Maybe we could continue our newfound trend of staying late at the ODM to discuss/brainstorm specifics?

I'm free then this week, happy to discuss then.