llvm/circt #907: Parsing Instance Connects Broken

seldridge, 2021-04-09:
Connections involving instances fail to parse, but work in the Scala FIRRTL Compiler.

The following currently fails:
```scala
circuit Instance:
  module Sub:
    output a: UInt<1>
    a <= UInt<1>(0)

  module Instance:
    output a: {a: UInt<1>}

    inst sub of Sub
    a <= sub
    a <- sub
```

The error is: 

```
# circt-translate --import-firrtl new/Instance.fir
new/Instance.fir:11:5: error: expected '.' in field reference
    a <- sub
    ^
```

Glancing at the parser, it looks like it expects `sub` to be in the value table, but it isn't. There is special logic to handled `sub is invalid`, but that needs to be extended to handle connections to submodules.

seldridge, 2021-04-09:
The approach for invalid is tedious to apply here. For the `sub is invalid` case, we just invalidate all the ports. However, for the `a <= sub` and `a <- sub` case, this is limited application of type lowering to do this correctly. We could instead construct a new aggregate wire that coalesces all the ports, connect the instance to that, and then connect that to `a` (or vice versa).

lattner, 2021-04-09:
I'll take a look

lattner, 2021-04-09:
Ok, well there are two different problems here, one is the parser issue which is straight-forward to fix.  This is a change to FIRParser.cpp:1211 and will geenralize the "bundle is invalid" logic nicely.

The problem is that we have no way to represent this in the IR.  I think we need a `firrtl.bundle (x,y,z)` operation like we discussed earlier today.  Can someone work on this?

youngar, 2021-05-24:
I took a swing at adding a BundleOp here: https://github.com/llvm/circt/pull/1127.  The details of how the new op worked to support both input and output ports and create the expected bundle type seemed overly complex (wire like semantics), and the decision has been made to not merge the PR.