llvm/circt #910: Construct firtool Pass Pipeline from Annotations

seldridge, 2021-04-10:
The Scala FIRRTL Compiler (SFC) uses a special annotation, [`firrtl.stage.RunFirrtlTransformAnnotation`](https://www.chisel-lang.org/api/firrtl/latest/firrtl/stage/RunFirrtlTransformAnnotation.html) to add a custom transform to the SFC compile pipeline.

As things like #909 land, it would be cool if a `firtool` user didn't have to explicitly request `--inline-stuff` and could instead rely on either inline annotations or the input annotation file to control this. This would help provide tighter integration with Chisel.

Concretely, if you see a `firrtl.stage.RunFirrtlTransfomAnnotation("firrtl.passes.InlineInstances")`, the #909 inliner should get scheduled.

(The whole structure of the SFC is "just a bag of transforms". When you want a "Verilog Compiler", this is really a request to "find a linearization of transforms that make up a Verilog Compiler that doesn't violate dependencies, resolving any invalidations along the way." The `RunFirrtlTransformAnnotation` is a way of appending to the list of transforms in this sub-problem.)

lattner, 2021-04-17:
We can support this in some simple cases, but I suspect this is one of the areas we'll intentionally diverge.  We can't support the full generality of what SFC supports (because we have different passes, intentionally)

seldridge, 2023-07-30:
I think we've solidly moved away from this approach. The pass pipeline is now almost entirely fixed and will handle all annotations that should exist. E.g., inlining is always run and is a no-op if no inline annotations exist.

closed 2023-07-30 by seldridge