llvm/circt #912: [FIRRTL] (lt) mismatch

drom, 2021-04-13:
The following FIRRTL program

```python
; seed: 46
circuit top_mod :
  module top_mod :
    output tmp30: UInt<1>
    tmp30 <= lt(SInt<4>(2), SInt<3>(-3))
```

Compiled with `firtool --lower-to-rtl --enable-lower-types` produces this Verilog:

```verilog
// Standard header to adapt well known macros to our needs.

module top_mod(
  output tmp30);

  assign tmp30 = 1'h1;	// a_top_mod.fir:3:3, :5:14
endmodule
```

Compiled with `firrtl-1.5-SNAPSHOT --dont-fold div` produces this Verilog:

```verilog
module top_mod(
  output  tmp30
);
  assign tmp30 = 1'h0;
endmodule
```

`Yosys 0.9+3911 (git sha1 4b31223e, clang 11.0.1 -fPIC -Os)` reports formal mismatch:

```
ERROR: Found 1 unproven $equiv cells in 'equiv_status -assert'.
```

drom, 2021-04-13:
Similar issues:

```python
; seed: 96
circuit top_mod :
  module top_mod :
    output tmp8: UInt<1>
    tmp8 <= neq(SInt<5>("b-11"), SInt<3>("o-3"))
```

firtool:

```verilog
// Standard header to adapt well known macros to our needs.

module top_mod(
  output tmp8);

  assign tmp8 = 1'h1;	// a_top_mod.fir:3:3, :5:13
endmodule
```

firrtl

```verilog
module top_mod(
  output  tmp8
);
  assign tmp8 = 1'h0;
endmodule
```

drom, 2021-04-13:
Similar issue:

```python
; seed: 343
circuit top_mod :
  module top_mod :
    output _tmp37: UInt<1>
    _tmp37 <= gt(SInt<28>("b100110110011101010100110"), SInt<27>("o-153762126"))
```

firtool:

```verilog
// Standard header to adapt well known macros to our needs.

module top_mod(
  output _tmp37);

  assign _tmp37 = 1'h0;	// a_top_mod.fir:3:3, :5:15
endmodule
```

firrtl:

```verilog
module top_mod(
  output  _tmp37
);
  assign _tmp37 = 1'h1;
endmodule
```

drom, 2021-04-13:
Similar issue:

```python
; seed: 437
circuit top_mod :
  module top_mod :
    output _tmp25: UInt<1>
    _tmp25 <= leq(SInt<17>("h-1617"), SInt<18>(3762))
```

firtool:

```verilog
// Standard header to adapt well known macros to our needs.

module top_mod(
  output _tmp25);

  assign _tmp25 = 1'h0;	// a_top_mod.fir:3:3, :5:15
endmodule
```

firrtl
```verilog
module top_mod(
  output  _tmp25
);
  assign _tmp25 = 1'h1;
endmodule
```

drom, 2021-04-13:
geq example

```python
; seed: 1378
circuit top_mod :
  module top_mod :
    output tmp8: UInt<1>
    tmp8 <= geq(SInt<9>("h-42"), SInt<11>("o-30"))
```

fabianschuiki, 2021-04-13:
Oops that seems related to the constant folding code I added a couple of weeks ago. Will take a look.

PR #917 (2021-04-14): [FIRRTL] Use operand type to determine sign for const folding
Use the operand types instead of the operation's result type when deciding whether to perform sign or zero extension of constants for folding. This is important for comparison operators, where the result type is always `uint<1>`, yet signed operands should be sign-extended. Fixes #912.
commit 9832061e (2021-04-14): [FIRRTL] Use operand type to determine sign for const folding (#917)

* Use the operand types instead of the operation's result type when
  deciding whether to perform sign or zero extension of constants for
  folding. This is important for comparison operators, where the result
  type is always `uint<1>`, yet signed operands should be sign-extended.
  Fixes #912.
closed 2021-04-14 by fabianschuiki