llvm/circt #913: [FIRRTL] (dshr, reg, add) mismatch

drom, 2021-04-13:
The following FIRRTL program

```python
; seed: 159
circuit top_mod :
  module top_mod :
    input clock: Clock
    input arst: UInt<1>
    input tmp4: SInt<11>
    output tmp29: SInt<11>
    reg state_11: UInt<1>, clock with: (reset => (arst, UInt<1>(0)))
    wire next_11: UInt<1>
    state_11 <= next_11
    next_11 <= add(state_11, UInt<1>(1))
    tmp29 <= dshr(tmp4, next_11)
```

Compiled with `firtool --lower-to-rtl --enable-lower-types` produces this Verilog:

```verilog
module top_mod(
  input         clock, arst,
  input  [10:0] tmp4,
  output [10:0] tmp29);

  wire _T;	// a_top_mod.fir:10:14
  reg  state_11;	// a_top_mod.fir:8:5
  wire next_11;	// a_top_mod.fir:9:5

  localparam _T_0 = 1'h0;	// a_top_mod.fir:8:57
  localparam _T_1 = 1'h1;	// a_top_mod.fir:8:5
  always @(posedge clock) begin	// a_top_mod.fir:8:5
    if (arst)	// a_top_mod.fir:8:5
      state_11 <= _T_0;	// a_top_mod.fir:8:5
    else	// a_top_mod.fir:8:5
      state_11 <= _T;	// a_top_mod.fir:10:14
  end // always @(posedge)
...
  assign _T = next_11;	// a_top_mod.fir:10:14
  wire [1:0] _T_2 = {_T_0, state_11} + 2'h1;	// a_top_mod.fir:11:16
  assign next_11 = _T_2[0];	// a_top_mod.fir:11:13
  assign tmp29 = $signed(tmp4) >>> $signed({10'h0, _T});	// a_top_mod.fir:3:3, :10:14, :12:14
endmodule
```

Compiled with `firrtl-1.5-SNAPSHOT --dont-fold div` produces this Verilog:

```verilog
module top_mod(
  input         clock,
  input         arst,
  input  [10:0] tmp4,
  output [10:0] tmp29
);
...
  reg  state_11;
  wire [1:0] _GEN_0 = state_11 + 1'h1;
  wire  next_11 = _GEN_0[0];
  assign tmp29 = $signed(tmp4) >>> next_11;
  always @(posedge clock) begin
    if (arst) begin
      state_11 <= 1'h0;
    end else begin
      state_11 <= next_11;
    end
  end
...
endmodule
```

`Yosys 0.9+3911 (git sha1 4b31223e, clang 11.0.1 -fPIC -Os)` reports formal mismatch:

```
ERROR: Found 10 unproven $equiv cells in 'equiv_status -assert'.
```

drom, 2021-04-13:
```verilog
assign tmp29 = $signed(tmp4) >>> $signed({10'h0, _T});
```
why shift value is signed?

drom, 2021-04-13:
maybe related:

```python
; seed: 345
circuit top_mod :
  module top_mod :
    input clock: Clock
    input reset: UInt<1>
    input inp_e: UInt<15>
    output _tmp38: UInt<1>
    reg state_29: UInt<5>, clock with: (reset => (reset, UInt<5>(1)))
    wire next_29: UInt<5>
    state_29 <= next_29
    next_29 <= add(state_29, UInt<1>(1))
    _tmp38 <= lt(inp_e, next_29)
```

drom, 2021-04-13:
that maybe issue with Yosys

fabianschuiki, 2021-04-13:
This could be due to ExportVerilog only checking the *result type* signedness when deciding how to emit the operands for the shift, even though the shift needs special treatment in the shift amount. If I recall correctly, the shift amount is always unsigned in Verilog and implicit reversal of the shift direction through negative values therefore is not allowed.

drom, 2021-04-14:
Not an CIRCT issue

closed 2021-04-14 by drom