llvm/circt #919: [FIRRTL] Type Canonicalization Considered Harmful

seldridge, 2021-04-14:
FIRRTL Dialect type canonicalization is destroying information that is needed to preserve connection semantics.

Consider the following circuit which has three different connections.
```scala
circuit Foo:
  module Foo:

    wire a: {flip a: {flip a: UInt<1>}}
    wire b: {flip a: {flip a: UInt<1>}}
    
    b <- a
    b.a <- a.a
    b.a.a <- a.a.a
```

If you run each of these through the SFC, you will get:

1. `b <- a` => `b.a.a <- a.a.a`
2. `b.a <- a.a` => `a.a <- b.a` (_The connection is reversed._)
3. `b.a.a <- a.a.a` => `b.a.a <- a.a.a`

However, the above circuit will type canonicalize into:
```scala
    wire a: {a: {a: UInt<1>}}
    wire b: {a: {a: UInt<1>}}
```

In this situation, the behavior of connection (2) is changed into `b.a <- a.a`.

I _think_ that wires are the only operation that have this behavior, but I'm not sure. The problem may show up anywhere that the SFC is relying on flip information that canonicalization is removing.

I think we can get around this by adding information to wires (and anything else that needs it) that stores the original, un-canonicalized type, and uses this information as an aid to resolve connection semantics.

seldridge, 2021-04-14:
I'm going to look at removing type canonicalization and see where I get. This example can be grabbed from https://github.com/llvm/circt/pull/897/commits/2d2e50d633f212659be2f99f56854b6632679da0.

seldridge, 2021-04-15:
I extended the test in the previous commit and landed that with `XFAIL: *` to disable them.

After thinking about this a lot, this issue is really the same thing as https://github.com/llvm/circt/issues/380, but it manifests differently for different flows:

1. Source/Sink flow cannot be verified unless you also flip connection order when canonicalizing (#380).
2. Duplex flow results in incorrect code because flip information determines who drives what.

We'd decided to accept (1) and relax what is an allowable connect. This is acceptable, but not great... If the MFC gives weaker checking than the SFC, this makes it less safe to use as a Chisel backend. However, (2) is critical.

lattner, 2021-04-17:
Ok, I think that it is worth just removing flip cannonicalization at this point.  It sounds like an experiment that didn't work out.

seldridge, 2021-04-18:
> Ok, I think that it is worth just removing flip cannonicalization at this point.

:+1: Working towards this on the https://github.com/llvm/circt/tree/dev/seldridge%2Fissue-919 branch. It touches everything, so I'll need some time to get it done.

> It sounds like an experiment that didn't work out.

Yeah, ugh... I hate this as the canonicalized types are super clean. 🤷‍♀️ That said, moving closer to the actual FIRRTL IR representation and leaving the clean abstractions for blue sky dialects should make everything easier.

seldridge, 2021-05-02:
Fixed via #944. The cases shown in the OP now work for connects. Partial connect support will be added in a reworked #897.

closed 2021-05-02 by seldridge