llvm/circt #921: [FIRRTL] (xor) mismatch

drom, 2021-04-15:
The following FIRRTL program

```python
; seed: 2216
circuit top_mod :
  module top_mod :
    input inp_j: UInt<3>
    output _tmp10: UInt<4>
    _tmp10 <= xor(asSInt(inp_j), SInt<4>("h0"))
```

Compiled with `firtool --lower-to-rtl --lower-types` produces this Verilog:

```verilog
// Standard header to adapt well known macros to our needs.

module top_mod(
  input  [2:0] inp_j,
  output [3:0] _tmp10);

  assign _tmp10 = {inp_j[2], inp_j};	// a_top_mod.fir:3:3, :6:15
endmodule
```

Compiled with `firrtl-1.5-SNAPSHOT --dont-fold div` produces this Verilog:

```verilog
module top_mod(
  input  [2:0] inp_j,
  output [3:0] _tmp10
);
  assign _tmp10 = inp_j;
endmodule
```

`Yosys 0.9+3911 (git sha1 4b31223e, clang 11.0.1 -fPIC -Os)` reports formal mismatch:

```
ERROR: Found 1 unproven $equiv cells in 'equiv_status -assert'.
```

drom, 2021-04-15:
I looking into fixing this issue.

drom, 2021-04-15:
FIRRTL sign extend for the `or` operation, and 'not-extend' (that is equivalent to zero-extend) in case of `xor` operation.
```python
circuit top_mod :
  module top_mod :
    input inp_j: UInt<3>
    output _tmp10: UInt<4>
    _tmp10 <= or(asSInt(inp_j), SInt<4>(0))
```

FIRRTL:

```verilog
module top_mod(
  input  [2:0] inp_j,
  output [3:0] _tmp10
);
  wire [2:0] _GEN_0 = inp_j;
  assign _tmp10 = {{1{_GEN_0[2]}},_GEN_0};
endmodule
```
Why it is different for `or` / `xor` ?
That maybe a FIRRTL bug?

fabianschuiki, 2021-04-15:
I think this is a FIRRTL bug. `_tmp10 = inp_j` does definitely do zero-extension instead of sign-extension, so that must have been accidentally folded away somewhere in the FIRRTL compiler. I could totally see the FIRRTL compiler look at `y = x | 0` and say "hey that's a nop, let me just replace that with `x`, but then in `y = x` the appropriate sign-extension might be lost because there's no operation taking place anymore.

fabianschuiki, 2021-04-15:
firtool seems to do the right thing

fabianschuiki, 2021-04-15:
For your initial example @drom I get different results if I adjust the bit width of the constant:
- `b <= xor(asSInt(a), SInt<4>("h0"))` yields `assign b = a;`
- `b <= xor(asSInt(a), SInt<5>("h0"))` yields `wire [2:0] _GEN_1 = a; assign b = _GEN_1[3:0];`
- `b <= xor(asSInt(a), SInt<3>("h0"))` yields `wire [2:0] _GEN_1 = a; assign b = {{1'd0}, _GEN_1};`

fabianschuiki, 2021-04-15:
The difference between `xor` and `or` is the result of constant propagation:
- `xor` replaced by `asUInt(asSInt(a))`
- `or` replaced by `pad(asSInt(a), 4)`

Seems to be pissing a `pad` in the `xor` case. Compare the [OR simplification](https://github.com/chipsalliance/firrtl/blob/20890bbd6bef2f33adace433a2d01f89458f0466/src/main/scala/firrtl/transforms/ConstantPropagation.scala#L241
) code to the [XOR simplification](https://github.com/chipsalliance/firrtl/blob/20890bbd6bef2f33adace433a2d01f89458f0466/src/main/scala/firrtl/transforms/ConstantPropagation.scala#L255) code. The latter is missing the `pad(..., e.tpe)`. I'll post a PR.

darthscsi, 2021-04-15:
SFC bug.

closed 2021-04-15 by darthscsi