llvm/circt #922: [CI] Update github workflow caching to be based on the actual workflow content

mikeurbach, 2021-04-15:
The npcomp project has a really nice feature where they get a hash of the actual github workflow content, and use that in the cache key for the LLVM/MLIR build. This lets you tweak build settings in that file without munging up the cache key every time, so we should probably do something similar.

https://github.com/llvm/mlir-npcomp/blob/28a0f02746f4a1315d8e17d9560f7ae1c255236c/.github/workflows/buildAndTest.yml#L67-L70

Ayushsunny, 2021-04-18:
Hello @mikeurbach I would like to fix this

mikeurbach, 2021-04-19:
Great! If you take a look at the above link, you can see a github workflow step that returns a hash of the current file's contents. I'm suggesting that we add a step to circt to do the same, and use that where we currently have `nocode-sharedlibs`: https://github.com/llvm/circt/blob/b9bf17c1c8cee2edd733299c204dd16674f460b8/.github/workflows/buildAndTest.yml#L40 There would be a couple places to update the key there. It also might make sense to use the same pattern on the other worfkflows. 

If you want to test your change, you should be able to try it against your branch in the github actions page.

Ayushsunny, 2021-04-21:
> Great! If you take a look at the above link, you can see a github workflow step that returns a hash of the current file's contents. I'm suggesting that we add a step to circt to do the same, and use [...]

Sir, I have made the changes to both the places, can you please check it now and review it.

Ayushsunny, 2021-04-21:
Hello @mikeurbach Sir, Sorry for the last PR.
I have made the new one without conflicts can you please review https://github.com/llvm/circt/pull/943

mikeurbach, 2021-04-23:
Thanks @Ayushsunny!

closed 2021-04-23 by mikeurbach