llvm/circt #929: [FIRRTL] Verification for MuxPrimOp fails after canonicalization

youngar, 2021-04-19:
Input:
```mlir
// ./bin/firtool -canonicalize ./can.mlir
firrtl.circuit "MuxInvalidOpt" {
firrtl.module @MuxInvalidOpt(%in : !firrtl.uint<1>, %out : !firrtl.flip<uint<4>>) {
  %c7_ui4 = firrtl.constant(7 : ui4) : !firrtl.uint<4>
  %c1_ui2 = firrtl.constant(1 : ui2) : !firrtl.uint<2>
  %c0_ui2 = firrtl.constant(0 : ui2) : !firrtl.uint<2>
  %0 = firrtl.mux (%in, %c7_ui4, %c0_ui2) : (!firrtl.uint<1>, !firrtl.uint<4>, !firrtl.uint<2>) -> !firrtl.uint<4>
  %1 = firrtl.mux (%in, %c1_ui2, %0) : (!firrtl.uint<1>, !firrtl.uint<2>, !firrtl.uint<4>) -> !firrtl.uint<4>
  firrtl.connect %out, %1 : !firrtl.flip<uint<4>>, !firrtl.uint<4>
}
}

```

Error:
```
./can.mlir:10:8: error: 'firrtl.mux' op result type should be '!firrtl.uint<2>'
  %1 = firrtl.mux (%in, %c1_ui2, %0) : (!firrtl.uint<1>, !firrtl.uint<2>, !firrtl.uint<4>) -> !firrtl.uint<4>
       ^
./can.mlir:10:8: note: see current operation: %0 = "firrtl.mux"(%in, %c1_ui2, %c0_ui2) : (!firrtl.uint<1>, !firrtl.uint<2>, !firrtl.uint<2>) -> !firrtl.uint<4>
```

IR:
```mlir
firrtl.circuit "MuxInvalidOpt"  {
  firrtl.module @MuxInvalidOpt(%in: !firrtl.uint<1>, %out: !firrtl.flip<uint<4>>) {
    %c1_ui2 = firrtl.constant(1 : ui2) : !firrtl.uint<2>
    %c0_ui2 = firrtl.constant(0 : ui2) : !firrtl.uint<2>
    %0 = firrtl.mux(%in, %c1_ui2, %c0_ui2) : (!firrtl.uint<1>, !firrtl.uint<2>, !firrtl.uint<2>) -> !firrtl.uint<4>
    firrtl.connect %out, %0 : !firrtl.flip<uint<4>>, !firrtl.uint<4>
  }
}
```

fabianschuiki, 2021-04-19:
Looks like the padding to the larger of both operand widths that is implicit in the mux op needs to be turned into an explicit `firrtl.pad` operation when we folding statically picks one of the operands.

seldridge, 2021-04-19:
The SFC will accept this (even though the specification says it is illegal). 

I propose we add a mux canonicalization to FIRRTL Dialect that will do the padding automatically.

darthscsi, 2021-04-19:
Fixed here: https://github.com/llvm/circt/commit/da83debdbfb114b912cf0528610615f393034753

closed 2021-04-19 by darthscsi