llvm/circt #933: Cross Module Reference (XMR) Primitive

seldridge, 2021-04-20:
Add support for a cross module reference (XMR) operation to a core dialect and to SystemVerilog. The lowering of the core dialect XMR to SystemVerilog is either resolved through:
1. Lowering to SystemVerilog XMRs
2. Boring ports through the hierarchy

This should support the ability to differentiate specific instances in the design and not require an unfolded design. The following example shows what this could look like. Here, there is only one module `@Bar`, but there are different XMRs into different instances that follow different paths:

```mlir
rtl.module @Bar(%a: i1) -> (%b: i1) {
  %x = sv.wire  : !rtl.inout<i1>
  rtl.output %a : i1
}
rtl.module @Foo() -> () {
  %bar1.b = rtl.instance "bar1" @Bar(%0) : (i1) -> i1
  %bar2.b = rtl.instance "bar2" @Bar(%0) : (i1) -> i1

  // Example XMR into bar1
  %baz = rtl.xmr @Bar("bar1"), %x : !rtl.inout<i1>
  // Example XMR into bar2
  %qux = rtl.xmr @Bar("bar2"), %x : !rtl.inout<i1>
}
```

The above XMR syntax is totally made up. This only exists to get a discussion going.

### Background

This notion of resolving XMRs is very similar to prior work on "soft connects". [[1]] Here, a string key is used to establish "sources" and "sinks" in the design. A Bluespec pass then resolves these handling one-to-many and many-to-one cases by inserting arbiters/queues in the design.

[1]: http://people.csail.mit.edu/emer/papers/2009.07.dac.leap_softconnect.pdf

Chisel and the Scala FIRRTL Compiler (SFC) have a "boring utils" API. [[2]] This, like soft connects, relies on sources and sinks (implemented as FIRRTL annotations) that are keyed by a string. The SFC then resolves these using its `WiringTransform` that finds the least-common-ancestor (LCA) between the source and sink and wires them up. The Chisel API allows either downwards XMRs (by choosing the string key for the user and selectively disabling deduplication) or the less explicit string key approach.

[2]: https://www.chisel-lang.org/api/3.4.2/chisel3/util/experimental/BoringUtils$.html

Verilog has support for doing hierarchical references in downwards, upwards, and spooky-action-at-a-distance (see code snippet below). This issue is _not_ concerned with capturing the full power of hierarchical references other than downwards references in emitted Verilog.

```verilog
module Top();
  Foo foo();
  Bar bar();

  logic x;

  /* Downwards reference */
  foo.x

endmodule();

module Foo();
  logic x;

  /* Upwards reference */
  Top.x;

endmodule

module Bar();

  /* Spooky-action-at-a-distance reference */
  Top.foo.x;

endmodule
```

mikeurbach, 2021-04-20:
This is an interesting feature, so let me try to understand the use case. When Juan was asking about how to expose memories from a design to a top-level module, would that be a good place to use cross module references? In this case, a downward reference from the top-level module to the memory?

When I read this, I'm wondering how these will be used, and if there are any guardrails to prevent abuse. In other words, it seems like you could build a circuit out of cross module references, rather than using the Values that are inputs/outputs to modules. And that would lead to a circuit that is much harder to analyze/transform. So would cross module references only show up when a user requests them in their design input? I guess I'm wondering what will lower into the proposed operation.

teqdruid, 2021-04-20:
I agree that it is an interesting and useful feature. One use case are ESI services: since they're theoretically global (ie available everywhere), XMR would be a way to hook the clients up to the server.

One note: SystemVerilog cross references are not considered synthesizable and I don't think many synthesis tools support them. Fortunately, changing ports and connections is viable for us.

seldridge, 2021-04-20:
@mikeurbach wrote:
> When Juan was asking about how to expose memories from a design to a top-level module, would that be a good place to use cross module references? In this case, a downward reference from the top-level module to the memory?

Great observation. This is an implementation of what Juan is asking about. Using XMRs is one way of doing this. Other approaches are:
1. Bore ports through the hierarchy. ("Lower" XMRs to ports/wires.)
2. Move the memories to the test bench.

@mikeurbach wrote:
> When I read this, I'm wondering how these will be used, and if there are any guardrails to prevent abuse. In other words, it seems like you could build a circuit out of cross module references, [...]

I did not envision guardrails. If a user builds a circuit consisting of port-less modules and only XMRs, that's a valid choice. 

@mikeurbach wrote:
> So would cross module references only show up when a user requests them in their design input? I guess I'm wondering what will lower into the proposed operation.

Yes, these would only show up if a user explicitly requests them. These should not be something that we actively try to lower to.

The [Chisel BoringUtils API is a pretty good example](https://www.chisel-lang.org/api/3.4.2/chisel3/util/experimental/BoringUtils$.html). The top module from that is below. This does an XMR that connects `constant.x` to `expect.y`.

```scala
class Top extends Module {
  val io = IO(new Bundle{})
  val constant = Module(new Constant)
  val expect = Module(new Expect)
  BoringUtils.bore(constant.x, Seq(expect.y))
}
```

This connection is not resolved by Chisel, but results in an annotation with a string key `"x_4"`. The Scala FIRRTL Compiler will then recompute the lowest-common-ancestor (the top module) and connect these together, adding ports as needed.

```json
[
  {
    "class":"firrtl.passes.wiring.SourceAnnotation",
    "target":"Top.Constant.x",
    "pin":"x_4"
  },
  {
    "class":"firrtl.passes.wiring.SinkAnnotation",
    "target":"Top.Expect.y",
    "pin":"x_4"
  }
]
```

From the FIRRTL perspective, it is natural to lower these annotation to XMR primitives in the least-common-ancestor and then lower those to actual XMRs or ports.

@teqdruid wrote:
> One use case are ESI services: since they're theoretically global (ie available everywhere), XMR would be a way to hook the clients up to the server.

I used to use these for hooking up a scan chain of fault injectors where the Scala FIRRTL Compiler would generate the fault injectors during compilation. You don't care about how they're hooked up, just that they are hooked up and that the compiler reports the resulting topology.

The soft connects paper was presented as "improving designer productivity" where you want to optionally hook up debug connections to a debug controller. If you have to manually add ports there's tons of different possible combinations of different debug connections a user may enable (and tons of different ports that you may have to add). Going the soft connect route avoids that headache.

@teqdruid wrote:
> One note: SystemVerilog cross references are not considered synthesizable and I don't think many synthesis tools support them. Fortunately, changing ports and connections is viable for us.

I think that's fine. Sometimes you want actual XMRs for some simulation thing on the side. Sometimes you want ports. Supporting a lowering to both (or one to the other) should solve handle the use cases here.

mikeurbach, 2021-04-20:
> Yes, these would only show up if a user explicitly requests them. These should not be something that we actively try to lower to.

Makes sense, thanks for the explanation. CC @JuanEsco063 since you may be interested in this feature.

lattner, 2021-04-25:
Are XMR's widely supported in verilog-consuming tools (incl OSS ones like verilator and yosis)?

If Chisel doesn't already produce these, then we should be very careful about introducing them.  This will have to be a new LowerOptions flag, which means we'll have to support traditional "boring" anyway.  I recommend that we start there before getting fancy.  Just generating structs will be a huge step forward on its own :-)

rkallal, 2021-05-04:
Yes on simulation consuming tools but no on tools used in the synthesis/place and route. XMRs also break encapsulation practices that you find in software and the same reasons for software encapsulation applies to hardware RTL code (private Flip Flops are just like private variables).

seldridge, 2021-09-23:
## Responses

@lattner wrote:
> Are XMR's widely supported in verilog-consuming tools (incl OSS ones like verilator and yosis)?

Verilator supports them. I've been using this `verilator --lint-only` to check the XMRs generated by Grand Central. 😉 

XMRs are not synthesizable (so any tool that is doing synthesis should reject them).

## New Motivating Example

Reviving this as I'm starting to run into a concrete XMR need and am struggling to do it with existing infrastructure.

I've ported a pass to the FIRRTL dialect called Grand Central that is doing three main things:

1) Creation of SV interfaces based on FIRRTL annotations describing interface structure
2) Binding of SV interfaces into specific modules
3) Generation of XMRs to drive the interfaces

The code that (3) eventually generates looks like [this in-tree integration test](https://github.com/llvm/circt/blob/main/test/Dialect/FIRRTL/SFCTests/GrandCentralInterfaces/Wire.fir#L129).

I am currently [relying on `sv::VerbatimOp` to emit these "XMRs" as strings determined at pass execution time](https://github.com/llvm/circt/blob/762f5a07162ea97198d27b31295e004b99164b8c/lib/Dialect/FIRRTL/Transforms/GrandCentral.cpp#L360). _This is criminally brittle as any downstream change to any part of the hierarchical reference will break the XMR_. [`HWLegalizeNames` currently can break Grand Central](https://github.com/llvm/circt/issues/1799) as [this expectedly failing test case shows due to use of the reserved Verilog keyword `signed`](https://github.com/llvm/circt/blob/main/test/Dialect/FIRRTL/SFCTests/GrandCentralInterfaces/HWRename.fir).

There are a lot of ways to fix this.  However, they all involve associating symbols with things which currently have none.  This is independent of the implementation of XMRs.  Namely if you have an XMR op or if you're using `sv::VerbatimOp` macros, you need symbols in either case.

What to do about symbols?  There are a lot of features that should go into a solution here:

1. Add optional symbols to declarations. (This is what `hw.instance` is doing to enable `sv.bind`. However, `sv.bind` is not specific enough to attach to just specific instances.)
2. Define a `hw.public @__a__ %a : <type>` op that can attach a symbol to values whose defining op doesn't or shouldn't have an optional symbol. (There is no use of this that I am aware of.)
3. Add a top-level symbol that encodes a path to something. XMRs can then use this path. (FIRRTL non-local annotations use this.)
4. Add the ability to indicate that components are on a top-level path. (FIRRTL non-local annotation use this.)

FIRRTL's non-local annotations (per-instance metadata) support is currently using (3) and (4).  The following example from the tests show such a use.  This doesn't incorporate either of (1) or (2) and instead relies on strings on the top-level symbol and (4) to attach information to instances which don't have symbols:

```mlir
firrtl.circuit "Foo"   {
  firrtl.nla @nla_2 [@Foo, @Bar, @Baz] ["bar", "baz", "Baz"]
  firrtl.nla @nla_1 [@Foo, @Bar, @Baz] ["bar", "baz", "Baz"]
  firrtl.module @Baz() attributes {annotations = [{a = "a", circt.nonlocal = @nla_1}, {b = "b", circt.nonlocal = @nla_2}]} {}\
  firrtl.module @Bar() {
    firrtl.instance @Baz  {annotations = [{circt.nonlocal = @nla_1, class = "circt.nonlocal"}, {circt.nonlocal = @nla_2, class = "circt.nonlocal"}], name = "baz"}
  }
  firrtl.module @Foo() {
    firrtl.instance @Bar  {annotations = [{circt.nonlocal = @nla_1, class = "circt.nonlocal"}, {circt.nonlocal = @nla_2, class = "circt.nonlocal"}], name = "bar"}
  }
}
```

I still don't have a concrete proposal here for how to handle this. What I think I have are the four pieces that could go into a solution.  This discussion ignores uses of nested symbol tables.  These are still on the table, but would require upstream MLIR work to get them to do what we want.

seldridge, 2023-10-19:
This has been handled primarily with `XMRRefOp` with support in:

- https://github.com/llvm/circt/pull/4422
- https://github.com/llvm/circt/pull/4421
- https://github.com/llvm/circt/pull/4349

closed 2023-10-19 by seldridge