llvm/circt #955: 65 bit constant gets changed

darthscsi, 2021-04-24:
LowerToRTL negates this unsigned constant (ui65 : 0x10000000000000000) leading to canonicalize thinking it is zero.

```module  {
  firrtl.circuit "foo"  {
    firrtl.module @foo(%io_in_valid: !firrtl.uint<1>, %foo: !firrtl.flip<uint<65>>) {
      %c18446744073709551616_ui65 = firrtl.constant(18446744073709551616 : ui65) : !firrtl.uint<65>
      %c0_ui65 = firrtl.constant(0 : ui65) : !firrtl.uint<65>
      %0 = firrtl.mux(%io_in_valid, %c18446744073709551616_ui65, %c0_ui65) : (!firrtl.uint<1>, !firrtl.uint<65>, !firrtl.uint<65>) -> !firrtl.uint<65>
      firrtl.connect %foo, %0 : !firrtl.flip<uint<65>>, !firrtl.uint<65>
    }
  }
}
```

darthscsi, 2021-04-24:
Example that just required canonicalize.  Start with 0x0FFFFFFFFFFFFFFFF, add one to get 0x10000000000000000.  The 'and' should not be eliminated as it result should have a top bit of (1 & io_in_valid).
 
```rtl.module @foo(%io_in_valid: i1) -> (%foo: i65) {
  %foo = rtl.constant 18446744073709551615 : i65
  %bar = rtl.constant 1 : i65
  %0 = comb.sext %io_in_valid : (i1) -> i65
  %baz = comb.add %foo, %bar : i65
  %1 = comb.and %0, %baz : i65
  rtl.output %1 : i65
}
```

prithayan, 2021-04-24:
The issue is in ``LowerToRTL.cpp``, the sign is not handled properly, and causes an unsigned number to be signed when its >64bits and the top bit is ``1``. I think the issue is in the folowing two functions,

```
LogicalResult FIRRTLLowering::visitExpr(ConstantOp op) {
  return setLowering(op, getOrCreateIntConstant(op.value()));
}
```

```
Value FIRRTLLowering::getOrCreateIntConstant(const APInt &value) {
  auto attr = builder.getIntegerAttr(
      builder.getIntegerType(value.getBitWidth()), value);
```

Can we update the above implementation to take into account the ``op.getType().isSigned()`` ? 
Tried to create an unsigned RTL constant, but getting the following error ``RTLIntegerType`` cannot be signed ? So, what's the correct construct here ?
```
1.mlir:4:18: error: 'rtl.constant' op result #0 must be an integer bitvector of one or more bits, but got 'ui65'
      %c0_ui65 = firrtl.constant(0 : ui65) : !firrtl.uint<65>
```

youngar, 2021-04-24:
I think the issue is in the [RTL AndOp fold](https://github.com/llvm/circt/blob/main/lib/Dialect/Comb/CombFolds.cpp#L175); The following line is supposed to construct a 65-bit allOnesValue:
`APInt value(/*numBits=*/width, -1, /*isSigned=*/false);`
Since `isSigned` is false, and the initial value parameter is 64 bits, the MS bit is 0.  We should pass in`/*isSigned=*/true`, or switch to using `auto value = getAllOnesValue(width)`.

darthscsi, 2021-04-24:
The problem is with attribute creation and can be triggered several
places.  It's possible to trigger it just with parsing and reprinting
without running any pass.

prithayan, 2021-04-24:
Maybe, ``APInt``, should really have an extra sign bit, in addition to the ``width``.

lattner, 2021-04-24:
@youngar is right, we just need to change the AndOp fold.  APInt is internally correct, this one is just being constructed wrong.

lattner, 2021-04-24:
I'd recommend `auto value = APInt::getAllOnesValue(width)` of the two since that is more explicit.

lattner, 2021-04-25:
I dropped in the obvious patch here: 10a145c3.  I don't think there is a pervasive problem with the design of attributes.

lattner, 2021-04-25:
Actually, I didn't see @darthscsi's comment here:

>The problem is with attribute creation and can be triggered several
> places.  It's possible to trigger it just with parsing and reprinting
> without running any pass.

I'll poke around, it is possible there is a bozo bug somewhere.  If so, it isn't a fundamental problem with attribute representation.

closed 2021-04-25 by lattner

lattner, 2021-04-25:
K, I don't see anything obvious wrong here, but you're right that the APInt API has several places that make it too easy to do implicit truncation of >64 bit values.  Lemme know if you see another testcase, but the original instigator of this seems like it is lowered correctly now:

```
  rtl.module @foo(%io_in_valid: i1) -> (%foo: i65) {
    %c-18446744073709551616_i65 = rtl.constant -18446744073709551616 : i65
    %0 = comb.sext %io_in_valid : (i1) -> i65
    %1 = comb.and %0, %c-18446744073709551616_i65 : i65
    rtl.output %1 : i65
  }
```

Which turns into this amusing but correct verilog:

```
  assign foo = {65{io_in_valid}} & 65'h10000000000000000;
```

lattner, 2021-04-25:
As of d25f82fa, we now produce the nicer:

`assign foo = {io_in_valid, 64'h0};`

prithayan, 2021-04-26:
I still see the issue in lower to RTL, the 65 bit constant is getting negated.

```
firrtl.circuit "foo"  {
    firrtl.module @foo(%io_in_valid: !firrtl.uint<1>, %foo: !firrtl.flip<uint<65>>) {
      %c18446744073709551616_ui65 = firrtl.constant(18446744073709551616 : ui65) : !firrtl.uint<65>                                                                                                                    %c0_ui65 = firrtl.constant(0 : ui65) : !firrtl.uint<65>
      %0 = firrtl.mux(%io_in_valid, %c18446744073709551616_ui65, %c0_ui65) : (!firrtl.uint<1>, !firrtl.uint<65>, !firrtl.uint<65>) -> !firrtl.uint<65>
      firrtl.connect %foo, %0 : !firrtl.flip<uint<65>>, !firrtl.uint<65>
    }
  }
```
``circt-opt -lower-firrtl-to-rtl --> ``  
```
module attributes {firrtl.mainModule = "foo"}  {
  sv.verbatim "// Standard header to adapt well known macros to our needs."
  sv.verbatim ""
  rtl.module @foo(%io_in_valid: i1) -> (%foo: i65) {
    %c0_i65 = rtl.constant 0 : i65
    %c-18446744073709551616_i65 = rtl.constant -18446744073709551616 : i65
    %0 = comb.mux %io_in_valid, %c-18446744073709551616_i65, %c0_i65 : i65
    rtl.output %0 : i65
  }
}
```

darthscsi, 2021-04-26:
@prithayan moved to #961

lattner, 2021-04-26:
The 65 bit constant isn't "negated".  It is a 2's complement bit pattern that the asmprinter happens to print with a negative value.  It is the correct value.