llvm/circt #958: Pull Handshake Affine lowering out into a separate pass.

mikeurbach, 2021-04-25:
The current StandardToHandshake pass contains a `rewriteAffineFor` method: https://github.com/llvm/circt/blob/5b923be8f516cba20d1bf6aa38cf45314c97793e/lib/Conversion/StandardToHandshake/StandardToHandshake.cpp#L1465, which implements some lowerings for Affine dialect constructs: https://github.com/llvm/circt/tree/main/test/Conversion/StandardToHandshake/Affine.

However, there are some noted TODOs in the current logic, like supporting nested loops, etc. These could all be resolved by using the upstream affine tools in MLIR to lower the affine dialect to standard. In essence, we want to do something like a combination of https://mlir.llvm.org/docs/Passes/#-lower-affine-lower-affine-operations-to-a-combination-of-standard-and-scf-operations and https://mlir.llvm.org/docs/Passes/#-convert-scf-to-std-convert-scf-dialect-to-standard-dialect-replacing-structured-control-flow-with-a-cfg.

As a first step, I propose to remove the `rewriteAffineFor` logic, and place it in a separate pass. The existing tests should pass, given the new pass is used. For example, we would update https://github.com/llvm/circt/blob/5b923be8f516cba20d1bf6aa38cf45314c97793e/test/Conversion/StandardToHandshake/Affine/test-affine-for.mlir#L1

To say `circt-opt -convert-affine-to-std -create-dataflow`. Longer term, the new `-convert-affine-to-std` pass could evolve to use upstream tools in order to resolve the TODOs, or we could potentially use upstream passes entirely to achieve the lowering.

This issue is about the first step, to separate out the Affine handling to a new pass.

Patosga, 2021-05-04:
I'm currently working on this issue.

mortbopet, 2021-12-09:
Closed by  #2321

closed 2021-12-09 by mortbopet