llvm/circt #961: LowerToRTL negates 65bit constants

darthscsi, 2021-04-26:
```module  {
  firrtl.circuit "bar"  {
    firrtl.module @bar(%foo: !firrtl.flip<uint<65>>) {
      %c18446744073709551616_ui65 = firrtl.constant(18446744073709551616 : ui65) : !firrtl.uint<65>
      firrtl.connect %foo, %c18446744073709551616_ui65 : !firrtl.flip<uint<65>>, !firrtl.uint<65>
    }
  }
}
```
produces:
```
module attributes {firrtl.mainModule = "bar"}  {
  sv.verbatim "// Standard header to adapt well known macros to our needs."
  sv.verbatim ""
  rtl.module @bar() -> (%foo: i65) {
    %c-18446744073709551616_i65 = rtl.constant -18446744073709551616 : i65
    rtl.output %c-18446744073709551616_i65 : i65
  }
}
```

lattner, 2021-04-26:
This isn't a bug - these have the same bit pattern.  Here's proof:

```
% cat x.mlir 

module  {
  rtl.module @bar() -> (%foo: i66) {
    %c-18446744073709551616_i65 = rtl.constant -18446744073709551616 : i65
    %one = rtl.constant 0 : i1
    %a = comb.concat %one, %c-18446744073709551616_i65 : (i1, i65) -> i66
    rtl.output %a : i66
  }
}

% circt-opt x.mlir -simple-canonicalizer
module  {
  rtl.module @bar() -> (%foo: i66) {
    %c18446744073709551616_i66 = rtl.constant 18446744073709551616 : i66
    rtl.output %c18446744073709551616_i66 : i66
  }
}
```

lattner, 2021-04-26:
When printing a signless value, the asmprinter has to decide how to print something.  It chooses to print "255 : i8" as "-1: i8" but the parser takes both formats since they mean the same thing.

lattner, 2021-04-26:
A better example in the i8 domain is 128 vs -128, same deal.

closed 2021-04-26 by lattner