llvm/circt #967: Union type

teqdruid, 2021-04-27:
ESI will have need (potentially soon) for a union type. (For preserving and exposing type information over multiplexed wires, among other things.)

I have two questions:

1. Should this go in the `rtl` dialect along with `array` and `struct`? Or the `sv` dialect? (I think the `rtl` dialect.)
2. Should it be discriminated (tagged) or a C-style union? Should we have both?

I'm torn on number 2. If it's in the `sv` dialect, the semantics are governed by the SV spec. If, however, it's going to be in the `rtl` dialect I could see arguments either way. I think a discriminated union makes sense so as not to lose type information -- SystemVerilog natively supports discriminated (tagged) unions.

Any thoughts?

lattner, 2021-04-27:
The safest/cleanest thing to do is to keep this in the ESI dialect.  SystemVerilog doesn't support this, so you'll have to lower this to structs/bitfields anyway.

teqdruid, 2021-04-27:
> SystemVerilog doesn't support this, so you'll have to lower this to structs/bitfields anyway.

See section 7.3 of the SV spec. SystemVerilog supports unions and tagged unions. It's better than C its union support.

lattner, 2021-04-27:
Oh, thanks for the pointer.  I think it makes perfect sense to add type system support for them then, either to rtl if they are generally synthesizable or to sv if not.

lattner, 2021-04-27:
It looks like packed unions have a lot better [tool support](https://symbiflow.github.io/sv-tests-results/) than tagged ones fwiw, maybe packed goes in RTL and unpacked in SV?

teqdruid, 2021-04-27:
A packed, tagged union is supported (by the spec) but I don't know how well supported it is in the synthesis tools. Does anyone know?

teqdruid, 2021-04-27:
#970 is the RTL model for an undiscriminated (non-tagged) union. I'm thinking if/when we add tagged unions, they'd be called `DiscriminatedUnionType` or something like that.

This will be output in verilog as a packed union. ExportVerilog support in the next PR.

teqdruid, 2021-04-28:
SystemVerilog requires all of the fields in a packed union to be the same size. I consider this to be a quirk of SystemVerilog so I don't want the same restriction to apply to RTL `UnionType`. I'd rather pad them when outputting SystemVerilog. Is this reasonable? If so, where should that padding happen? I'm thinking either ExportVerilog or RTLCleanup.

teqdruid, 2021-04-28:
FYI: Verilator doesn't support tagged unions, so they are not so widely supported.

teqdruid, 2021-04-28:
@darthscsi Could you comment on the right place to do the padding?

darthscsi, 2021-04-29:
I'm trying to think how to pad them cleanly without generating structs.  Do you have a plan here?  The two obvious choices are sturct with the padding in it or a bit vector with a cast/trunc on extraction.  If a struct, we might starting to need a "LegalizeTypes" style pass (probably RTLCleanup would be fine).  If the latter, then ExportVerilog. 

I think I would have ExportVerilog pad the types anytime it printed it and when you had an extract/index operation, generate the depadding code (maybe as simple as .unionfield.realpart).

Alternately reject any non-equal unions in ExportVerilog assuming they will be fixed before then.

teqdruid, 2021-04-29:
> The two obvious choices are sturct with the padding in it

I started down the path of padding by stuffing the offending fields into structs which I would pad in a pre-exportverilog pass. I was using RTLCleanup since you mentioned that it was already required. With this approach, ExportVerilog could just reject any union with differing member sizes.

> a bit vector with a cast/trunc on extraction

I'm shooting for readable, type-safe verilog since these unions will be exposed to manually-written code.

> I think I would have ExportVerilog pad the types anytime it printed it and when you had an extract/index operation, generate the depadding code (maybe as simple as .unionfield.realpart).

This was the alternative I was considering. I suspect it'd be easier, though I feel like representing the padding in the IR data types is the "right" thing to do. If, for instance, another conversion/translation has to reason about the type in a way that corresponds to the generated verilog that code would be SOL.

teqdruid, 2023-04-17:
This is finally done

closed 2023-04-17 by teqdruid