llvm/circt #971: Enumeration type

teqdruid, 2021-04-28:
ESI will also need an enum type. In which dialect should it live? `rtl` or `sv`?

teqdruid, 2021-04-28:
I'm building out the variable-length `list` type in ESI. It will lower to a union value and then synthesize FSMs (which I want to emit an enum).

lattner, 2021-04-28:
Enums are going to be more like interfaces where you need separate declaration and references, right?  Or is there an inline syntax for enums?

teqdruid, 2021-04-28:
```
module test (
  input clk,
  input enum {D, E} a
);
  enum {C, B} x;
  always@(posedge clk) begin
    if (a == D)
      x <= B;
    else
      x <= C;
  end
endmodule
```

Is valid and accepted by both verilator and questa.

teqdruid, 2021-04-28:
Also, I'd ideally want to specify backing numbers for each member so I can do things like specify a one-hot encoding.

lattner, 2021-05-01:
Interesting, does it correctly resolve things like this?

```
module test (
  input clk,
  input enum {B, D} a
);
  enum {C = 2, B = 4} x;
  always@(posedge clk) begin
    if (a == D)
      x <= B;      // Does it know which B??
```

teqdruid, 2021-05-08:
The enum declaration has an error:
```
** Error: enum_test.sv(5): Enum literal name 'B' already exists.
```

teqdruid, 2021-05-08:
Even typedefs don't help:

```
typedef enum {B, D} E1;
typedef enum {C = 2, B = 4} E2;
module test (
  input clk,
  input E1 a
);
  E2 x;
  always@(posedge clk) begin
    if (a == D)
      x <= B;
  end
endmodule
```

```
** Error: enum_test.sv(2): Enum literal name 'B' already exists.
```

So it appears that enum value names are global identifiers!

teqdruid, 2021-05-08:
Typedef'd enums vs not don't seem to make much of a difference, except for two cases:

```
enum {B, D} x;
module test (
  input clk
);
  logic [31:0] B;
  always@(posedge clk) begin
      x <= B;
  end
endmodule

** Error (suppressible): enum_test.sv(7): (vlog-8386) An enum variable 'x' of type '<enum#1>' may only be assigned the same enum typed variable or one of its values. Variable B requires an explicit cast.
```

So if there's an identifier conflict in a particular scope, there's no way to get it to resolve enum.

You also can't do this:

```
module test ();
enum {B, D} y;
enum {B, D} x;
endmodule

** Error: enum_test.sv(3): Enum literal name 'B' already exists.
** Error: enum_test.sv(3): Enum literal name 'D' already exists.
```

SystemVerilog is fun!

teqdruid, 2021-05-08:
To be fair, this seems to match C/C++ behavior.

lattner, 2021-05-08:
I think this comes down to "what is the circt model for an enum type".  I think that the structure should be:
 - It has a name for the enum type.
 - It has a series of enumerators with names and they always have integer values associated with them (the printer can elide consecutive identical ones)
 - The decl is at the top level scope of the file

When printing, we can ameliorate the name collision thing by always printing the enumerators with the enum name as a prefix, giving us some barbarian namespacing:

```
enum {X_B, X_D} X;
enum {Y_B, Y_D} Y;
```

This should be good enough to unblock any immediate needs.

teqdruid, 2021-05-08:
Agreed on all counts.

dtzSiFive, 2023-06-05:
Can this be closed now, with current enum support?

teqdruid, 2023-06-07:
Damn this is an old issue! Yeah, I think so.

closed 2023-06-07 by teqdruid