llvm/circt #976: reg "sink not fully initialized" (false negative)

drom, 2021-04-29:
The following FIRRTL program

```python
circuit top_mod :
  module top_mod :
    input clock: Clock
    input reset: UInt<1>
    reg tmp42: UInt<10>, clock with: (reset => (reset, UInt<10>(0)))
```

Compiled with `firtool --lower-to-rtl --lower-types` generates this error:

```
a_top_mod.fir:6:5: error: sink not fully initialized.
    reg tmp42: UInt<10>, clock with: (reset => (reset, UInt<10>(0)))
    ^
a_top_mod.fir:6:5: note: see current operation: %tmp42 = "firrtl.regreset"(%clock, %reset, %c0_ui10) {annotations = [], name = "tmp42"} : (!firrtl.clock, !firrtl.uint<1>, !firrtl.uint<10>) -> !firrtl.uint<10>
```

Compiled with `firrtl-1.5-SNAPSHOT --dont-fold div` produces this Verilog:

```verilog
module top_mod(
  input   clock,
  input   reset
);
endmodule
```

seldridge, 2021-04-29:
The SFC is adding a self-connection during it's `ExpandWhensAndCheck` pass. The output of it's IR after that point is:

```scala
circuit top_mod :
  module top_mod :
    input clock : Clock
    input reset : UInt<1>

    reg tmp42 : UInt<10>, clock with :
      reset => (reset, UInt<10>("h0"))
    tmp42 <= tmp42
```

I think we need to update CIRCT to add this self-connection (or as part of parsing?).

This feels a little weird to me as it seems like it's masking what is likely a Chisel user error. However, I can see where it does make sense. Registers do not have to be explicitly driven because they are holding their own state.

Regardless, we should not error on this. 

Thanks for the bug report, @drom!

youngar, 2021-04-29:
I opened a PR for this: https://github.com/llvm/circt/pull/979. @seldridge Thanks for the pointer on how to implement this, it made it super easy to fix.

Does SFC actually create a connection when there was none before, or was that for illustration?

seldridge, 2021-04-29:
> Does SFC actually create a connection when there was none before, or was that for illustration?

It creates the connection where there was none before. 

I quickly came to that conclusion by running the SFC with `-cll firrtl.passes.ExpandWhensAndCheck:trace`.

darthscsi, 2021-04-29:
I added the mux canonicalization in RTL to remove the unconditional self-write fallback for conditionally changed registers.  I was wondering where the old value was coming from :)

youngar, 2021-04-30:
PR merged!

closed 2021-04-30 by youngar