llvm/circt #986: [LLHD] Allow llhd.inst to instantiate rtl.module

fabianschuiki, 2021-04-30:
LLHD currently does not allow for RTL modules to be instantiated. Being able to so will be an important step towards co-simulating SystemVerilog/LLHD and RTL Dialect designs. This is likely just a modification in the verifier for the `llhd.inst` operation.

Eventually the following should work:
```
rtl.module @Foo(%a: i1) -> (%b: i1) {
  rtl.output %a : i1
 }

llhd.entity @Bar() -> () {
  %c0_i1 = llhd.const 0 : i1
  %a = llhd.sig "a" %c0_i1 : i1
  %b = llhd.sig "b" %c0_i1 : i1
  llhd.inst "u0" @Foo (%a) -> (%b) : (!llhd.sig<i1>) -> (!llhd.sig<i1>)
}
```
Currently this produces:
```
error: 'llhd.inst' op 'Foo' does not reference a valid proc or entity
```
An important aspect of this support is to bridge the type system gap where LLHD instances require signals to be connected to the ports, but the RTL dialect has no notion of a signal. Basically the signature of the `llhd.inst` should match that of the `rtl.module`, but with all types wrapped in `!llhd.sig<...>`.

mikeurbach, 2021-04-30:
In terms of intermingling LLHD and RTL in the IR, my thought is that any conversions/legalizations/etc. should be done via dialect conversion within the RTLToLLHD pass. It already takes care of converting an `rtl.instance` into a `llhd.inst`, and has all the facilities available to bridge the types systems via materializations: https://github.com/llvm/circt/blob/3e070c2ccb13f12e8988cf5353682f0f58b91088/lib/Conversion/RTLToLLHD/RTLToLLHD.cpp#L190.

My thought was the RTL module and instance concepts are similar enough to LLHD entity and instance concepts that we could just use RTLToLLHD to convert them in the short term, and have llhd-sim support the RTL ops directly in the long term. Is there a use case that requires `llhd.inst` to support `rtl.module` directly, rather than converting with RTLToLLHD?

Other than a handful of RTL ops (constant, arrays, structs), and SV wires, I think we are already pretty close to being able to convert IR with RTL/Comb/Seq into IR with LLHD/Comb/Seq.

fabianschuiki, 2021-04-30:
The main use case I see for this is linking and integrating with testbenches coming from other dialects. For example, a testbench compiled from SystemVerilog with moore will contain a lot of behavioural LLHD operations that we probably never want to merge into the RTL dialect for separation of concerns (maybe a "Sim" dialect?). A common use case I could see is taking an RTL design (e.g. coming from FIRRTL) and linking that with testbench LLHD code (e.g. from SV) to simulate.

A second use case would be optimization in the simulator. I expect us to be much better at optimizing RTL modules because of the lack of timing and strict SSA notion. So one transformation in a simulator could be to take an LLHD design (assuming coming from SV) and "hoisting" the `llhd.entity`s to `rtl.module`s where possible to allow them to be optimized.

For these we need an instantiation of RTL module in LLHD modules that implies a bridging of the timing-aware LLHD world to the timing-unaware RTL world.

closed 2022-03-11 by maerhart