llvm/circt #993: MLIR pass manager not running firrtl.module passes (e.g. CSE) in parallel

lattner, 2021-05-01:
This is because the LLVM taskgroup implementation is really bad.  It doesn't allow nested parallelism, and the top level firrtl.circuit operation is seen as a parallelism region (even though there is only ever one of them).  sigh.

lattner, 2021-05-01:
This is due to this thing in Parallel.cpp:

```
// Latch::sync() called by the dtor may cause one thread to block. If is a dead
// lock if all threads in the default executor are blocked. To prevent the dead
// lock, only allow the first TaskGroup to run tasks parallelly. In the scenario
// of nested parallel_for_each(), only the outermost one runs parallelly.
```

lattner, 2021-05-01:
Fixed/Worked around in upstream in this patch: https://reviews.llvm.org/D101699.  It turns out that TaskGroup could be a lot better, but this should fix it in the important FIRRTL case.

lattner, 2021-05-03:
Push in LLVM 5fa9d4163421. This should come in in our next merge.

closed 2021-05-03 by lattner

mikeurbach, 2021-05-03:
FYI I have a branch building on CI that contains it: https://github.com/llvm/circt/pull/997

avl-llvm, 2023-04-24:
@lattner  Hi Chris,

I would like to propose another workaround than suggested by this issue.
I am currently implementing a check for threadIndex https://reviews.llvm.org/D148916,
which will prevent data races on threadIndex. The possibility to execute parallelFor on the main
thread introduced by https://reviews.llvm.org/D101699  makes it possible
to have a data race if parallelFor is called from different threads:

```
 std::function<void()> Fn = [&]() {
  parallelFor(Passes) { << if Passes contains single entry, the getThreadIndex() will be called on current thread
                        << (instead of threads of ThreadPoolExecutor). 
    getThreadIndex()    << will return 0 for different threads.
  }
};

ThreadPool Pool;

for (Files) {
  Pool.async(Fn);

Pool.wait();

```
I propose to remove the check for NumItems https://reviews.llvm.org/D148916#inline-1438703 
and use the following pattern for the case nested parallelFor:

Instead:

```
parallelFor (Files) {
   parallelFor(Passes) {
   }
}

```
Use:

```
std::function<void()> Fn = [&]() {
  parallelFor(Passes) {
  }
};

ThreadPool Pool;

for (Files) {
  Pool.async(Fn);

Pool.wait();

```

That change would solve the problem from this issue.
It does not create possibility to have the ThreadIndex data race. 
It does not have deadlocks.
And it would give more parallelism for "Files" processing.

What do you think?