llvm/circt #998: Non-deterministic failures in lower-types.mlir tests

mikeurbach, 2021-05-03:
After https://github.com/llvm/circt/pull/997, I seem to be finding some non-deterministic test failures related to LowerTypes.

One issue is related to an index out of bounds error (attributes.txt). I tried adding a bounds check, but that exposed what appears to be the real issue: a firrtl.invalidvalue is being erased which still has uses. I tracked it down to this test: https://github.com/llvm/circt/blob/03de9739896ec4d2ad6df60948cce24185a67dd0/test/Dialect/FIRRTL/lower-types.mlir#L370.

Another issue, which pops up occasionally when the above is commented out, has to do with attempting to drop_front on a string that doesn't have the required number of characters (drop_front.txt). I haven't tracked this down yet, but it is related to the InstanceOp lowering in LowerTypes.

It's strange to me this doesn't get caught in CI, but I guess my workstation has enough parallelism to reveal it, while our build machines don't. I'm also not sure yet if this is a personal problem or others will face it. Is it worth reverting the LLVM version bump for now, since it seems related to parallelism?

[attributes.txt](https://github.com/llvm/circt/files/6417990/attributes.txt)
[drop_front.txt](https://github.com/llvm/circt/files/6417992/drop_front.txt)

mikeurbach, 2021-05-03:
Tagging @lattner @youngar for visibility, I believe you are the ones who have been digging into the parallel_for_each stuff.

youngar, 2021-05-04:
My hunch is that the second issue is related to the fact that the InstanceOp is using the port names from the module.  While lowering, the module and the instance are racing to modify and use the port names.  I think this is also causing the index out of bounds error when printing the IR.  I don't think the use of port names in InstanceOp lowering is needed, we can just pass in an empty string.

I'm not sure what is going on with the first part of the issue, could it be related to module ports in general, and not specific to the InvalidValue test? I'll keep looking in to this!

mikeurbach, 2021-05-04:
> Could it be related to module ports in general, and not specific to the InvalidValue test? I'll keep looking in to this!

Quite possibly. I think this pass has never actually had its data structures accessed concurrently until Chris' patch to LLVM.

lattner, 2021-05-04:
The simplest workaround in the immediate term is to switch this to being a module pass with a for loop.

fabianschuiki, 2021-05-04:
> My hunch is that the second issue is related to the fact that the InstanceOp is using the port names from the module. While lowering, the module and the instance are racing to modify and [use](https://github.com/llvm/circt/blob/main/lib/Dialect/FIRRTL/Transforms/LowerTypes.cpp#L302) the port names.

I noticed a possibly related thing recently when playing around with the RTL-to-LLHD lowering: The `rtl.instance` printing is using the referenced module's port names for printing. But during dialect conversion the target module may be modified in ways the instance doesn't expect, crashing when `--debug-only=dialect-conversion` is used. Maybe we should switch to making the immediate printing/inspection of ops (RTL/FIRRTL instances) independent from the referenced module? (Like operands of and op would keep their type even if the definition disappears or they're nulled)

youngar, 2021-05-04:
I opened a PR which fixes the assert which was firing: https://github.com/llvm/circt/pull/1003

>The rtl.instance printing is using the referenced module's port names for printing. But during dialect conversion the target module may be intermittently modified in ways the instance doesn't expect

After my patch there is still the problem of IR being unable to print properly due to inconsistent IR when the assert fires (attributes.txt is actually an assert inside an assert).

mikeurbach, 2021-05-04:
> After my patch there is still the problem of IR being unable to print properly due to inconsistent IR when the assert fires (attributes.txt is actually an assert inside an assert).

Right, it asserts when trying to print, and I believe the reason it was trying to print was because an op was deleted with uses. I'll take another look on this now your other patch is in.

mikeurbach, 2021-05-04:
The patch looks good to me. I just ran the test case that produced the attributes.txt error originally about 20 times, and saw no failures. Is there a reliable way to reproduce this now?

youngar, 2021-05-04:
> The patch looks good to me. I just ran the test case that produced the attributes.txt error originally about 20 times, and saw no failures. Is there a reliable way to reproduce this now?

Hey, sorry for the late reply.  I think we can close this issue now, since the main problem has been fixed. Each of the original problems was just a different symptom of the same problem.

I opened a separate issue for the op printer crash https://github.com/llvm/circt/issues/1004.  As for reproducing it, there shouldn't be any normal way to get it to crash.  The only way to get it to crash is to dump the IR mid-transformation.

closed 2021-05-04 by mikeurbach