llvm/circt #1001: LowerTypes Parallelism Bug?

seldridge, 2021-05-04:
I'm seeing issues on `main` where FIRRT's LowerTypes transform is trying to erase block args which still have uses. This only happens with multithreading on.

You should be able to reproduce with the following `Foo.mlir`:

```mlir
module  {
  firrtl.circuit "Foo"  {
    firrtl.module @Bar(%a: !firrtl.bundle<a: uint<1>>, %b: !firrtl.flip<bundle<a: uint<1>>>) {
      firrtl.connect %b, %a : !firrtl.flip<bundle<a: uint<1>>>, !firrtl.bundle<a: uint<1>>
    }
    firrtl.module @Foo(%a: !firrtl.bundle<a: uint<1>>, %b: !firrtl.flip<bundle<a: uint<1>>>) {
      %bar_a, %bar_b = firrtl.instance @Bar  {name = "bar"} : !firrtl.flip<bundle<a: uint<1>>>, !firrtl.bundle<a: uint<1>>
      firrtl.connect %bar_a, %a : !firrtl.flip<bundle<a: uint<1>>>, !firrtl.bundle<a: uint<1>>
      firrtl.connect %b, %bar_b : !firrtl.flip<bundle<a: uint<1>>>, !firrtl.bundle<a: uint<1>>
    }
  }
}
```

Then put this in a while loop:

```bash
while circt-opt -pass-pipeline='firrtl.circuit(firrtl-lower-types)' Foo.mlir; do echo "works"; done
```

This should eventually crash.

Alternatively, running without parallelism on shouldn't produce a crash:

```bash
while circt-opt -mlir-disable-threading -pass-pipeline='firrtl.circuit(firrtl-lower-types)' Foo.mlir; do echo "works"; done
```

An example error log is below:

<details>
 <summary>Full log</summary>

```text
Assertion failed: (use_empty() && "Cannot destroy a value that still has uses!"), function ~IRObjectWithUseList, file /Users/schuylere/repos/github.com/llvm/circt/llvm/mlir/include/mlir/IR/UseDefLists.h, line 37.
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.	Program arguments: circt-opt -pass-pipeline=firrtl.circuit(firrtl-lower-types) Foo.mlir
1.	2.	Stack dump without symbol names (ensure you have llvm-symbolizer in your PATH or set the environment var `LLVM_SYMBOLIZER_PATH` to point to it):
0  circt-opt                0x000000010fc8897d llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) + 61
1  circt-opt                0x000000010fc88f2b PrintStackTraceSignalHandler(void*) + 27
2  circt-opt                0x000000010fc86c8b llvm::sys::RunSignalHandlers() + 123
3  circt-opt                0x000000010fc8b598 SignalHandler(int) + 232
4  libsystem_platform.dylib 0x00007fff20384d7d _sigtramp + 29
5  circt-opt                0x00000001111e9f80 _dyld_private + 0
6  libsystem_c.dylib        0x00007fff20294411 abort + 120
7  libsystem_c.dylib        0x00007fff202937e8 err + 0
8  circt-opt                0x00000001102778c2 mlir::IRObjectWithUseList<mlir::OpOperand>::~IRObjectWithUseList() + 98
9  circt-opt                0x0000000110277858 mlir::detail::ValueImpl::~ValueImpl() + 24
10 circt-opt                0x0000000110277838 mlir::detail::BlockArgumentImpl::~BlockArgumentImpl() + 24
11 circt-opt                0x0000000110277815 mlir::detail::BlockArgumentImpl::~BlockArgumentImpl() + 21
12 circt-opt                0x0000000110274f4c mlir::BlockArgument::destroy() + 44
13 circt-opt                0x000000011027670d mlir::Block::eraseArguments(llvm::function_ref<bool (mlir::BlockArgument)>) + 173
14 circt-opt                0x0000000110276658 mlir::Block::eraseArguments(llvm::BitVector const&) + 72
15 circt-opt                0x00000001102765d3 mlir::Block::eraseArguments(llvm::ArrayRef<unsigned int>) + 179
16 circt-opt                0x000000010fcc4fed (anonymous namespace)::TypeLoweringVisitor::visitDecl(circt::firrtl::FModuleOp) + 1549
17 circt-opt                0x000000010fcc4233 (anonymous namespace)::TypeLoweringVisitor::lowerModule(mlir::Operation*) + 83
18 circt-opt                0x000000010fcc27f3 auto (anonymous namespace)::LowerTypesPass::runAsync()::$_8::operator()<unsigned long>(unsigned long) const + 147
19 circt-opt                0x000000010fcc270f void llvm::parallel::detail::parallel_for_each_n<unsigned long, (anonymous namespace)::LowerTypesPass::runAsync()::$_8>(unsigned long, unsigned long, (anonymous namespace)::LowerTypesPass::runAsync()::$_8) + 415
20 circt-opt                0x000000010fcc135d void llvm::parallelForEachN<(anonymous namespace)::LowerTypesPass::runAsync()::$_8>(unsigned long, unsigned long, (anonymous namespace)::LowerTypesPass::runAsync()::$_8) + 109
21 circt-opt                0x000000010fcc111f (anonymous namespace)::LowerTypesPass::runAsync() + 191
22 circt-opt                0x000000010fcc02dd (anonymous namespace)::LowerTypesPass::runOnOperation() + 61
23 circt-opt                0x0000000110afaadb mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) + 587
24 circt-opt                0x0000000110afb137 mlir::detail::OpToOpPassAdaptor::runPipeline(llvm::iterator_range<llvm::pointee_iterator<std::__1::unique_ptr<mlir::Pass, std::__1::default_delete<mlir::Pass> >*, mlir::Pass> >, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) + 359
25 circt-opt                0x0000000110b0fa3f mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_8::operator()(llvm::MutableArrayRef<mlir::OpPassManager>) const + 559
26 circt-opt                0x0000000110b0f558 void llvm::parallel::detail::parallel_for_each<llvm::SmallVector<mlir::OpPassManager, 1u>*, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_8>(llvm::SmallVector<mlir::OpPassManager, 1u>*, llvm::SmallVector<mlir::OpPassManager, 1u>*, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_8) + 152
27 circt-opt                0x0000000110afc984 void llvm::parallelForEach<llvm::SmallVector<mlir::OpPassManager, 1u>*, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_8>(llvm::SmallVector<mlir::OpPassManager, 1u>*, llvm::SmallVector<mlir::OpPassManager, 1u>*, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_8) + 180
28 circt-opt                0x0000000110afbf6f mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool) + 1279
29 circt-opt                0x0000000110afad8d mlir::detail::OpToOpPassAdaptor::runOnOperation(bool) + 77
30 circt-opt                0x0000000110afaac6 mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) + 566
31 circt-opt                0x0000000110afb137 mlir::detail::OpToOpPassAdaptor::runPipeline(llvm::iterator_range<llvm::pointee_iterator<std::__1::unique_ptr<mlir::Pass, std::__1::default_delete<mlir::Pass> >*, mlir::Pass> >, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) + 359
32 circt-opt                0x0000000110afda69 mlir::PassManager::run(mlir::Operation*) + 985
33 circt-opt                0x00000001103fccf8 performActions(llvm::raw_ostream&, bool, bool, llvm::SourceMgr&, mlir::MLIRContext*, mlir::PassPipelineCLParser const&) + 520
34 circt-opt                0x00000001103fb36b processBuffer(llvm::raw_ostream&, std::__1::unique_ptr<llvm::MemoryBuffer, std::__1::default_delete<llvm::MemoryBuffer> >, bool, bool, bool, bool, mlir::PassPipelineCLParser const&, mlir::DialectRegistry&) + 331
35 circt-opt                0x00000001103fb166 mlir::MlirOptMain(llvm::raw_ostream&, std::__1::unique_ptr<llvm::MemoryBuffer, std::__1::default_delete<llvm::MemoryBuffer> >, mlir::PassPipelineCLParser const&, mlir::DialectRegistry&, bool, bool, bool, bool, bool) + 358
36 circt-opt                0x00000001103fc0fe mlir::MlirOptMain(int, char**, llvm::StringRef, mlir::DialectRegistry&, bool) + 3310
37 circt-opt                0x000000010fab5b5f main + 159
38 libdyld.dylib            0x00007fff2035af3d start + 1
39 libdyld.dylib            0x0000000000000003 start + 18446603339975774407
```

</details>

mikeurbach, 2021-05-04:
See #998

seldridge, 2021-05-04:
Closing as dupe. Thanks for the other issue @mikeurbach!

closed 2021-05-04 by seldridge