llvm/circt #1004: [FIRRTL] Crashing in the Op printer for InstanceOp

youngar, 2021-05-04:
The InstanceOp printer relies on the port names retained in the ModuleOp.  When a pass makes a change to the port names, such as happens in LowerTypes, the InstanceOp and ModuleOp can have a different number of ports.  When printing the instance operation while the IR is inconsistent, the printer will crash.  This can only happen when printing the InstanceOp while the IR is mid-transformation, otherwise the verifier would catch the issue.

You can get it to crash in the IR printer by inserting a `op->dump()` in `TypeLoweringVisitor::visitDecl(InstanceOp op)`. 

Maybe a good rule of thumb is that an operations ASM printer should not rely on the consistency of any other operation? Otherwise we should attempt to make IR printer resilient to this issue.

More information in thread starting from here: https://github.com/llvm/circt/issues/998#issuecomment-831703089

lattner, 2021-05-04:
One approach to this is to use the `--mlir-print-op-generic  ` flag to the tool, which ignores custom printers.   It is listed in --help-hidden output from firtool etc.  

That said, if this is particularly fragile, it is ok with me to add a bit of defensive code.

lattner, 2021-05-04:
@fabianschuiki I realize that this flag was in the --help-hidden output which is why I didn't see it on firtool, but it is there.

fabianschuiki, 2021-05-05:
There was also this discussion: https://github.com/llvm/circt/pull/988#discussion_r625569281

> One approach to this is to use the `--mlir-print-op-generic` flag to the tool, which ignores custom printers.

@lattner The problem is that the [asm printer (SSANameState)](https://github.com/llvm/llvm-project/blob/354604a2a7149b5efd52134efa4765cf8c32e386/mlir/lib/IR/AsmPrinter.cpp#L1004-L1007) calls `getAsmResultNames()` regardless of whether generic or custom printers are used. So the `--mlir-print-op-generic` flag won't prevent this particular crash.

> That said, if this is particularly fragile, it is ok with me to add a bit of defensive code.

I think it would be great if we could make any code path that gets called from the generic printer as robust as the generic printer itself :100:.

lattner, 2021-05-05:
Sure, feel free to add a bit of defensive code here.  I suspect it is a bug that the generic printer is calling `getAsmResultNames` when that flag is passed!

fabianschuiki, 2021-05-10:
Related: #1034