llvm/circt #1013: Annotation Verification

seldridge, 2021-05-06:
FIRRTL Annotations that are parsed into FIRRTL Dialect should be verified in two ways:

1. Any annotations which do not apply to the circuit should be a disable-able error.
2. Annotations which are applied and include subfield/subindex sub-targets should be validated that they point to legal things.

(1) is important because it helps us to know that we're pulling in the right annotation file, that the annotation file isn't malformed, and that our annotation parsing works. By default, weird-looking annotations are promoted to circuit target annotations. This should start erroring so that we know what custom annotations we need to worry about.

(2) Matters for things like lowering. FIRRTL allows for field names to be numeric. This can result in annotations incorrectly aliasing to the wrong thing if they aren't verified first. Consider the following where `a[0]` doesn't exist and the lowering step will incorrectly apply this to a new wire `a_0`:

```scala
circuit Baz: %[[{"a":null,"target":"~Baz|Baz>a[0]"}]]
  module Baz:
    wire a: {0: UInt<1>}
```

There are more pedestrian cases of just incorrect annotations pointing to things that don't exist:

```scala
circuit Baz: %[[{"a":null,"target":"~Baz|Baz>a.b"}]]
  module Baz:
    wire a: UInt<1>
```

hanchenye, 2021-06-25:
Done in #1280 and #1327

seldridge, 2021-06-25:
Re-opening this to track the first issue of annotations that _don't apply_ to the circuit should also error. E.g.:

```scala
circuit Baz: %[[{"a":null,"target":"~Baz|Baz>b"}]]
  module Baz:
    input a: UInt<1>
```

The work in #1280 and #1327 look great, though I wonder if #1280 and the issue above could be more easily handled with just catching annotations which don't apply to the circuit.

hanchenye, 2021-06-25:
Good point... Missed this issue. For now, this annotation will be inserted into the `annotationsMap` but not be attached to any operation or port by the parser. Don't have a good answer on this for now :cry:

PR #1440 (2021-07-22): [FIRRTL] Error on Annotations that Do Not Apply
Change the FIRRTL parser to error if any Annotations are unapplied to
the circuit.  Put differently, this will error if any Target in an
Annotation (or a generated Target from Annotation scattering) points at
something which does not exist in the circuit.

Residual Annotations indicate either a user problem like inconsistent
FIRRTL and Annotation files or some problem with scattering logic or the
parser itself.  E.g., you can hit this if you have busted scattering
code (that creates targets that do not exist) or for cases where
non-local Annotati [... cut]
closed 2021-07-22 by seldridge