llvm/circt #1027: [Handshake] Integration tests failing

teqdruid, 2021-05-07:
```
Failed Tests (2):
  CIRCT :: Dialect/Handshake/loop.mlir
  CIRCT :: Dialect/Handshake/memory.mlir
```

`loop.mlir` seems to pass non-deterministically.

mikeurbach, 2021-05-07:
What simulator are you using? Those were passing for me with Vivado. If it's really simulator-dependent, I will only enable those tests for Vivado.

teqdruid, 2021-05-07:
I'm running with Questa.

mikeurbach, 2021-05-07:
Ok, I pushed this just now to only turn these on for Vivado: https://github.com/llvm/circt/commit/18bf7f2cfe4fc10630e262d26d8061346652ac6b.

teqdruid, 2021-05-07:
> Ok, I pushed this just now to only turn these on for Vivado: 18bf7f2

I assume that's a temporary bandaid?

mikeurbach, 2021-05-07:
Frankly I have no idea why one commerical simulator will succeed and one will fail. I'm very interested in testing Handshake with Vivado, but I'm less interested in comparing Vivado to Questa. I think there's value in having these tests, even if they don't work on every simulator. I wouldn't consider it a long term solution, but I don't see any problem keeping it like this until someone cares to try another simulator.

teqdruid, 2021-05-07:
These don't look like simulation failures...

[testfails.log](https://github.com/llvm/circt/files/6444549/testfails.log)

teqdruid, 2021-05-07:
I think it would, yes. I thought I did set it up so that each test would execute in a separate dir. That may have been in the ESI cosim test runner though...

When I run the tests without any lit parallelism they pass:
```
15:35 $ lit build/integration_test/Dialect/Handshake/ -sv -j1

Testing Time: 6.10s
  Passed: 3
```

mikeurbach, 2021-05-07:
Got it, that makes sense. Thanks for checking.

teqdruid, 2021-05-07:
I've got circt-rtl-sim.py compiling and running everything in its own subdir and the handshake tests are now working.