llvm/circt #1034: Segfault in test/Dialect/RTL/errors.mlir

darthscsi, 2021-05-09:
/home/andrew/src/circt/test/Dialect/RTL/errors.mlir segfaults in getModuleResultNameAttr for the test of incorrect return result count.  The generic op printer when called from emitError uses getAsmResultNames gets the names of the returns on the referenced module based on the number of results on the instance, which is exactly what the test case is looking for.

I haven't looked into why a segfault passes lit.

darthscsi, 2021-05-09:
```
andrew@narya:~/src/circt/build$ /home/andrew/src/circt/build/bin/circt-opt /home/andrew/src/circt/test/Dialect/RTL/errors.mlir -split-input-file -verify-diagnostics
/home/andrew/src/circt/build/bin/circt-opt /home/andrew/src/circt/test/Dialect/RTL/errors.mlir -split-input-file
[Normal stuff elided]
circt-opt: ../llvm/build/tools/mlir/include/mlir/IR/BuiltinAttributes.h.inc:62: mlir::Attribute mlir::ArrayAttr::operator[](unsigned int) const: Assertion `idx < size() && "index out of bounds"' failed.
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.      Program arguments: /home/andrew/src/circt/build/bin/circt-opt /home/andrew/src/circt/test/Dialect/RTL/errors.mlir -split-input-file
...
```

-verify-diagnostics masks the segfault.

lattner, 2021-05-15:
Fixed in 86911192

closed 2021-05-15 by lattner