llvm/circt #1047: Tools in CI crash across the board

fabianschuiki, 2021-05-13:
Added two minimal changes which now cause CI to produce tools and tests that all crash in weird locations: https://github.com/llvm/circt/runs/2575814179. Seems unrelated to the change. PR with the changes reverted to check if they somehow caused the problem: #1046.

fabianschuiki, 2021-05-13:
#1046 breaks as well. The PR has the identical state as the last build that succeeded yesterday (https://github.com/llvm/circt/runs/2572839614). Maybe some build cache got corrupted?

seldridge, 2021-05-13:
As a further data point, `circt/images` got through tests just fine (https://github.com/circt/images/runs/2574886938?check_suite_focus=true#step:3:9229) using [5712680b435cc6ae4bc9439f3112e0894b5bade2](https://github.com/circt/images/runs/2574886938?check_suite_focus=true#step:3:94) (which is what either the last `llvm/circt` commit to pass or the effective state with #1046 applied). 🤷‍♀️

mikeurbach, 2021-05-13:
Very strange... I am compiling locally with the new version of LLVM and I'll try to help triage.

mikeurbach, 2021-05-13:
We did recently change our buildbot and how LLVM is cached. The cache key is a combination of the LLVM commit sha and a md5 of the workflow file's contents: https://github.com/llvm/circt/blob/main/.github/workflows/buildAndTest.yml#L45. I believe I saw the workflow rebuilding the LLVM cache as expected in the most recent LLVM bump.

youngar, 2021-05-13:
The output of the caching job makes it look like the hash key has a space in it:
```
Run actions/cache@v2
  with:
    path: llvm/build
  llvm/install
  
    key: Linux-llvm-33f908c42881fa02963f0c64f8be5088717664cc-952a59dd748f2e22ab02a6603bf38bee  /home/runner/work/circt/circt/.github/workflows/buildAndTest.yml
    ```

mikeurbach, 2021-05-13:
Interesting... I'm not sure where the space and path to the file is coming from in that print out. We are setting the key like this: https://github.com/llvm/circt/blob/e117ac01bdd9954bafbd3732f30d662372f2bd91/.github/workflows/buildAndTest.yml#L45

fabianschuiki, 2021-05-13:
@youngar pointed out that we just use the output of `md5sum` for the workflow hash, which contains the space and file name after the hash.

youngar, 2021-05-13:
The NPComp key (where we stole the idea from) has tons of spaces:
```
Run actions/cache@v2
  with:
    path: external/llvm-project
    key: Linux-llvm-20.04-install- 6d263b6f1c97fe6c45c75443e7daf6cd0c1c4222 external/llvm-project (6d263b6f1)-5dd262dc1c6dc41a092f589d20595f80  /home/runner/work/mlir-npcomp/mlir-npcomp/.github/workflows/buildAndTest.yml
  env:
    LLVM: external/llvm-project
Cache not found for input keys: Linux-llvm-20.04-install- 6d263b6f1c97fe6c45c75443e7daf6cd0c1c4222 external/llvm-project (6d263b6f1)-5dd262dc1c6dc41a092f589d20595f80  /home/runner/work/mlir-npcomp/mlir-npcomp/.github/workflows/buildAndTest.yml
```

seldridge, 2021-05-13:
Yeah, the spaces seem fine... The GitHub actions codebase eventually bottoms out with a `localeCompare` on the two keys. That seemed to do the right thing with spaces for the options they're using. 🤔

youngar, 2021-05-13:
Even if spaces are not the issue, it might make sense to use the github provided [function to hash files](https://docs.github.com/en/actions/reference/context-and-expression-syntax-for-github-actions#hashfiles), which is what is used in the cache docs.

teqdruid, 2021-05-14:
The nightly tests don't produce non-working builds like this. Clearly something changed in the GH build environment. Since we're not able to debug in that environment, I'll switch us to using a Docker environment.

teqdruid, 2021-05-14:
@GeorgeLyon @lattner @stephenneuendorffer @darthscsi
I'm trying to publish a Docker image for our CI builds so we have some sort of build stability and are not affected by GitHub environment changes. The new image isn't appearing. By default, new images are private so that's probably the reason. You are all owners on the CIRCT org, can you check https://github.com/orgs/circt/packages for an "images/circt-ci-build" image and make it public?

mikeurbach, 2021-05-14:
Would the latest integration test image work for now? That image runs all the regular tests nightly as well.

mikeurbach, 2021-05-14:
I'm +1 on using a docker image in general, this is not the first time we've gotten burned by githubs machine image.

teqdruid, 2021-05-14:
> Would the latest integration test image work for now?

That image runs 20.04 (not a big deal IMO) and has capnp installed so it runs capnp builds... I tend to break non-capnp builds so this is a check on me.

I think we should switch over if I can't get the new image working soon-ish.

teqdruid, 2021-05-14:
The new workflow is in this PR: https://github.com/llvm/circt/pull/1053 if someone wants to run/merge it after one of the owners flicks the switch.

mikeurbach, 2021-05-14:
It looks like github released a new version of ubuntu 18.04 on 5/10: https://github.com/actions/virtual-environments/commit/8441c418f8f1d468bd6a5bb63f73f1e4b0bd5ff3. Not sure what yet but I assume something in there is the issue. Docker ftw!

teqdruid, 2021-05-16:
An update on this: I have the build working in a 20.04 docker image. It's just failing the `git clang-format` step only because I haven't had time to fix it.

It turns out that the tests crash in an 18.04 docker image as well and I have a local repro (using docker) which I haven't pushed yet (haven't been at my desk in a while). Is anyone using Ubuntu 18.04?

closed 2021-05-17 by teqdruid

teqdruid, 2021-05-17:
I upgraded to Ubuntu 20.04 in docker and the builds are green again. I have a local repro of what was going on and have file a new issue #1065.

PR #1061 (2021-05-17): [CI] Upgrade to Ubuntu 20.04 docker image
If this CI build passes, we're good to go. If anyone sees three green checks, please merge as I will be away from my desk.

Fixes #1047 ... hopefully.