llvm/circt #1048: Compilation failed with Ninja on Apple M1

shi27feng, 2021-05-13:
Hi all,
Does anyone know what causes the below error?
Thanks in advance

```shell
❯ ninja
[1/2791] Building IntrinsicsAMDGPU.h...
FAILED: include/llvm/IR/IntrinsicsAMDGPU.h
cd /Users/fengshi/Workspace/circt/llvm/build && /Users/fengshi/Workspace/circt/llvm/build/bin/llvm-tblgen -gen-intrinsic-enums -intrinsic-prefix=amdgcn -I /Users/fengshi/Workspace/circt/llvm/llvm/include/llvm/IR -I/Users/fengshi/Workspace/circt/llvm/build/include -I/Users/fengshi/Workspace/circt/llvm/llvm/include /Users/fengshi/Workspace/circt/llvm/llvm/include/llvm/IR/Intrinsics.td --write-if-changed -o include/llvm/IR/IntrinsicsAMDGPU.h -d include/llvm/IR/IntrinsicsAMDGPU.h.d
dyld: Library not loaded: @rpath/libtinfo.6.dylib
  Referenced from: /Users/fengshi/Workspace/circt/llvm/build/bin/llvm-tblgen
  Reason: image not found
/bin/sh: line 1: 78988 Abort trap: 6           /Users/fengshi/Workspace/circt/llvm/build/bin/llvm-tblgen -gen-intrinsic-enums -intrinsic-prefix=amdgcn -I /Users/fengshi/Workspace/circt/llvm/llvm/include/llvm/IR -I/Users/fengshi/Workspace/circt/llvm/build/include -I/Users/fengshi/Workspace/circt/llvm/llvm/include /Users/fengshi/Workspace/circt/llvm/llvm/include/llvm/IR/Intrinsics.td --write-if-changed -o include/llvm/IR/IntrinsicsAMDGPU.h -d include/llvm/IR/IntrinsicsAMDGPU.h.d
[2/2791] Building IntrinsicsARM.h...
FAILED: include/llvm/IR/IntrinsicsARM.h
cd /Users/fengshi/Workspace/circt/llvm/build && /Users/fengshi/Workspace/circt/llvm/build/bin/llvm-tblgen -gen-intrinsic-enums -intrinsic-prefix=arm -I /Users/fengshi/Workspace/circt/llvm/llvm/include/llvm/IR -I/Users/fengshi/Workspace/circt/llvm/build/include -I/Users/fengshi/Workspace/circt/llvm/llvm/include /Users/fengshi/Workspace/circt/llvm/llvm/include/llvm/IR/Intrinsics.td --write-if-changed -o include/llvm/IR/IntrinsicsARM.h -d include/llvm/IR/IntrinsicsARM.h.d
dyld: Library not loaded: @rpath/libtinfo.6.dylib
  Referenced from: /Users/fengshi/Workspace/circt/llvm/build/bin/llvm-tblgen
  Reason: image not found
/bin/sh: line 1: 78990 Abort trap: 6           /Users/fengshi/Workspace/circt/llvm/build/bin/llvm-tblgen -gen-intrinsic-enums -intrinsic-prefix=arm -I /Users/fengshi/Workspace/circt/llvm/llvm/include/llvm/IR -I/Users/fengshi/Workspace/circt/llvm/build/include -I/Users/fengshi/Workspace/circt/llvm/llvm/include /Users/fengshi/Workspace/circt/llvm/llvm/include/llvm/IR/Intrinsics.td --write-if-changed -o include/llvm/IR/IntrinsicsARM.h -d include/llvm/IR/IntrinsicsARM.h.d
[3/2791] Building IntrinsicsPowerPC.h...
FAILED: include/llvm/IR/IntrinsicsPowerPC.h
cd /Users/fengshi/Workspace/circt/llvm/build && /Users/fengshi/Workspace/circt/llvm/build/bin/llvm-tblgen -gen-intrinsic-enums -intrinsic-prefix=ppc -I /Users/fengshi/Workspace/circt/llvm/llvm/include/llvm/IR -I/Users/fengshi/Workspace/circt/llvm/build/include -I/Users/fengshi/Workspace/circt/llvm/llvm/include /Users/fengshi/Workspace/circt/llvm/llvm/include/llvm/IR/Intrinsics.td --write-if-changed -o include/llvm/IR/IntrinsicsPowerPC.h -d include/llvm/IR/IntrinsicsPowerPC.h.d
dyld: Library not loaded: @rpath/libtinfo.6.dylib
  Referenced from: /Users/fengshi/Workspace/circt/llvm/build/bin/llvm-tblgen
  Reason: image not found
/bin/sh: line 1: 78987 Abort trap: 6           /Users/fengshi/Workspace/circt/llvm/build/bin/llvm-tblgen -gen-intrinsic-enums -intrinsic-prefix=ppc -I /Users/fengshi/Workspace/circt/llvm/llvm/include/llvm/IR -I/Users/fengshi/Workspace/circt/llvm/build/include -I/Users/fengshi/Workspace/circt/llvm/llvm/include /Users/fengshi/Workspace/circt/llvm/llvm/include/llvm/IR/Intrinsics.td --write-if-changed -o include/llvm/IR/IntrinsicsPowerPC.h -d include/llvm/IR/IntrinsicsPowerPC.h.d
[4/2791] Building IntrinsicEnums.inc...
FAILED: include/llvm/IR/IntrinsicEnums.inc
cd /Users/fengshi/Workspace/circt/llvm/build && /Users/fengshi/Workspace/circt/llvm/build/bin/llvm-tblgen -gen-intrinsic-enums -I /Users/fengshi/Workspace/circt/llvm/llvm/include/llvm/IR -I/Users/fengshi/Workspace/circt/llvm/build/include -I/Users/fengshi/Workspace/circt/llvm/llvm/include /Users/fengshi/Workspace/circt/llvm/llvm/include/llvm/IR/Intrinsics.td --write-if-changed -o include/llvm/IR/IntrinsicEnums.inc -d include/llvm/IR/IntrinsicEnums.inc.d
dyld: Library not loaded: @rpath/libtinfo.6.dylib
  Referenced from: /Users/fengshi/Workspace/circt/llvm/build/bin/llvm-tblgen
  Reason: image not found
/bin/sh: line 1: 78985 Abort trap: 6           /Users/fengshi/Workspace/circt/llvm/build/bin/llvm-tblgen -gen-intrinsic-enums -I /Users/fengshi/Workspace/circt/llvm/llvm/include/llvm/IR -I/Users/fengshi/Workspace/circt/llvm/build/include -I/Users/fengshi/Workspace/circt/llvm/llvm/include /Users/fengshi/Workspace/circt/llvm/llvm/include/llvm/IR/Intrinsics.td --write-if-changed -o include/llvm/IR/IntrinsicEnums.inc -d include/llvm/IR/IntrinsicEnums.inc.d
[5/2791] Building IntrinsicImpl.inc...
FAILED: include/llvm/IR/IntrinsicImpl.inc
cd /Users/fengshi/Workspace/circt/llvm/build && /Users/fengshi/Workspace/circt/llvm/build/bin/llvm-tblgen -gen-intrinsic-impl -I /Users/fengshi/Workspace/circt/llvm/llvm/include/llvm/IR -I/Users/fengshi/Workspace/circt/llvm/build/include -I/Users/fengshi/Workspace/circt/llvm/llvm/include /Users/fengshi/Workspace/circt/llvm/llvm/include/llvm/IR/Intrinsics.td --write-if-changed -o include/llvm/IR/IntrinsicImpl.inc -d include/llvm/IR/IntrinsicImpl.inc.d
dyld: Library not loaded: @rpath/libtinfo.6.dylib
  Referenced from: /Users/fengshi/Workspace/circt/llvm/build/bin/llvm-tblgen
  Reason: image not found
/bin/sh: line 1: 78983 Abort trap: 6           /Users/fengshi/Workspace/circt/llvm/build/bin/llvm-tblgen -gen-intrinsic-impl -I /Users/fengshi/Workspace/circt/llvm/llvm/include/llvm/IR -I/Users/fengshi/Workspace/circt/llvm/build/include -I/Users/fengshi/Workspace/circt/llvm/llvm/include /Users/fengshi/Workspace/circt/llvm/llvm/include/llvm/IR/Intrinsics.td --write-if-changed -o include/llvm/IR/IntrinsicImpl.inc -d include/llvm/IR/IntrinsicImpl.inc.d
[6/2791] Building IntrinsicsBPF.h...
FAILED: include/llvm/IR/IntrinsicsBPF.h
cd /Users/fengshi/Workspace/circt/llvm/build && /Users/fengshi/Workspace/circt/llvm/build/bin/llvm-tblgen -gen-intrinsic-enums -intrinsic-prefix=bpf -I /Users/fengshi/Workspace/circt/llvm/llvm/include/llvm/IR -I/Users/fengshi/Workspace/circt/llvm/build/include -I/Users/fengshi/Workspace/circt/llvm/llvm/include /Users/fengshi/Workspace/circt/llvm/llvm/include/llvm/IR/Intrinsics.td --write-if-changed -o include/llvm/IR/IntrinsicsBPF.h -d include/llvm/IR/IntrinsicsBPF.h.d
dyld: Library not loaded: @rpath/libtinfo.6.dylib
  Referenced from: /Users/fengshi/Workspace/circt/llvm/build/bin/llvm-tblgen
  Reason: image not found
/bin/sh: line 1: 78981 Abort trap: 6           /Users/fengshi/Workspace/circt/llvm/build/bin/llvm-tblgen -gen-intrinsic-enums -intrinsic-prefix=bpf -I /Users/fengshi/Workspace/circt/llvm/llvm/include/llvm/IR -I/Users/fengshi/Workspace/circt/llvm/build/include -I/Users/fengshi/Workspace/circt/llvm/llvm/include /Users/fengshi/Workspace/circt/llvm/llvm/include/llvm/IR/Intrinsics.td --write-if-changed -o include/llvm/IR/IntrinsicsBPF.h -d include/llvm/IR/IntrinsicsBPF.h.d
[7/2791] Building IntrinsicsNVPTX.h...
FAILED: include/llvm/IR/IntrinsicsNVPTX.h
cd /Users/fengshi/Workspace/circt/llvm/build && /Users/fengshi/Workspace/circt/llvm/build/bin/llvm-tblgen -gen-intrinsic-enums -intrinsic-prefix=nvvm -I /Users/fengshi/Workspace/circt/llvm/llvm/include/llvm/IR -I/Users/fengshi/Workspace/circt/llvm/build/include -I/Users/fengshi/Workspace/circt/llvm/llvm/include /Users/fengshi/Workspace/circt/llvm/llvm/include/llvm/IR/Intrinsics.td --write-if-changed -o include/llvm/IR/IntrinsicsNVPTX.h -d include/llvm/IR/IntrinsicsNVPTX.h.d
dyld: Library not loaded: @rpath/libtinfo.6.dylib
  Referenced from: /Users/fengshi/Workspace/circt/llvm/build/bin/llvm-tblgen
  Reason: image not found
/bin/sh: line 1: 78986 Abort trap: 6           /Users/fengshi/Workspace/circt/llvm/build/bin/llvm-tblgen -gen-intrinsic-enums -intrinsic-prefix=nvvm -I /Users/fengshi/Workspace/circt/llvm/llvm/include/llvm/IR -I/Users/fengshi/Workspace/circt/llvm/build/include -I/Users/fengshi/Workspace/circt/llvm/llvm/include /Users/fengshi/Workspace/circt/llvm/llvm/include/llvm/IR/Intrinsics.td --write-if-changed -o include/llvm/IR/IntrinsicsNVPTX.h -d include/llvm/IR/IntrinsicsNVPTX.h.d
[8/2791] Building IntrinsicsAArch64.h...
FAILED: include/llvm/IR/IntrinsicsAArch64.h
[... cut ...]
/bin/sh: line 1: 78984 Abort trap: 6           /Users/fengshi/Workspace/circt/llvm/build/bin/llvm-tblgen -gen-intrinsic-enums -intrinsic-prefix=hexagon -I /Users/fengshi/Workspace/circt/llvm/llvm/include/llvm/IR -I/Users/fengshi/Workspace/circt/llvm/build/include -I/Users/fengshi/Workspace/circt/llvm/llvm/include /Users/fengshi/Workspace/circt/llvm/llvm/include/llvm/IR/Intrinsics.td --write-if-changed -o include/llvm/IR/IntrinsicsHexagon.h -d include/llvm/IR/IntrinsicsHexagon.h.d
ninja: build stopped: subcommand failed.
```

teqdruid, 2021-05-13:
These are LLVM compile failures, so you're better off asking there. I don't
think they support compiling on M1 yet, but I am unsure.

lattner, 2021-05-13:
LLVM definitely builds on M1.  I'd agree that asking on llvm-dev mailing list is the best way to go.

fyquah, 2021-07-02:
In case this has not been resolved:

The README mentions compiling with `-DLLVM_TARGETS_TO_BUILD="x86; riscv"`. I think you need to set `LLVM_TARGETS_TO_BUILD` as `host`. Your cmake invocation should look something like the following:

```bash
 cmake -G Ninja ../llvm \
    -DLLVM_ENABLE_PROJECTS="mlir" \
    -DLLVM_TARGETS_TO_BUILD="host" \
    -DLLVM_ENABLE_ASSERTIONS=ON \
    -DCMAKE_BUILD_TYPE=DEBUG \
    -DCMAKE_EXPORT_COMPILE_COMMANDS=ON
```

lattner, 2021-07-02:
Nice catch.  Could you update the readme please?  This counts as "obvious" in my book :-)

fyquah, 2021-07-02:
The README has been updated to reflect this. The updated commands in the README has been tested on a mac M1, and it builds.

closed 2021-07-02 by fyquah