llvm/circt #1059: [firrtl] LowerTypes isn't lowering HwachaSequencer.fir correctly

lattner, 2021-05-16:
firtool -verilog on HwachaSequencer.fir from the firrtl regress suite fails because not all bundle types are getting lowered on ports in the HwachaSequencer module:

```
firtool HwachaSequencer.fir -o HwachaSequencer.fir.v -verilog -mlir-timing --ignore-fir-locators    HwachaSequencer.fir:2:3: error: cannot lower this port type to HW
  module HwachaSequencer : 
  ^
HwachaSequencer.fir:2:3: note: see current operation: "firrtl.module"() ( {
^bb0(%clock: !firrtl.clock, %reset: !firrtl.uint<1>, %io: !firrtl.bundle<op: flip<bundle<ready: flip<uint<1>>, valid: uint<1>, bits: bundle<fn: bundle<union: uint<10>>, sreg: bundle<ss1: uint<64>, ss2: uint<64>, ss3: uint<64>>, base: bundle<vp: bundle<id: uint<4>, valid: uint<1>, scalar: uint<1>, pred: uint<1>>, vs1: bundle<id: uint<8>, valid: uint<1>, scalar: uint<1>, pred: uint<1>, prec: uint<2>>, vs2: bundle<id: uint<8>, valid: uint<1>, scalar: uint<1>, pred: uint<1>, prec: uint<2>>, vs3: bundle<id: uint<8>, valid: uint<1>, scalar: uint<1>, pred: uint<1>, prec: uint<2>>, vd: bundle<id: uint<8>, valid: uint<1>, scalar: uint<1>, pred: uint<1>, prec: uint<2>>>, reg: bundle<vp: bundle<id: uint<8>>, vs1: bundle<id: uint<8>>, vs2: bundle<id: uint<8>>, vs3: bundle<id: uint<8>>, vd: bundle<id: uint<8>>>, active: bundle<vint: uint<1>, vipred: uint<1>, vimul: uint<1>, vidiv: uint<1>, vfma: uint<1>, vfdiv: uint<1>, vfcmp: uint<1>, vfconv: uint<1>, vrpred: uint<1>, vrfirst: uint<1>, vamo: uint<1>, vldx: uint<1>, vstx: uint<1>, vld: uint<1>, vst: uint<1>>>>>, master: bundle<state: bundle<valid: vector<uint<1>, 8>, e: vector<bundle<fn: bundle<union: uint<10>>, sreg: bundle<ss1: uint<64>, ss2: uint<64>, ss3: uint<64>>, base: bundle<vp: bundle<id: uint<4>, valid: uint<1>, scalar: uint<1>, pred: uint<1>>, vs1: bundle<id: uint<8>, valid: uint<1>, scalar: uint<1>, pred: uint<1>, prec: uint<2>>, vs2: bundle<id: uint<8>, valid: uint<1>, scalar: uint<1>, pred: uint<1>, prec: uint<2>>, vs3: bundle<id: uint<8>, valid: uint<1>, scalar: uint<1>, pred: uint<1>, prec: uint<2>>, vd: bundle<id: uint<8>, valid: uint<1>, scalar: uint<1>, pred: uint<1>, prec: uint<2>>>, rate: uint<1>, active: bundle<viu: uint<1>, vipu: uint<1>, vimu: uint<1>, vidu: uint<1>, vfmu: uint<1>, vfdu: uint<1>, vfcu: uint<1>, vfvu: uint<1>, vrpu: uint<1>, vrfu: uint<1>, vpu: uint<1>, vgu: uint<1>, vcu: uint<1>, vlu: uint<1>, vsu: uint<1>, vqu: uint<1>>, raw: vector<uint<1>, 8>, war: vector<uint<1>, 8>, waw: vector<uint<1>, 8>, last: uint<1>, rports: uint<2>, wport: bundle<sram: uint<4>, pred: uint<3>>>, 8>, head: uint<3>>, update: bundle<valid: vector<uint<1>, 8>, reg: vector<bundle<vp: bundle<id: uint<8>>, vs1: bundle<id: uint<8>>, vs2: bundle<id: uint<8>>, vs3: bundle<id: uint<8>>, vd: bundle<id: uint<8>>>, 8>>, clear: flip<vector<uint<1>, 8>>>, pending: bundle<mem: uint<1>, all: uint<1>>, vf: bundle<stop: flip<uint<1>>, last: uint<1>>, counters: bundle<memoryUOps: uint<3>, arithUOps: uint<3>, predUOps: uint<3>>, debug: bundle<head: uint<3>, tail: uint<3>, maybe_full: uint<1>, empty: uint<4>>>):
```

lattner, 2021-05-16:
Actually, this was because I wasn't running lower types!  When enabling it I get a crash:

```
$ firtool regress/HwachaSequencer.fir -o HwachaSequencer.fir.v -verilog -mlir-timing --ignore-fir-locators   -lower-types        
Assertion failed: (entry && "bundle lowering was not set"), function getBundleLowering, file ../lib/Dialect/FIRRTL/Transforms/LowerTypes.cpp, line 1077.
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.	Program arguments: firtool regress/HwachaSequencer.fir -o HwachaSequencer.fir.v -verilog -mlir-timing --ignore-fir-locators -lower-types
1.	2.	In-Flight Diagnostics:
    loc("regress/HwachaSequencer.fir":2155:11): error: SubaccessOp not handled.
    loc("regress/HwachaSequencer.fir":2156:11): error: SubaccessOp not handled.
Stack dump without symbol names (ensure you have llvm-symbolizer in your PATH or set the environment var `LLVM_SYMBOLIZER_PATH` to point to it):
...
8  firtool                  0x000000010ad377d0 (anonymous namespace)::TypeLoweringVisitor::getBundleLowering(mlir::Value, llvm::StringRef) + 208
9  firtool                  0x000000010ad37520 (anonymous namespace)::TypeLoweringVisitor::visitExpr(circt::firrtl::SubfieldOp) + 384
10 firtool                  0x000000010ad3738b void circt::firrtl::ExprVisitor<(anonymous namespace)::TypeLoweringVisitor, void>::dispatchExprVisitor(mlir::Operation*)::'lambda'(auto)::operator()<circt::firrtl::SubfieldOp>(auto) const + 43

```

seldridge, 2021-05-16:
Judging by the error message, I think this is a dupe of #656 where subaccess isn't currently handled. 

Things like this show up in that file: 
```
        v[tail] <= UInt<1>("h01") @[sequencer-master.scala 267:35]
```