llvm/circt #1060: [firrtl] Width inference doesn't support when's yet

lattner, 2021-05-16:
We cannot compile the regress/Ops.fir testcase in the firrtl regression suite:

```
$ firtool regress/Ops.fir -infer-widths -expand-whens -verilog
regress/Ops.fir:51:27: error: 'firrtl.when' op not supported in width inference
                          when eq(sel, UInt(5)):
                          ^
regress/Ops.fir:51:27: note: see current operation: "firrtl.when"(%5) ( {
  "firrtl.connect"(%ou, %c0_ui2) : (!firrtl.uint<13>, !firrtl.uint<2>) -> ()
},  {
  "firrtl.connect"(%ou, %c0_ui2) : (!firrtl.uint<13>, !firrtl.uint<2>) -> ()
}) : (!firrtl.uint<1>) -> ()
```

This may also be a phase ordering issue between infer-widths and expand-whens.

seldridge, 2021-05-16:
Good find. 

SFC does width inference before expand whens. If we keep that order (and I don't see an obvious reason to deviate), then we should beef up the MFC expand whens to handle this.

fabianschuiki, 2021-05-18:
Agreed that this should work with width inference. The whens shouldn't actually impact width inference at all, so it's likely just a missing no-op somewhere. I'll take a look.

fabianschuiki, 2021-05-18:
Fixed in #1075.

PR #1075 (2021-05-19): [FIRRTL] Ignore ops not affecting width inference
* Ignore operations that have no bearing on width inference. Until now these would generate an error.
* Have the `VarExpr` of the `InferWidths` pass directly track its constraint expression rather than using an indirection through a separate map. This simplifies things and rectifies the post-order traversal that will be necessary for graphs with cycles.

Fixes #1060.
commit 076a48ec (2021-05-19): [FIRRTL] Ignore ops not affecting width inference (#1075)

* Have the `VarExpr` of the `InferWidths` pass directly track its
  constraint expression rather than using an indirection through a
  separate map. This simplifies things and rectifies the post-order
  traversal that will be necessary for graphs with cycles.

* Ignore operations that have no bearing on width inference. Until now
  these would generate an error. Fixes #1060.
closed 2021-05-19 by fabianschuiki