llvm/circt #1071: [LLHD] Buffer over-read during simulation

youngar, 2021-05-18:
In `LLHD/Engine.cpp`, a memory bytebuffer is converted to an APInt.  In order to do this, the buffer is converted from `uint8_t*` to `uint64_t*`, and the size of the buffer is rounded up to the next 64 bits.  After this, the constructor if APInt reads past the end of the original bytebuffer.

```c++
      APInt buff(
          curr.size * 8,
          llvm::makeArrayRef(reinterpret_cast<uint64_t *>(curr.value.get()),
                             llvm::divideCeil(curr.size, 8)));
```

There seems to be a purpose built function to do what we want, although I haven't checked that about how the memory buffer is supposed to be used on big endian systems.
```c++
      APInt buff(curr.size * 8, 0);
      LoadIntFromMemory(buff, curr.value.get(), curr.size);
```

There are at least two places that will have to be fixed:
https://github.com/llvm/circt/blob/main/lib/Dialect/LLHD/Simulator/Engine.cpp#L131
https://github.com/llvm/circt/blob/main/lib/Dialect/LLHD/Simulator/State.cpp#L113

fabianschuiki, 2021-05-18:
That's an excellent catch, thanks a lot for digging this up. I think the `APInt` usage in the LLHD simulation engine should generally be replaced by dedicated JIT-compiled implementations of the operations. There shouldn't be a need for arbitrary-precision integer operations in this fashion at runtime that couldn't be mapped to actual LLVM IR.

mikeurbach, 2022-10-11:
I just noticed this while debugging #4081, and ASAN will catch this.

fabianschuiki, 2022-10-11:
@maerhart could we just round up the allocations in llhd-sim to ensure they always cover an integer multiple of 64 bits?

maerhart, 2022-10-14:
Yes, rounding up to a consistent multiple of bits would make a lot of sense. Currently the implementation is hard to understand because sometimes the number of bits is passed as an argument value and sometimes the number of bytes. Sometimes an llvm alloca is inserted with an already rounded up to 8 bit type and sometimes the type is used directly no matter how many bits it has.
Always rounding up to 64 bits would be a valid strategy, however, I think we can also easily get correctness by guaranteeing a multiple of 8 bits. Rounding up to bigger multiples of 32 or 64 bits could then be considered as performance optimizations.

closed 2022-10-14 by mikeurbach