llvm/circt #1088: [FIRRTL] infer constant width

drom, 2021-05-19:
The following FIRRTL program:

```python
circuit top_mod :
  module top_mod :
    output tmp166: SInt<4>
    tmp166 <= asSInt(UInt(9))
```

Compiled with `firtool a_top_mod.fir --lower-to-hw --lower-types --expand-whens --infer-widths --lowering-options=noAlwaysFF --verilog -o=a_top_mod_new.v` fails with the following error:

```
a_top_mod.fir:5:12: error: destination width 4 is not greater than or equal to source width 5
    tmp166 <= asSInt(UInt(9))
           ^
a_top_mod.fir:5:12: note: see current operation: "firrtl.connect"(%tmp166, %c9_si5) : (!firrtl.sint<4>, !firrtl.sint<5>) -> ()
```

Compiled with `firrtl-1.5-SNAPSHOT` produces this Verilog:

```verilog
module top_mod(
  output [3:0] tmp166
);
  assign tmp166 = -4'sh7;
endmodule
```

fabianschuiki, 2021-05-19:
Seems like the literal is overflowing the bit width in SFC, and MFC rejects the overflow.

drom, 2021-05-19:
Is `asSInt` a casting operation?

lattner, 2021-05-19:
I'm not sure if this is related, but the APSInt in firrtl.constant may be "wider than necessary" when there is an unknown width.  Using the existing width directly probably isn't tight enough of a bound

drom, 2021-05-19:
Maybe related:

The following FIRRTL program:

```python
circuit top_mod :
  module top_mod :
    output tmp44: SInt<5>
    tmp44 <= asSInt(SInt("o13"))
```

Compiled with `firtool a_top_mod.fir --lower-to-hw --lower-types --expand-whens --infer-widths --lowering-options=noAlwaysFF --verilog -o=a_top_mod_new.v` fails with the following error:

```
a_top_mod.fir:4:11: error: destination width 5 is not greater than or equal to source width 6
    tmp44 <= asSInt(SInt("o13"))
          ^
a_top_mod.fir:4:11: note: see current operation: "firrtl.connect"(%tmp44, %c11_si6) : (!firrtl.sint<5>, !firrtl.sint<6>) -> ()
```

Compiled with `firrtl-1.5-SNAPSHOT` produces this Verilog:

```verilog
module top_mod(
  output [4:0] tmp44
);
  assign tmp44 = 5'shb;
endmodule
```

fabianschuiki, 2021-05-20:
> @fabianschuiki Seems like the literal is overflowing the bit width in SFC, and MFC rejects the overflow.

I misremembered the semantics fo `asSInt`. That merely reinterprets the bits as signed, which for `9` produces a negative number in this context. All well in SFC land :smiley:

> @lattner I'm not sure if this is related, but the APSInt in firrtl.constant may be "wider than necessary" when there is an unknown width. Using the existing width directly probably isn't tight enough of a bound

I think this is exactly the problem. I'm using APInt's active bits here, which I mistook for something like a `ceil∘log2` mechanism. Will dig into the API there again to see what is available that gives the position of the MSB in an APInt.

fabianschuiki, 2021-05-20:
As @youngar points out, I was too dumb to actually call `getActiveBits` on the `APInt`, and instead used `getBitWidth`... :see_no_evil:

fabianschuiki, 2021-05-20:
The problem is the implicit truncation semantics in FIRRTL if wires have a known width. We handle this in the FIR parser by inserting `firrtl.bits` to truncate the value. In the absence of a known width (or if the RHS is just a constant), no such truncation is inserted, which then blows up if width inference ends up expanding the size. Seems like width inference should actually also add `firrtl.bits` as appropriate for connects.

There's also some weird behavior where connecting a too-wide constant to an output generates a `firrtl.partialconnect`. I'll open a separate issue to track and discuss this.

PR #1102 (2021-05-20): [FIRRTL] Fix constant width inference and connection truncation
* Width inference can cause the RHS of a connect operation to become wider than the LHS. The behaviour prescribed by the Scala FIRRTL compiler is to insert a `bits` operation to truncate the RHS to the appropriate width. Add this behaviour to the InferWidths pass. Fixes #1088, fixes #1089, fixes #1091, fixes #1092, fixes #1093, fixes #1094, fixes #1095, fixes #1096, fixes #1097, fixes #1099.
* Fix an issue where unsized constants would infer a width that could be wider than the minimum number of bits necessary to represent the con [... cut]
commit 762ce0af (2021-05-20): [FIRRTL] Fix constant width inference and connection truncation (#1102)

* Fix an issue where unsized constants would infer a width that could be
  wider than the minimum number of bits necessary to represent the
  constant. This fixes width inference to properly compute the number of
  bits necessary to represent a signed/unsigned constant. Also updates
  the APInt store in ConstantOp after width inference. Fixes #1090,
  fixes #1098.

* Width inference can cause the RHS of a connect operation to become
  wider than the LHS. The behaviour prescribed by the Scala FIRRTL
  compiler is to insert [... cut]
closed 2021-05-20 by fabianschuiki