llvm/circt #1108: [FIRRTL] infer width (mux)

drom, 2021-05-20:
The following FIRRTL program:

```python
circuit top_mod :
  module top_mod :
    input sel: UInt
    output tmp116: UInt<8>
    tmp116 <= mux(sel, UInt<8>(24), UInt<8>(42))
```

Compiled with `firtool a_top_mod.fir --lower-to-hw --expand-whens --infer-widths --lowering-options=noAlwaysFF --verilog -o=a_top_mod_new.v` fails with the following error:

```
a_top_mod.fir:5:15: error: 'firrtl.mux' op operand #0 must be UInt<1>, but got '!firrtl.uint'
    tmp116 <= mux(sel, UInt<8>(24), UInt<8>(42))
              ^
a_top_mod.fir:5:15: note: see current operation: %0 = "firrtl.mux"(%sel, %c24_ui8, %c42_ui8) : (!firrtl.uint, !firrtl.uint<8>, !firrtl.uint<8>) -> !firrtl.uint<8>
```

Compiled with `firrtl-1.5-SNAPSHOT` produces this Verilog:

```verilog
module top_mod(
  input        sel,
  output [7:0] tmp116
);
  assign tmp116 = sel ? 8'h18 : 8'h2a;
endmodule
```

seldridge, 2021-05-20:
This restriction is coming from the ODS constraints on `UInt1Type`. I think it's safe to relax that to either be either 1-bit or unknown width.

lattner, 2021-05-20:
Yes that makes sense

seldridge, 2021-05-21:
I closed this with `fb94acd` just now. However, this circuit actually crashes downstream in `InferWidths`. It's counterintuitive to me that this actually gets inferred correctly, but it looks like the SFC sets a constraint based on a signal being the mux line of a select.

youngar, 2021-05-21:
Good call, looks like[ it does](https://github.com/chipsalliance/firrtl/blob/master/src/main/scala/firrtl/passes/InferWidths.scala#L119).  Should we reopen this or open a new issue?

seldridge, 2021-05-21:
I've got a patch ready to go. Give me 2 minutes.

seldridge, 2021-05-21:
With these two patches in place, I'm now getting output that matches the SFC:

```verilog
module top_mod(
  input        sel,
  output [7:0] tmp116);

  assign tmp116 = sel ? 8'h18 : 8'h2A;  // 1108/top_mod.fir:2:3, :5:{15,24,37}
endmodule
```

drom, 2021-05-21:
I am getting WAY MORE core dumps after this fix.

drom, 2021-05-21:
```
circuit top_mod :
  module top_mod :
    input inp_h: SInt<58>
    input inp_n: SInt<67>
    output tmp32: SInt<67>
    tmp32 <= mux(tail(asUInt(inp_h), 57), inp_n, SInt<18>("o-327727"))
```

gives this crash dump

```
==18542== Invalid read of size 8
==18542==    at 0x5A70FE: (anonymous namespace)::ConstraintSolver::addGeqConstraint((anonymous namespace)::VarExpr*, (anonymous namespace)::Expr*) (InferWidths.cpp:339)
==18542==    by 0x5AB022: _ZZN12_GLOBAL__N_116InferenceMapping12mapOperationEPN4mlir9OperationEENKUlT_E19_clIN5circt6firrtl9MuxPrimOpEEEDaS4_ (InferWidths.cpp:642)
==18542==    by 0x5AB116: llvm::TypeSwitch<mlir::Operation*, void>& llvm::TypeSwitch<mlir::Operation*, void>::Case<circt::firrtl::MuxPrimOp, (anonymous namespace)::InferenceMapping::mapOperation(mlir::Operation*)::{lambda(auto:1)#21}>((anonymous namespace)::InferenceMapping::mapOperation(mlir::Operation*)::{lambda(auto:1)#21}&&) (TypeSwitch.h:164)
==18542==    by 0x5A7BCF: (anonymous namespace)::InferenceMapping::mapOperation(mlir::Operation*) (InferWidths.cpp:647)
==18542==    by 0x5A782F: (anonymous namespace)::InferenceMapping::map(circt::firrtl::FModuleOp)::{lambda(mlir::Operation*)#1}::operator()(mlir::Operation*) const (InferWidths.cpp:502)
==18542==    by 0x5B6FB4: mlir::WalkResult llvm::function_ref<mlir::WalkResult (mlir::Operation*)>::callback_fn<(anonymous namespace)::InferenceMapping::map(circt::firrtl::FModuleOp)::{lambda(mlir::Operation*)#1}>(long, mlir::Operation*) (STLExtras.h:177)
==18542==    by 0x5B33AF: std::enable_if<llvm::disjunction<std::is_same<mlir::Operation*, mlir::Operation*>, llvm::disjunction<std::is_same, mlir::Region*>, llvm::disjunction<std::is_same, mlir::Block*> >::value, mlir::WalkResult>::type mlir::detail::walk<(mlir::WalkOrder)1, (anonymous namespace)::InferenceMapping::map(circt::firrtl::FModuleOp)::{lambda(mlir::Operation*)#1}&, mlir::Operation*, mlir::WalkResult>(mlir::Operation*, (anonymous namespace)::InferenceMapping::map(circt::firrtl::FModuleOp)::{lambda(mlir::Operation*)#1}&) (Visitors.h:134)
==18542==    by 0x5B06EE: std::enable_if<std::is_same<mlir::WalkResult, (anonymous namespace)::InferenceMapping::map(circt::firrtl::FModuleOp)::{lambda(mlir::Operation*)#1}&>::value, std::is_same>::type mlir::Block::walk<(mlir::WalkOrder)1, (anonymous namespace)::InferenceMapping::map(circt::firrtl::FModuleOp)::{lambda(mlir::Operation*)#1}&, mlir::WalkResult>(llvm::ilist_iterator<std::enable_if<std::is_same<mlir::WalkResult, (anonymous namespace)::InferenceMapping::map(circt::firrtl::FModuleOp)::{lambda(mlir::Operation*)#1}&>::value, std::is_same>::type::ilist_detail::node_options<mlir::Operation, false, false, void>, false, false>, std::enable_if<std::is_same<mlir::WalkResult, (anonymous namespace)::InferenceMapping::map(circt::firrtl::FModuleOp)::{lambda(mlir::Operation*)#1}&>::value, std::is_same>::type::ilist_detail::node_options<mlir::Operation, false, false, void>, (anonymous namespace)::InferenceMapping::map(circt::firrtl::FModuleOp)::{lambda(mlir::Operation*)#1}&) (Block.h:312)
==18542==    by 0x5AD575: mlir::WalkResult mlir::Block::walk<(mlir::WalkOrder)1, (anonymous namespace)::InferenceMapping::map(circt::firrtl::FModuleOp)::{lambda(mlir::Operation*)#1}&, mlir::WalkResult>((anonymous namespace)::InferenceMapping::map(circt::firrtl::FModuleOp)::{lambda(mlir::Operation*)#1}&) (Block.h:275)
==18542==    by 0x5A9DEB: std::enable_if<std::is_same<mlir::WalkResult, (anonymous namespace)::InferenceMapping::map(circt::firrtl::FModuleOp)::{lambda(mlir::Operation*)#1}>::value, std::is_same>::type mlir::Region::walk<(mlir::WalkOrder)1, (anonymous namespace)::InferenceMapping::map(circt::firrtl::FModuleOp)::{lambda(mlir::Operation*)#1}, mlir::WalkResult>((anonymous namespace)::InferenceMapping::map(circt::firrtl::FModuleOp)::{lambda(mlir::Operation*)#1}&&) (Region.h:282)
==18542==    by 0x5A78E6: (anonymous namespace)::InferenceMapping::map(circt::firrtl::FModuleOp) (InferWidths.cpp:502)
==18542==    by 0x5A77D7: (anonymous namespace)::InferenceMapping::map(circt::firrtl::CircuitOp) (InferWidths.cpp:488)
==18542==  Address 0x10 is not stack'd, malloc'd or (recently) free'd
==18542== 
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.      Program arguments: ../../llvm/circt/build/bin/firtool a_top_mod.fir --lower-to-hw --expand-whens --infer-widths --lowering-options=noAlwaysFF --mlir-timing --verilog -o=a_top_mod_new.v
1.       #0 0x000000000048512d llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (../../llvm/circt/build/bin/firtool+0x48512d)
 #1 0x0000000000482f64 llvm::sys::RunSignalHandlers() (../../llvm/circt/build/bin/firtool+0x482f64)
 #2 0x00000000004830c3 SignalHandler(int) (../../llvm/circt/build/bin/firtool+0x4830c3)
 #3 0x0000000004890a30 __restore_rt (/lib64/libpthread.so.0+0x13a30)
 #4 0x00000000005a70fe (anonymous namespace)::ConstraintSolver::addGeqConstraint((anonymous namespace)::VarExpr*, (anonymous namespace)::Expr*) /home/drom/work/github/llvm/circt/build/../lib/Dialect/FIRRTL/Transforms/InferWidths.cpp:339:28
 #5 0x00000000005ab023 auto (anonymous namespace)::InferenceMapping::mapOperation(mlir::Operation*)::'lambda19'(auto)::operator()<circt::firrtl::MuxPrimOp>(auto) const /home/drom/work/github/llvm/circt/build/../lib/Dialect/FIRRTL/Transforms/InferWidths.cpp:641:36
[... cut ...]
==18542==  If you believe this happened as a result of a stack
==18542==  overflow in your program's main thread (unlikely but
==18542==  possible), you can try to increase the size of the
==18542==  main thread stack using the --main-stacksize= flag.
==18542==  The main thread stack size used in this run was 8388608.
./process1.sh: line 16: 18542 Segmentation fault      (core dumped) valgrind -q ../../llvm/circt/build/bin/firtool a_top_mod.fir --lower-to-hw --expand-whens --infer-widths --lowering-options=noAlwaysFF --mlir-timing --verilog -o=$VFILE2
```

seldridge, 2021-05-21:
Should be fixed now. I really botched that one.

Sorry about that.

:clap: JUST :clap: USE :clap: THE :clap: EXISTING :clap: APIS :clap: @seldridge :clap: :clap: :clap:

:clap: DON'T :clap: BLIND :clap: CAST :clap: STUFF :clap: :clap: :clap:

commit fb94acd3 (2021-05-21): [FIRRTL] Allow UnknownWidths in UInt1Type

Relax the predicate on UInt1Type to also allow an unknown width UInt (in
addition to the existing, alternative requirement of a 1-bit UInt).
This addresses a false positive where a Mux can allowably have an
unknown width select parameter.

Fix #1108.
commit 6be5f63c (2021-05-21): [FIRRTL] Fix Regression on Mux w/ Known Width

Fix a regression introduced in 20a1d111 for width inference involving
the Mux select line.  20a1d111 was yolo-ing a cast causing width
inference to crash if it ran on a Mux with a select line that had a
known width.  tl;dr: I should learn how to use the APIs...

Fix #1108 (again).
closed 2021-05-21 by seldridge