llvm/circt #1109: [FIRRTL] infer width (io) (core dump)

drom, 2021-05-20:
The following FIRRTL program:

```python
circuit top_mod :
  module top_mod :
    input inp_a: UInt
    output _tmp46: UInt
    _tmp46 <= inp_a
```

Compiled with `firtool a_top_mod.fir --lower-to-hw --expand-whens --infer-widths --lowering-options=noAlwaysFF --verilog -o=a_top_mod_new.v` fails with the following core dump:

```
0.      Program arguments: ../../llvm/circt/build/bin/firtool a_top_mod.fir --lower-to-hw --expand-whens --infer-widths --lowering-options=noAlwaysFF --mlir-timing --verilog -o=a_top_mod_new.v
1.       #0 0x000000000048512d llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (../../llvm/circt/build/bin/firtool+0x48512d)
 #1 0x0000000000482f64 llvm::sys::RunSignalHandlers() (../../llvm/circt/build/bin/firtool+0x482f64)
 #2 0x00000000004830c3 SignalHandler(int) (../../llvm/circt/build/bin/firtool+0x4830c3)
 #3 0x0000000004890a30 __restore_rt (/lib64/libpthread.so.0+0x13a30)
 #4 0x0000000004cb24a5 raise (/lib64/libc.so.6+0x3d4a5)
 #5 0x0000000004c9b864 abort (/lib64/libc.so.6+0x26864)
 #6 0x0000000004c9b749 _nl_load_domain.cold (/lib64/libc.so.6+0x26749)
 #7 0x0000000004caa9b6 (/lib64/libc.so.6+0x359b6)
 #8 0x00000000005b7305 llvm::isa_impl_cl<(anonymous namespace)::RootExpr, (anonymous namespace)::Expr const*>::doit((anonymous namespace)::Expr const*) /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/Support/Casting.h:105:36
 #9 0x00000000005b59c2 llvm::isa_impl_wrap<(anonymous namespace)::RootExpr, (anonymous namespace)::Expr const*, (anonymous namespace)::Expr const*>::doit((anonymous namespace)::Expr const* const&) /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/Support/Casting.h:132:3
#10 0x00000000005b3c0f llvm::isa_impl_wrap<(anonymous namespace)::RootExpr, (anonymous namespace)::Expr const* const, (anonymous namespace)::Expr const*>::doit((anonymous namespace)::Expr const* const&) /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/Support/Casting.h:124:3
#11 0x00000000005b126b bool llvm::isa<(anonymous namespace)::RootExpr, (anonymous namespace)::Expr const*>((anonymous namespace)::Expr const* const&) /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/Support/Casting.h:144:1
#12 0x00000000005ae5b4 llvm::cast_retty<(anonymous namespace)::RootExpr, (anonymous namespace)::Expr const*>::ret_type llvm::dyn_cast<(anonymous namespace)::RootExpr, (anonymous namespace)::Expr const>((anonymous namespace)::Expr const*) /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/Support/Casting.h:345:22
#13 0x00000000005abb3b auto llvm::detail::TypeSwitchBase<llvm::TypeSwitch<(anonymous namespace)::Expr const*, (anonymous namespace)::Expr* const*>, (anonymous namespace)::Expr const*>::castValue<(anonymous namespace)::RootExpr, (anonymous namespace)::Expr const*>((anonymous namespace)::Expr const*, std::enable_if<!(llvm::detail::detector<void, llvm::detail::TypeSwitchBase<llvm::TypeSwitch<(anonymous namespace)::Expr const*, (anonymous namespace)::Expr* const*>, (anonymous namespace)::Expr const*>::has_dyn_cast_t, (anonymous namespace)::Expr const*, (anonymous namespace)::RootExpr>::value_t::value), void>::type*) /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/ADT/TypeSwitch.h:82:3
#14 0x00000000005abb9a llvm::TypeSwitch<(anonymous namespace)::Expr const*, (anonymous namespace)::Expr* const*>& llvm::TypeSwitch<(anonymous namespace)::Expr const*, (anonymous namespace)::Expr* const*>::Case<(anonymous namespace)::RootExpr, (anonymous namespace)::Expr::begin() const::'lambda'(auto*)&>((anonymous namespace)::Expr::begin() const::'lambda'(auto*)&) /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/ADT/TypeSwitch.h:115:58
#15 0x00000000005a8efb llvm::TypeSwitch<(anonymous namespace)::Expr const*, (anonymous namespace)::Expr* const*>& llvm::detail::TypeSwitchBase<llvm::TypeSwitch<(anonymous namespace)::Expr const*, (anonymous namespace)::Expr* const*>, (anonymous namespace)::Expr const*>::Case<(anonymous namespace)::RootExpr, (anonymous namespace)::VarExpr, (anonymous namespace)::KnownExpr, (anonymous namespace)::AddExpr, (anonymous namespace)::PowExpr, (anonymous namespace)::MaxExpr, (anonymous namespace)::MinExpr, (anonymous namespace)::Expr::begin() const::'lambda'(auto*)>((anonymous namespace)::Expr::begin() const::'lambda'(auto*)&&) /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/ADT/TypeSwitch.h:41:42
#16 0x00000000005a6df6 (anonymous namespace)::Expr::begin() const /home/drom/work/github/llvm/circt/build/../lib/Dialect/FIRRTL/Transforms/InferWidths.cpp:224:42
#17 0x00000000005a6e6a llvm::GraphTraits<(anonymous namespace)::Expr*>::child_begin((anonymous namespace)::Expr*) /home/drom/work/github/llvm/circt/build/../lib/Dialect/FIRRTL/Transforms/InferWidths.cpp:247:3
#18 0x00000000005ad092 llvm::po_iterator<(anonymous namespace)::Expr*, llvm::SmallPtrSet<(anonymous namespace)::Expr*, 8u>, false, llvm::GraphTraits<(anonymous namespace)::Expr*> >::traverseChild() /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/ADT/PostOrderIterator.h:136:64
#19 0x00000000005b28d7 llvm::po_iterator<(anonymous namespace)::Expr*, llvm::SmallPtrSet<(anonymous namespace)::Expr*, 8u>, false, llvm::GraphTraits<(anonymous namespace)::Expr*> >::po_iterator((anonymous namespace)::Expr*) /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/ADT/PostOrderIterator.h:115:3
#20 0x00000000005af808 llvm::po_iterator<(anonymous namespace)::Expr*, llvm::SmallPtrSet<(anonymous namespace)::Expr*, 8u>, false, llvm::GraphTraits<(anonymous namespace)::Expr*> >::begin((anonymous namespace)::Expr* const&) /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/ADT/PostOrderIterator.h:145:3
#21 0x00000000005aca1f llvm::po_iterator<(anonymous namespace)::Expr*, llvm::SmallPtrSet<llvm::GraphTraits<(anonymous namespace)::Expr*>::NodeRef, 8u>, false, llvm::GraphTraits<(anonymous namespace)::Expr*> > llvm::po_begin<(anonymous namespace)::Expr*>((anonymous namespace)::Expr* const&) /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/ADT/PostOrderIterator.h:184:72
#22 0x00000000005a95ca llvm::iterator_range<llvm::po_iterator<(anonymous namespace)::Expr*, llvm::SmallPtrSet<llvm::GraphTraits<(anonymous namespace)::Expr*>::NodeRef, 8u>, false, llvm::GraphTraits<(anonymous namespace)::Expr*> > > llvm::post_order<(anonymous namespace)::Expr*>((anonymous namespace)::Expr* const&) /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/ADT/PostOrderIterator.h:189:43
#23 0x00000000005a74f2 (anonymous namespace)::ConstraintSolver::solve() /home/drom/work/github/llvm/circt/build/../lib/Dialect/FIRRTL/Transforms/InferWidths.cpp:410:64
#24 0x00000000005a8c4c (anonymous namespace)::InferWidthsPass::runOnOperation() /home/drom/work/github/llvm/circt/build/../lib/Dialect/FIRRTL/Transforms/InferWidths.cpp:936:3
#25 0x000000000090e2f5 mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (../../llvm/circt/build/bin/firtool+0x90e2f5)
#26 0x000000000090e450 mlir::detail::OpToOpPassAdaptor::runPipeline(llvm::iterator_range<llvm::pointee_iterator<std::unique_ptr<mlir::Pass, std::default_delete<mlir::Pass> >*, mlir::Pass> >, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) (../../llvm/circt/build/bin/firtool+0x90e450)
#27 0x000000000090db14 mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool) (../../llvm/circt/build/bin/firtool+0x90db14)
#28 0x000000000090e0b6 mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (../../llvm/circt/build/bin/firtool+0x90e0b6)
#29 0x000000000090e450 mlir::detail::OpToOpPassAdaptor::runPipeline(llvm::iterator_range<llvm::pointee_iterator<std::unique_ptr<mlir::Pass, std::default_delete<mlir::Pass> >*, mlir::Pass> >, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) (../../llvm/circt/build/bin/firtool+0x90e450)
#30 0x0000000000911517 mlir::PassManager::run(mlir::Operation*) (../../llvm/circt/build/bin/firtool+0x911517)
#31 0x000000000040c785 processBuffer(std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer> >, llvm::StringRef, mlir::TimingScope&, mlir::MLIRContext&, std::function<mlir::LogicalResult (mlir::OwningModuleRef)>) /home/drom/work/github/llvm/circt/build/../t [... cut ...]
#33 0x000000000040cffc main /home/drom/work/github/llvm/circt/build/../tools/firtool/firtool.cpp:352:39
#34 0x0000000004c9cb25 __libc_start_main (/lib64/libc.so.6+0x27b25)
#35 0x000000000040bd7e _start /home/abuild/rpmbuild/BUILD/glibc-2.33/csu/../sysdeps/x86_64/start.S:122:0
./process1.sh: line 16: 11134 Aborted                 (core dumped) valgrind -q ../../llvm/circt/build/bin/firtool a_top_mod.fir --lower-to-hw --expand-whens --infer-widths --lowering-options=noAlwaysFF --mlir-timing --verilog -o=$VFILE2
f```

seldridge, 2021-05-21:
This issue seems to occur for any circuit which has no constraints. E.g., you can produce the same error with two wires:

fabianschuiki, 2021-05-21:
This was a dumb oversight on my side where an unconstrained variable would do an out-of-bounds memory access. Why aren't we using Rust :crab: again :wink: ?

closed 2021-05-21 by fabianschuiki

lattner, 2021-05-21:
C++, my old nemesis

commit 6159f82a (2021-05-21): [FIRRTL] Fix width inference in the presence of unbound vars

* Fix an issue where unbound variables in width inference could cause
  a segfault. Fixes #1109.