llvm/circt #1115: [FIRRTL] infer width (shr) (core dump)

drom, 2021-05-21:
The following FIRRTL program:

```python
circuit top_mod :
  module top_mod :
    output tmp59: SInt<2>
    tmp59 <= shr(SInt("h5cbca"), 21)
```

Compiled with `firtool a_top_mod.fir --lower-to-hw --expand-whens --infer-widths --lowering-options=noAlwaysFF --verilog -o=a_top_mod_new.v` fails with the following core dump:

```
firtool: ../llvm/llvm/include/llvm/ADT/APInt.h:995: void llvm::APInt::lshrInPlace(unsigned int): Assertion `ShiftAmt <= BitWidth && "Invalid shift amount"' failed.
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.      Program arguments: ../../llvm/circt/build/bin/firtool a_top_mod.fir --lower-to-hw --expand-whens --infer-widths --lowering-options=noAlwaysFF --mlir-timing --verilog -o=a_top_mod_new.v
1.       #0 0x000000000048512d llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (../../llvm/circt/build/bin/firtool+0x48512d)
 #1 0x0000000000482f64 llvm::sys::RunSignalHandlers() (../../llvm/circt/build/bin/firtool+0x482f64)
 #2 0x00000000004830c3 SignalHandler(int) (../../llvm/circt/build/bin/firtool+0x4830c3)
 #3 0x0000000004890a30 __restore_rt (/lib64/libpthread.so.0+0x13a30)
 #4 0x0000000004cb24a5 raise (/lib64/libc.so.6+0x3d4a5)
 #5 0x0000000004c9b864 abort (/lib64/libc.so.6+0x26864)
 #6 0x0000000004c9b749 _nl_load_domain.cold (/lib64/libc.so.6+0x26749)
 #7 0x0000000004caa9b6 (/lib64/libc.so.6+0x359b6)
 #8 0x000000000071c778 llvm::APInt::lshrInPlace(unsigned int) /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/ADT/APInt.h:996:21
 #9 0x000000000071c737 llvm::APInt::lshr(unsigned int) const /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/ADT/APInt.h:990:12
#10 0x000000000080deaa circt::comb::ExtractOp::fold(llvm::ArrayRef<mlir::Attribute>) /home/drom/work/github/llvm/circt/build/../lib/Dialect/Comb/CombFolds.cpp:133:60
#11 0x00000000008097e1 mlir::LogicalResult mlir::Op<circt::comb::ExtractOp, mlir::OpTrait::ZeroRegion, mlir::OpTrait::OneResult, mlir::OpTrait::OneTypedResult<mlir::IntegerType>::Impl, mlir::OpTrait::ZeroSuccessor, mlir::OpTrait::OneOperand, mlir::MemoryEffectOpInterface::Trait>::foldSingleResultHook<circt::comb::ExtractOp>(mlir::Operation*, llvm::ArrayRef<mlir::Attribute>, llvm::SmallVectorImpl<mlir::OpFoldResult>&) /home/drom/work/github/llvm/circt/build/../llvm/mlir/include/mlir/IR/OpDefinition.h:1702:53
#12 0x00000000006a2a4b mlir::Operation::fold(llvm::ArrayRef<mlir::Attribute>, llvm::SmallVectorImpl<mlir::OpFoldResult>&) (../../llvm/circt/build/bin/firtool+0x6a2a4b)
#13 0x000000000064efca mlir::OpBuilder::tryFold(mlir::Operation*, llvm::SmallVectorImpl<mlir::Value>&) (../../llvm/circt/build/bin/firtool+0x64efca)
#14 0x0000000000575c64 void mlir::OpBuilder::createOrFold<circt::comb::ExtractOp, mlir::Type&, mlir::Value&, unsigned int&>(llvm::SmallVectorImpl<mlir::Value>&, mlir::Location, mlir::Type&, mlir::Value&, unsigned int&) /home/drom/work/github/llvm/circt/build/../llvm/mlir/include/mlir/IR/Builders.h:426:18
#15 0x000000000057045e std::enable_if<circt::comb::ExtractOp::hasTrait<mlir::OpTrait::OneResult>(), mlir::Value>::type mlir::OpBuilder::createOrFold<circt::comb::ExtractOp, mlir::Type&, mlir::Value&, unsigned int&>(mlir::Location, mlir::Type&, mlir::Value&, unsigned int&) /home/drom/work/github/llvm/circt/build/../llvm/mlir/include/mlir/IR/Builders.h:439:25
#16 0x0000000000569729 std::enable_if<circt::comb::ExtractOp::hasTrait<mlir::OpTrait::OneResult>(), mlir::Value>::type mlir::ImplicitLocOpBuilder::createOrFold<circt::comb::ExtractOp, mlir::Type&, mlir::Value&, unsigned int&>(mlir::Type&, mlir::Value&, unsigned int&) /home/drom/work/github/llvm/circt/build/../llvm/mlir/include/mlir/IR/ImplicitLocOpBuilder.h:84:3
#17 0x000000000054b58a mlir::LogicalResult (anonymous namespace)::FIRRTLLowering::setLoweringTo<circt::comb::ExtractOp, mlir::Type, mlir::Value, unsigned int>(mlir::Operation*, mlir::Type, mlir::Value, unsigned int) /home/drom/work/github/llvm/circt/build/../lib/Conversion/FIRRTLToHW/LowerToHW.cpp:1378:51
#18 0x0000000000547457 (anonymous namespace)::FIRRTLLowering::visitExpr(circt::firrtl::ShrPrimOp) /home/drom/work/github/llvm/circt/build/../lib/Conversion/FIRRTLToHW/LowerToHW.cpp:2273:1
#19 0x000000000055c13b mlir::LogicalResult circt::firrtl::ExprVisitor<(anonymous namespace)::FIRRTLLowering, mlir::LogicalResult>::dispatchExprVisitor(mlir::Operation*)::'lambda'(auto)::operator()<circt::firrtl::ShrPrimOp>(auto) const /home/drom/work/github/llvm/circt/build/../include/circt/Dialect/FIRRTL/FIRRTLVisitors.h:49:13
#20 0x000000000055c0b1 llvm::TypeSwitch<mlir::Operation*, mlir::LogicalResult>& llvm::TypeSwitch<mlir::Operation*, mlir::LogicalResult>::Case<circt::firrtl::ShrPrimOp, circt::firrtl::ExprVisitor<(anonymous namespace)::FIRRTLLowering, mlir::LogicalResult>::dispatchExprVisitor(mlir::Operation*)::'lambda'(auto)&>(mlir::LogicalResult&&) /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/ADT/TypeSwitch.h:116:22
#21 0x000000000055c00b llvm::TypeSwitch<mlir::Operation*, mlir::LogicalResult>& llvm::detail::TypeSwitchBase<llvm::TypeSwitch<mlir::Operation*, mlir::LogicalResult>, mlir::Operation*>::Case<circt::firrtl::ShrPrimOp, circt::firrtl::TailPrimOp, circt::firrtl::AsPassivePrimOp, circt::firrtl::AsNonPassivePrimOp, circt::firrtl::StdIntCastOp, circt::firrtl::HWStructCastOp, circt::firrtl::AnalogInOutCastOp, circt::firrtl::ExprVisitor<(anonymous namespace)::FIRRTLLowering, mlir::LogicalResult>::dispatchExprVisitor(mlir::Operation*)::'lambda'(auto)&>(circt::firrtl::ExprVisitor<(anonymous namespace)::FIRRTLLowering, mlir::LogicalResult>::dispatchExprVisitor(mlir::Operation*)::'lambda'(auto)&) /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/ADT/TypeSwitch.h:41:42
#22 0x000000000055bf2b llvm::TypeSwitch<mlir::Operation*, mlir::LogicalResult>& llvm::detail::TypeSwitchBase<llvm::TypeSwitch<mlir::Operation*, mlir::LogicalResult>, mlir::Operation*>::Case<circt::firrtl::ShlPrimOp, circt::firrtl::ShrPrimOp, circt::firrtl::TailPrimOp, circt::firrtl::AsPassivePrimOp, circt::firrtl::AsNonPassivePrimOp, circt::firrtl::StdIntCastOp, circt::firrtl::HWStructCastOp, circt::firrtl::AnalogInOutCastOp, circt::firrtl::ExprVisitor<(anonymous namespace)::FIRRTLLowering, mlir::LogicalResult>::dispatchExprVisitor(mlir::Operation*)::'lambda'(auto)&>(circt::firrtl::ExprVisitor<(anonymous namespace)::FIRRTLLowering, mlir::LogicalResult>::dispatchExprVisitor(mlir::Operation*)::'lambda'(auto)&) /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/ADT/TypeSwitch.h:42:3
#23 0x000000000055be39 llvm::TypeSwitch<mlir::Operation*, mlir::LogicalResult>& llvm::detail::TypeSwitchBase<llvm::TypeSwitch<mlir::Operation*, mlir::LogicalResult>, mlir::Operation*>::Case<circt::firrtl::PadPrimOp, circt::firrtl::ShlPrimOp, circt::firrtl::ShrPrimOp, circt::firrtl::TailPrimOp, circt::firrtl::AsPassivePrimOp, circt::firrtl::AsNonPassivePrimOp, circt::firrtl::StdIntCastOp, circt::firrtl::HWStructCastOp, circt::firrtl::AnalogInOutCastOp, circt::firrtl::ExprVisitor<(anonymous namespace)::FIRRTLLowering, mlir::LogicalResult>::dispatchExprVisitor(mlir::Operation*)::'lambda'(auto)&>(circt::firrtl::ExprVisitor<(anonymous namespace)::FIRRTLLowering, mlir::LogicalResult>::dispatchExprVisitor(mlir::Operation*)::'lambda'(auto)&) /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/ADT/TypeSwitch.h:42:3
#24 0x000000000055bd47 llvm::TypeSwitch<mlir::Operation*, mlir::LogicalResult>& llvm::detail::TypeSwitchBase<llvm::TypeSwitch<mlir::Operation*, mlir::LogicalResult>, mlir::Operation*>::Case<circt::firrtl::MuxPrimOp, circt::firrtl::PadPrimOp, circt::firrtl::ShlPrimOp, circt::firrtl::ShrPrimOp, circt::firrtl::TailPrimOp, circt::firrtl::AsPassivePrimOp, circt::firrtl::AsNonPassivePrimOp, circt::firrtl::StdIntCastOp, circt::firrtl::HWStructCastOp, circt::firrtl::AnalogInOutCastOp, circt::firrtl::ExprVisitor<(anonymous namespace)::FIRRTLLowering, mlir::LogicalResult>::dispatchExprVisitor(mlir::Operation*)::'lambda'(auto)&>(circt::firrtl::ExprVisitor<(anonymous namespace)::FIRRTLLowering, mlir::LogicalResult>::dispatchExprVisitor(mlir::Operation*)::'lambda'(auto)&) /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/ADT/TypeSwitch.h:42:3
#25 0x000000000055bc55 llvm::TypeSwitch<mlir::Operation*, mlir::LogicalResult>& llvm::detail::TypeSwitchBase<llvm::TypeSwitch<mlir::Operation*, mlir::LogicalResult>, mlir::Operation*>::Case<circt::firrtl::HeadPrimOp, circt::firrtl::MuxPrimOp, circt::f [... cut ...]
#75 0x0000000004c9cb25 __libc_start_main (/lib64/libc.so.6+0x27b25)
#76 0x000000000040bd7e _start /home/abuild/rpmbuild/BUILD/glibc-2.33/csu/../sysdeps/x86_64/start.S:122:0
./process1.sh: line 16: 24161 Aborted                 (core dumped) valgrind -q ../../llvm/circt/build/bin/firtool a_top_mod.fir --lower-to-hw --expand-whens --infer-widths --lowering-options=noAlwaysFF --mlir-timing --verilog -o=$VFILE2
```

Compiled with `firrtl-1.5-SNAPSHOT` produces this Verilog:

```verilog
module top_mod(
  output [1:0] tmp59
);
  assign tmp59 = 2'sh0;
endmodule
```

lattner, 2021-05-21:
Looks like a lower2hw issue, I'll take a quick look

lattner, 2021-05-21:
fixed in 4c2d81de

closed 2021-05-21 by lattner