llvm/circt #1122: [FIRRTL] (tail) false positive

drom, 2021-05-21:
The following FIRRTL program

```python
circuit top_mod :
  module top_mod :
    output tmp42: SInt<11>
    tmp42 <= validif(tail(UInt("h11f"), 13), SInt<11>(0))
```

Compiled with `firtool a_top_mod.fir --lower-to-hw --expand-whens --infer-widths --lowering-options=noAlwaysFF --verilog -o=a_top_mod_new.v` produces this Verilog:

```verilog
// Standard header to adapt well known macros to our needs.

module top_mod(
  output [10:0] tmp42);

  assign tmp42 = 11'h0;	// a_top_mod.fir:2:3, :4:46
endmodule
```

Compiled with `firrtl-1.5-SNAPSHOT` produces this Error message:

```
Exception in thread "main" firrtl.passes.CheckWidths$TailWidthException: : [target ~top_mod|top_mod] Parameter 13 in tail operator is larger than input width 9.
```

fabianschuiki, 2021-05-22:
This is probably something that should go into the verifier of `TailPrimOp`.

fabianschuiki, 2021-05-25:
This is fixed in #1146.