llvm/circt #1131: [FIRRTL] (orr) folder missing

drom, 2021-05-24:
Not a bug, but missing constant folder.

The following FIRRTL program

```python
circuit top_mod :
  module top_mod :
    output tmp62: UInt<1>
    wire tmp42: SInt
    tmp42 <= SInt<13>("o-2562")
    tmp62 <= orr(tmp42)
```

Compiled with `firtool` produces this Verilog:

```verilog
module top_mod(
  output tmp62);

  wire [12:0] tmp42;	// a_top_mod.fir:4:5

  assign tmp42 = 13'h1A8E;	// a_top_mod.fir:5:{11,14}
  assign tmp62 = |tmp42;	// a_top_mod.fir:2:3, :6:14
endmodule
```

Compiled with `firrtl-1.5-SNAPSHOT` produces this Verilog:

```verilog
module top_mod(
  output  tmp62
);
  assign tmp62 = 1'h1;
endmodule
```

seldridge, 2021-05-24:
What's actually going on here is that the folders are [gated on having widths known](https://github.com/llvm/circt/blob/68d0845326e35c9e1040ead52102348299c1ec3d/lib/Dialect/FIRRTL/FIRRTLFolds.cpp#L711). Canonicalization won't touch the uninferred widths (by inspecting `-print-ir-after-all` output). FIRRTL dialect Canonicalization runs before width inference and is not re-run after. 

The end result is that canonicalization appears to not happen.

The safest way to handle this would be to re-run canonicalization after width inference. However, there may be a cleaner way like beefing up canonicalization to better handle uninferred widths that can be trivially worked around.

lattner, 2021-05-24:
I'd prefer to move width-inference to before canonicalization

darthscsi, 2021-05-25:
Enabling infer-widths by default is blocked on supporting instances.  Here is a branch with a restructures firtool pipeline which indeed catches this: https://github.com/llvm/circt/tree/dev/darthscsi/default.inferwidth

fabianschuiki, 2021-05-25:
I agree that width inference should run before canonicalization, mainly because it doesn't need the input to be canonicalized, and it enables additional canonicalizations to run.

lattner, 2021-05-25:
Right ,canonicalization is going to be pretty useless pre-width inference, it isn't worth it.

fabianschuiki, 2021-06-11:
Running `firtool --infer-widths` on the FIR input now produces the same output as SFC. Once we make width inference on by default (#1242), you shouldn't even need to pass in the flag anymore.

closed 2021-06-11 by fabianschuiki