llvm/circt #1137: [FIRRTL] infer width (not, tail, dshl ?)

drom, 2021-05-25:
The following FIRRTL program

```python
circuit top_mod :
  module top_mod :
    input inp_j: UInt<56>
    output tmp43: UInt
    output tmp47: SInt
    tmp43 <= not(inp_j)
    tmp47 <= dshl(SInt("o10266562"), tail(tmp43, 49))
```

Compiled with `firtool a_top_mod.fir --lower-to-hw --expand-whens --infer-widths --lowering-options=noAlwaysFF --verilog -o=a_top_mod_new.v` produces this Verilog:

```verilog
module top_mod(
  input  [55:0]  inp_j,
  output [55:0]  tmp43,
  output [149:0] tmp47);

  wire [55:0] _tmp43_output;	// a_top_mod.fir:2:3

  assign _tmp43_output = ~inp_j;	// a_top_mod.fir:6:{11,14}
  wire [55:0] _T = _tmp43_output;	// a_top_mod.fir:7:38
  assign tmp43 = _T;	// a_top_mod.fir:2:3
  assign tmp47 = 150'h216D72 << {143'h0, _T[6:0]};	// a_top_mod.fir:2:3, :7:{14,38}
endmodule

```

Compiled with `firrtl-1.5-SNAPSHOT` produces this Verilog:

```verilog
module top_mod(
  input  [55:0] inp_j,
  output [55:0] tmp43,
  output [21:0] tmp47
);
  wire [149:0] _GEN_1 = 150'sh216d72 << tmp43[6:0];
  assign tmp43 = ~inp_j;
  assign tmp47 = _GEN_1[21:0];
endmodule
```

fabianschuiki, 2021-05-25:
This is a weird issue. Intuitively the result of `firtool` seems correct to me, and the output of the Scala `firrtl` implementation seems wrong: `dshl` produces a result that is the width of its first operand, plus the maximum value of the second operand. Since `tmp43` is 56 bits wide, the `tail(tmp43)` is 7 bits wide, which has a maximum value of 127. The `SInt("o10266562")` literal is 23 bits wide, so the `dshl` overall should be 23 + 127 = 150 bits wide, as `firtool` infers.

@seldridge any ideas why FIRRTL infers such a weird width?

seldridge, 2021-05-25:
Something funky is going on it. This may be an SFC issue.

The following, similar formulation produces the 150-bit width:

```scala
circuit top_mod :
  module top_mod :
    input inp_j: UInt<56>
    output tmp43: UInt
    output tmp47: SInt
    tmp43 <= not(inp_j)
    wire x: UInt
    x <= tail(tmp43, 49)
    tmp47 <= dshl(SInt("o10266562"), x)
```

After the SFC `InferWidths` this is:
```scala
circuit top_mod :
  module top_mod :
    input inp_j : UInt<56>
    output tmp43 : UInt<56>
    output tmp47 : SInt<150>

    tmp43 <= not(inp_j)
    wire x : UInt<7>
    x <= tail(tmp43, 49)
    tmp47 <= dshl(SInt<23>("h216d72"), x)
```

azidar, 2021-05-25:
I think this is a bug where dshl is being converted incorrectly to a dshlw?

seldridge, 2021-05-25:
I think it's actually Scala `InferWidths` here as if you use a node instead of a wire for my expanded example, you get a width of 22:

```scala
circuit top_mod :
  module top_mod :
    input inp_j: UInt<56>
    output tmp43: UInt
    output tmp47: SInt
    tmp43 <= not(inp_j)
    node x = tail(tmp43, 49)
    tmp47 <= dshl(SInt("o10266562"), x)
```

After `InferWidths` this produces:

```scala
circuit top_mod :
  module top_mod :
    input inp_j : UInt<56>
    output tmp43 : UInt<56>
    output tmp47 : SInt<22>

    tmp43 <= not(inp_j)
    node x = tail(tmp43, 49)
    tmp47 <= dshl(SInt<23>("h216d72"), x)
```

fabianschuiki, 2021-05-27:
Is already someone working on this on the SFC side, or should I go ahead and take a look?

azidar, 2021-05-27:
I'm not, feel free!

fabianschuiki, 2021-06-11:
Closing this as it's not an MFC bug.

closed 2021-06-11 by fabianschuiki

drom, 2021-06-11:
should we move it to SFC?

fabianschuiki, 2021-06-11:
Yeah good point. I'll open a PR on SFC.