llvm/circt #1142: [FIRRTL] width inference (mux) mismatch

drom, 2021-05-25:
The following FIRRTL program

```python
circuit top_mod :
  module top_mod :
    input _tmp15: UInt<22>
    output tmp19: UInt
    tmp19 <= mux(UInt<1>(1), UInt(192), _tmp15)
```

Compiled with `firtool` produces this Verilog:

```verilog
module top_mod(
  input  [21:0] _tmp15,
  output [7:0]  tmp19);

  assign tmp19 = 8'hC0;	// a_top_mod.fir:2:3, :5:30
endmodule
```

Compiled with `firrtl-1.5-SNAPSHOT` produces this Verilog:

```verilog
module top_mod(
  input  [21:0] _tmp15,
  output [21:0] tmp19
);
  assign tmp19 = 22'hc0;
endmodule
```

tmp19 width is different

`Yosys 0.9+3911 (git sha1 4b31223e, clang 11.0.1 -fPIC -Os)` reports formal mismatch:

```
ERROR: Can't match gold port `tmp19_gold' to a gate port.
```

fabianschuiki, 2021-05-25:
This is related to canonicalization running before width inference: The canonicalizer folds the `mux` to its first operand due to the constant select signal, but that operand has no inferred width yet. There is an implicit `pad()` operation around a constant-folded mux that gets lost here. I don't think there's a proper way to solve this besides moving width inference up to run before the other passes.

lattner, 2021-05-25:
We should eliminate that canonicalization if it isn't correct -- make it only kick in when all widths are known.

fabianschuiki, 2021-05-26:
#1153 prevents this kind of canonicalization.

PR #1153 (2021-05-26): [FIRRTL] Don't fold mux with unknown result width
Fix an issue around mux canonicalization where folding a mux to one of its operands can result in invalidation of subsequent IR if the mux result does not have a known width. For example, picking an unsized constant can lead to the constant resolving to a width more narrow than assumed in the remaining IR. This change refuses to fold and canonicalize any multiplexers that have unkown result width.

Fixes #1142.
commit cf57f47e (2021-05-26): [FIRRTL] Don't fold mux with unknown result width (#1153)

* Fix an issue around mux canonicalization where folding a mux to one
  of its operands can result in invalidation of subsequent IR if the
  mux result does not have a known width. For example, picking an
  unsized constant can lead to the constant resolving to a width more
  narrow than assumed in the remaining IR. This change refuses to fold
  and canonicalize any multiplexers that have unkown result width.
  Fixes #1142.
closed 2021-05-26 by fabianschuiki