llvm/circt #1145: [FIRRTL] Width Inference fails with instances

darthscsi, 2021-05-25:
eg:
```
/home/andrew/src/circt/test/firtool/blackbox.mlir:30:5: error: 'firrtl.instance' op not supported in width inference
```

fabianschuiki, 2021-05-25:
Yeah the current implementation does not touch instances because it expects the free variables in the widths returned by the `firrtl.instance` op to line up with the free widths defined on the instantiated module's port. This then allows connects to the instance signals to impose constraints on the module's ports. This is still on the todo list, but shouldn't be too difficult to implement.

darthscsi, 2021-05-25:
It's not that it doesn't touch instances, it is that it errors out on instances which blocks enabling the pass by default.

fabianschuiki, 2021-05-25:
Yeah. I can make it less panicky and have it at least work with instances that have fully inferred types. There's an additional blocker: width inference will just assert when it encounters a composite type like a bundle or vector. To enable width inference by default, we should probably make it just not touch these cases to not disturb the rest of the pipeline. That requires a bunch of tweaks.

fabianschuiki, 2021-06-11:
#1226 should make this problem go away.

closed 2021-06-11 by fabianschuiki