llvm/circt #1151: [Py] Integration tests broken -- cannot find `mlir.ir`

teqdruid, 2021-05-25:
https://github.com/llvm/circt/runs/2665697495?check_suite_focus=true

For some reason, the integration tests can no longer find the mlir python libs. This may have something to do with the latest submodule bump.

mikeurbach, 2021-05-26:
Strange, they've been passing for me locally. I can dig into it tomorrow.

teqdruid, 2021-05-26:
Yeah, same here which makes it all the more weird.

mikeurbach, 2021-05-26:
Ok, I guess I will go down this rabbit hole now

teqdruid, 2021-05-26:
Sorry, I haven't had time to dig into this. Thanks. Hopefully it's not too
deep a rabbit hole.

mikeurbach, 2021-05-26:
This seems to be the smoking gun: https://github.com/llvm/circt/runs/2665036812?check_suite_focus=true#step:6:339

Related to this works to standardize CMake flag naming: https://github.com/llvm/llvm-project/commit/9c21ddb70ab56eb3ca5b0f99faa18bb3af17b3df#diff-e5c7daf3222025d57c93afa2e5efb1e6735b915350046159be7cb09d27ec94b8R121

I'll update that flag in our CI. Not sure why I didn't see this locally... probably because I already had a working copy of the Python bindings installed from previous runs with the old flag.

teqdruid, 2021-05-26:
> Related to this works to standardize CMake flag naming: llvm/llvm-project@
> 9c21ddb
> #diff-e5c7daf3222025d57c93afa2e5efb1e6735b915350046159be7cb09d27ec94b8R121

That makes sense. Thanks for digging into this!

~John

mikeurbach, 2021-05-26:
No problem. Manual integration test running now with a new LLVM cache key and updated flag: https://github.com/llvm/circt/actions/runs/879474782

closed 2021-05-26 by mikeurbach