llvm/circt #1159: [FIRRTL] remove unused modules

drom, 2021-05-26:
The following FIRRTL program

```python
circuit top_mod :
  module mod0 :
    input inp: UInt<11>
    output out: UInt<11>
    out <= inp
  module top_mod :
    input inp: UInt<11>
    output out: UInt<11>
    out <= inp
```

Compiled with `firtool a_top_mod.fir --lower-to-hw --expand-whens --infer-widths --lowering-options=noAlwaysFF --verilog -o=a_top_mod_new.v` produces this Verilog:

```verilog
module mod0(
  input  [10:0] inp,
  output [10:0] out);

  assign out = inp;	// a_top_mod.fir:2:3
endmodule

module top_mod(
  input  [10:0] inp,
  output [10:0] out);

  assign out = inp;	// a_top_mod.fir:6:3
endmodule
```

Compiled with `firrtl-1.5-SNAPSHOT` produces this Verilog:

```verilog
module top_mod(
  input  [10:0] inp,
  output [10:0] out
);
  assign out = inp;
endmodule
```

`mod0` should be removed from the output because it is not instantiated inside `top_mod`

youngar, 2021-05-27:
the inlining pass actually removes unused modules (even if there is no inling :) )

seldridge, 2023-07-30:
Fixed per the above comment and on top-of-tree.

closed 2023-07-30 by seldridge