llvm/circt #1172: Check initialization of all memory ports

prithayan, 2021-05-28:
SFC errors out if all memory ports are not initialized. But MFC does not complain.
SFC error:
```
Exception in thread "main" firrtl.passes.CheckInitialization$RefNotInitializedException:  : [module MemSimple]  Reference _M is not fully initialized.
   : _M.rw.wdata <= VOID
``` 

```python
circuit MemSimple:
 module MemSimple :
    input clock1  : Clock
    input clock2  : Clock
    input inpred  : UInt<1>
    input indata  : SInt<42>
    output result : SInt<42>
    output result2 : SInt<42>

    mem _M :
          data-type => SInt<42>
          depth => 12
          read-latency => 0
          write-latency => 1
          reader => read
          writer => write
          readwriter => rw
          read-under-write => undefined

    result <= _M.read.data
    result2 <= _M.rw.rdata

    _M.read.addr <= UInt<1>("h0")
    _M.read.en <= UInt<1>("h1")
    _M.read.clk <= clock1
    _M.rw.addr <= UInt<1>("h0")
    _M.rw.en <= UInt<1>("h1")
    _M.rw.clk <= clock1
    _M.rw.wmask <= UInt<1>("h1")
    _M.rw.wmode <= UInt<1>("h1")
    _M.write.addr <= validif(inpred, UInt<3>("h0"))
    _M.write.en <= mux(inpred, UInt<1>("h1"), UInt<1>("h0"))
    _M.write.clk <= clock2
    _M.write.data <= validif(inpred, indata)
    _M.write.mask <= validif(inpred, UInt<1>("h1"))
```

Output from MFC,
Uninitialized port is set to X `(wo_data_0  (42'bx)`
```verilog
module MemSimple(
  input         clock1, clock2, inpred,
  input  [41:0] indata,
  output [41:0] result, result2);

  wire [3:0]  _T; // 1.fir:10:5
  wire [3:0]  _T_0; // 1.fir:10:5
  wire [3:0]  _T_1; // 1.fir:10:5
  wire [41:0] _M_ro_data_0; // 1.fir:10:5
  wire [41:0] _M_ro_data_1; // 1.fir:10:5
  wire [3:0]  _M_read_addr; // 1.fir:10:5
  wire [3:0]  _M_rw_addr; // 1.fir:10:5
  wire [3:0]  _M_write_addr;  // 1.fir:10:5

  FIRRTLMem_2_2_0_42_12_0_1_0 _M (  // 1.fir:10:5
    .ro_clock_0 (clock1),
    .ro_en_0    (1'h1), // 1.fir:24:19
    .ro_addr_0  (_T_1), // 1.fir:10:5
    .ro_clock_1 (clock1),
    .ro_en_1    (1'h0), // 1.fir:23:21
    .ro_addr_1  (_T_0), // 1.fir:10:5
    .wo_clock_0 (clock1),
    .wo_en_0    (1'h1), // 1.fir:24:19
    .wo_addr_0  (_T_0), // 1.fir:10:5
    .wo_mask_0  (1'h1), // 1.fir:24:19
    .wo_data_0  (42'bx),  // 1.fir:10:5
    .wo_clock_1 (clock2),
    .wo_en_1    (inpred),
    .wo_addr_1  (_T), // 1.fir:10:5
    .wo_mask_1  (1'h1), // 1.fir:24:19
    .wo_data_1  (indata),
    .ro_data_0  (_M_ro_data_0),
    .ro_data_1  (_M_ro_data_1)
  );
  assign _T_1 = _M_read_addr; // 1.fir:10:5
  assign _T_0 = _M_rw_addr; // 1.fir:10:5
  assign _T = _M_write_addr;  // 1.fir:10:5
  assign _M_read_addr = 4'h0; // 1.fir:23:18
  assign _M_rw_addr = 4'h0; // 1.fir:23:18, :26:16
  assign _M_write_addr = 4'h0;  // 1.fir:23:18, :31:19
  assign result = _M_ro_data_0; // 1.fir:2:2
  assign result2 = _M_ro_data_1;  // 1.fir:2:2
endmodule
```

youngar, 2021-05-28:
Thanks for the report!  Initialization checking is enabled by the `--expand-whens` firtool option.  This test case actually revealed a crash when printing in the message: it assumed that subfield op argument was a bundle type when it was a flip type 🤦 .  As soon as I get the patch in, this is the error message you can expect:
```
./test-bbmem.fir:10:5: error: sink "_M.rw_w.data" not fully initialized
    mem _M :
    ^
./test-bbmem.fir:10:5: note: see current operation: %_M_read, %_M_rw_r, %_M_rw_w, %_M_write = "firrtl.mem"() {annotations = [], depth = 12 : i64, name = "_M", portNames = ["read", "rw_r", "rw_w", "write"], readLatency = 0 : i32, ruw = 0 : i32, writeLatency = 1 : i32} : () -> (!firrtl.flip<bundle<addr: uint<4>, en: uint<1>, clk: clock, data: flip<sint<42>>>>, !firrtl.flip<bundle<addr: uint<4>, en: uint<1>, clk: clock, data: flip<sint<42>>>>, !firrtl.flip<bundle<addr: uint<4>, en: uint<1>, clk: clock, data: sint<42>, mask: uint<1>>>, !firrtl.flip<bundle<addr: uint<4>, en: uint<1>, clk: clock, data: sint<42>, mask: uint<1>>>)
```

youngar, 2021-05-28:
merged the fix: https://github.com/llvm/circt/pull/1173
I'm going to open a PR for turning on the pass by default: https://github.com/llvm/circt/pull/1174

closed 2021-05-28 by youngar