llvm/circt #1177: [SV] Make `VerbatimExpr` accept arbitrary result types

fabianschuiki, 2021-05-28:
Is there a technical reason why the `VerbatimExpr` of the SV dialect has a result type restricted to `AnySignlessInteger`? Would it make sense to generalize this to accept `AnyType`?

As a concrete example, I would like to have a piece of verbatim SV act as an expression that returns a FIRRTL type. It feels like the `VerbatimExpr` op would be the perfect thing to use:
```
%0 = sv.verbatim.expr "foo.bar" : () -> !firrtl.uint<42>  // could be a bundle, vector, any other type
%1 = firrtl.add %0, %otherStuff : (!firrtl.uint<42>, !firrtl.uint<42>) -> !firrtl.uint<43>
```

lattner, 2021-05-28:
It'd make sense for sv.verbatim.expr to be generalized, e.g. to support hw.struct etc.  

That said, I don't think it make sense to support firrtl.  These are different worlds and we don't currently mix them - we have no other cases were we have SV and FIRRTL things in the same module.

We can add a firrtl.verbatim though!

fabianschuiki, 2021-05-31:
That's an excellent point. Sounds like `firrtl.verbatim` is a much better approach here, then lowering to `sv.verbatim` as you would expect. Will take a look.