llvm/circt #1179: [Py] Move `design_entry` and anything opinionated out of "bindings"

teqdruid, 2021-05-29:
We need to figure out a line between syntactic sugar on the bindings and "opinionated language"-y things.

Not sure where this should move yet, but it should be in the CIRCT repo just for practicality reasons -- about 1/2 the changes we make to the language-y stuff require changes to either the python bindings proper or the CIRCT C++ code.

teqdruid, 2021-06-01:
I'm thinking a new top-level directory called "frontends" is appropriate for these sorts of things. I'm also thinking that the design entry python stuff we've been working on should be called "pyCDE" for Python CIRCT Design Entry. So it'd be under "frontends/pyCDE".

mikeurbach, 2021-06-01:
> I'm thinking a new top-level directory called "frontends" 

This is what mlir-npcomp does to house their PyTorch frontend separately from the rest of the general "numerical python" compiler infrastructure. In that case, the PyTorch-specific Python and C++ code lives in `frontends/`, but there is a `Torch` dialect used by the frontend, which lives in the usual `include/` and `lib/` directories. Not making any value judgement, but there is another data point from the MLIR community about how to address the split.

To argue with myself for a bit, say we do create a `frontends/` space in CIRCT. Would Edith be considered as a "frontend"? If it doesn't make sense to ever have Edith in the CIRCT tree, then perhaps it doesn't make sense to have pyCDE in the CIRCT tree.

Anyway, some fuel for the fire. Looking forward to discussing it more in the ODM.

teqdruid, 2021-06-01:
> To argue with myself for a bit, say we do create a `frontends/` space in CIRCT. Would Edith be considered as a "frontend"?

I would have no objection to Edith and MLIRSwift living under `frontends/`.

> CIRCT tree

I think it's important to make a distinction between "tree" (poorly defined) and "repo". Nothing under `frontends/` need be part of the CIRCT build and test system. If a frontend is being co-developed with CIRCT (as `pyCDE` is), sharing a repo is ideal purely for practical reasons -- not having to keep yet another repo in sync 1:1. Ideally, bugs and test failures in the frontends wouldn't cause CIRCT tests to fail (just like if they were two different repos) so CIRCT developers who aren't working on the frontends wouldn't be responsible for fixing other people's frontends when their patch breaks said frontend (just like if they were two different repos).

mikeurbach, 2021-06-01:
> Nothing under frontends/ need be part of the CIRCT build and test system.

Fair point, and this is exactly how it works in mlir-npcomp. There are separate build steps and checks for the PyTorch frontend.

teqdruid, 2021-06-02:
We spoke about this in today's meeting and agreed that this is an acceptable solution.

PR #1195 (2021-06-03): [PyCDE] Moving the most opinionated python out of bindings (#1179)
- Creating new frontends top level directory which contains frontends which are   optionally included.
- Moving @module and @generator into PyCDE frontend.
- Keeping `connect` in the bindings since it's used in the bindings tests.
- Fixing the integration tests.

Closes #1179.
commit 51639cad (2021-06-03): [PyCDE] Moving the most opinionated python out of bindings (#1179) (#1195)

- Creating new frontends top level directory which contains frontends which are optionally included.
- Moving @module and @Generator into PyCDE frontend.
- Keeping connect in the bindings since it's used in the bindings tests.
- Fixing the integration tests.

Closes #1179.
closed 2021-06-03 by teqdruid