llvm/circt #1186: [FIRRTL] (validif) mismatch

drom, 2021-06-02:
Not a bug, but missing constant folder for `andr, orr, xorr` operations.

The following FIRRTL program

```python
circuit top_mod :
  module top_mod :
    input inp_o: SInt<6>
    output tmp11: SInt<6>
    tmp11 <= validif(UInt<1>(0), inp_o)
```

Compiled with `firtool` produces this Verilog:

```verilog
module top_mod(
  input  [5:0] inp_o,
  output [5:0] tmp11);

  assign tmp11 = 6'h0;	// a_top_mod.fir:2:3, :5:14
endmodule
```

Compiled with `firrtl-1.5-SNAPSHOT` produces this Verilog:

```verilog
module top_mod(
  input  [5:0] inp_o,
  output [5:0] tmp11
);
  assign tmp11 = inp_o;
endmodule
```

`Yosys 0.9+3911 (git sha1 4b31223e, clang 11.0.1 -fPIC -Os)` reports formal mismatch:

```
ERROR: Found 6 unproven $equiv cells in 'equiv_status -assert'
```

lattner, 2021-06-02:
Both are correct, but it looks like SFC is folding this differently.  I'm +1 on following its lead here.  @seldridge, FIRParser.cpp contains this code that is doing this:

```
     // validif(0, x) -> always invalid.
      if (cst.value().isNullValue()) {
        result = builder.create<InvalidValueOp>(opTypes[1]);
        return success();
      }
```

Should I change this to return the second operand for compatiblity?

seldridge, 2021-06-02:
We talked about this in our daily sync and we were leaning towards just removing that and letting normal canonicalization do its thing (which I think does the correct thing here?).

lattner, 2021-06-02:
works for me!  I don't think the "optimization" matters, so I'd prefer to be as close to SFC as possible.  This will make fuzzing etc be easier.

PR #1191 (2021-06-02): [FIRParser] Don't optimize validif in the parser
This pattern is already handled by the FIRRTL folders.  As a side effect
of this change, we are now aligning with how the Scala FIRRTL compiler
handles validif with a constant condition:

`validif(0, invalid, value)`:
 - old: condition always false, return invalid.
 - new: mux with an invalid operand, return the other operand.

This fixes #1186.
commit b87bd044 (2021-06-02): [FIRParser] Optimize validif the same way as SFC (#1191)

We are now aligning with how the Scala FIRRTL compiler
handles validif with a constant condition:

`validif(0, value, invalid)`:
 - old: condition always false, return invalid.
 - new: mux with an invalid operand, return the other operand.

This fixes #1186.
closed 2021-06-02 by youngar