llvm/circt #1188: [FIRRTL] Integration test times out

teqdruid, 2021-06-02:
[image]

https://github.com/llvm/circt/runs/2727815005?check_suite_focus=true

seldridge, 2021-06-02:
How does one go about debugging this? Is there a way to get this to build with waveform dumping support in the Verilated binary?

lattner, 2021-06-02:
It looks like imconstprop is deleting a reg assignment in "LFSR" module.  I'll take a look

lattner, 2021-06-02:
The key thing happening is that we're deleting the logic that reads and writes the D0123456 reg here:

```
  module LFSR :
    input clock : Clock
    input reset : UInt<1>
    output io_out : UInt<6>
    output io_out3 : UInt<3>

    reg D0123456 : UInt<6>, clock with :
      reset => (UInt<1>("h0"), D0123456) @[main.scala 16:25]
    node _T = bits(D0123456, 4, 0) @[main.scala 20:30]
    node _T_1 = bits(D0123456, 5, 5) @[main.scala 20:47]
    node _T_2 = cat(_T, _T_1) @[main.scala 20:36]
    node _T_3 = bits(D0123456, 4, 4) @[main.scala 20:59]
    node nxt_D0123456 = xor(_T_2, _T_3) @[main.scala 20:50]
    node _T_4 = bits(D0123456, 1, 1) @[main.scala 25:22]
    node _T_5 = bits(D0123456, 3, 3) @[main.scala 25:37]
    node _T_6 = cat(_T_4, _T_5) @[main.scala 25:26]
    node _T_7 = bits(D0123456, 5, 5) @[main.scala 25:52]
    node _T_8 = cat(_T_6, _T_7) @[main.scala 25:41]
    io_out <= D0123456 @[main.scala 24:10]
    io_out3 <= _T_8 @[main.scala 25:11]
    D0123456 <= mux(reset, UInt<6>("h1"), nxt_D0123456) @[main.scala 16:25 main.scala 16:25 main.scala 22:12]
```

This looks correct to me: the initializer specified is the "self init" pattern, which is aka "uninitialized"/invalid.  The logic in that module reads the invalid value and does other things that permute the invalid and store it back.  Therefore we can nuke all of that logic.

I suspect that the integration test worked before out of luck and incorrect x behavior in verilator.  I'll strengthen the init condition to be correct.

lattner, 2021-06-02:
Fixed in d7cca952.

lattner, 2021-06-02:
incidentally, I don't have verilator so I can't run this test.  I'd love some validation of whether this fixes the problem or not.  There are a bunch of other regs (the golden_'s) in the test with the same issue.

mikeurbach, 2021-06-02:
@lattner I have Verilator installed and when I run the integration tests after d7cca952 I do see `EmitVerilog/exampleCodeGen.fir` continue to time out.

lattner, 2021-06-02:
Thanks Mike.  Do you have a sense of what the issue is?  I can add initializers to the other reg's if that would help.  @darthscsi , you added this case - where did it come from?  Should it just be removed?

darthscsi, 2021-06-02:
This is from a chisel program testing firrtl->simulation.  The reset behavior is what chisel does,  note that besides the disabled resets (@seldridge told me it was from the lack of an internal non-reset register from in the scala firrtl compiler), there are muxes on reset which do the "real" reset.

As a test, we should keep this as it is testing the direct output of SFC through the pipeline.

seldridge, 2021-06-02:
I haven't investigated this, but I think the issue is that the top module is not using anything in the LFSR. It's just two inputs and a bunch of nodes. IMConstProp may be doing "the right thing" by aggressively removing things.

darthscsi, 2021-06-02:
> This looks correct to me: the initializer specified is the "self init" pattern, which is aka "uninitialized"/invalid. The logic in that module reads the invalid value and does other things that permute the invalid and store it back. Therefore we can nuke all of that logic.

Reset is not initialization.

But, the register has 0 tied to reset, so it isn't reset.  But, it always has valid data written to it from the mux at the end, which implements the real reset.
```D0123456 <= mux(reset, UInt<6>("h1"), nxt_D0123456) ```
Ensures that after reset, the register is always valid.  Thus I don't see how you can remove all reads from it.  It is only invalid between initialization and reset.

darthscsi, 2021-06-02:
Consider this chisel:
```
import chisel3._
import chisel3.stage.ChiselStage

class Foo extends MultiIOModule {
  val foo = IO(Output(UInt(4.W)))
  val bar = IO(Input(UInt(4.W)))
    
  val x = RegInit(5.U(4.W))
  foo := x
  x:= bar
}

println(ChiselStage.emitFirrtl(new Foo))
```
Produces high firrtl:
```
circuit Foo :
  module Foo :
    input clock : Clock
    input reset : UInt<1>
    output foo : UInt<4>
    input bar : UInt<4>

    reg x : UInt<4>, clock with :
      reset => (reset, UInt<4>("h5")) @[main.scala 10:18]
    foo <= x @[main.scala 11:7]
    x <= bar @[main.scala 12:4]
```
from which firrtl -X low produces:
```
circuit Foo :
  module Foo :
    input clock : Clock
    input reset : UInt<1>
    output foo : UInt<4>
    input bar : UInt<4>

    reg x : UInt<4>, clock with :
      reset => (UInt<1>("h0"), x) @[main.scala 10:18]
    foo <= x @[main.scala 11:7]
    x <= mux(reset, UInt<4>("h5"), bar) @[main.scala 10:18 main.scala 10:18 main.scala 12:4]
    ```

seldridge, 2021-06-02:
@darthscsi is correct here. This is the low FIRRTL pattern for an uninitialized reset.

> This looks correct to me: the initializer specified is the "self init" pattern, which is aka "uninitialized"/invalid. The logic in that module reads the invalid value and does other things that permute the invalid and store it back. Therefore we can nuke all of that logic.

By this reasoning any register with an invalid value can be removed. However, "invalid" doesn't really exist! (For that matter `x` doesn't exist either.) That register has _some_ value when you turn it on. You can write logic that uses this pattern. (Chisel's round robin arbiter uses this pattern for the "last grant" register---and this has [been a point of confusion in the past](https://github.com/chipsalliance/chisel3/pull/267)). Just because it's uninitialized (or assigned to invalid or `x`) doesn't mean it's a candidate for deletion.

lattner, 2021-06-02:
> That register has some value when you turn it on. 

Yes, this is what `firrtl.invalid` and `sv.x` model: an unknown value.

I think it is possible that I'm very confused here (not unusual!) and that we may have a major problem.  Is this logic in FIRParser ~line 2823 correct?

```
   // The Scala implementation of FIRRTL represents registers without resets
    // as a self referential register... and the pretty printer doesn't print
    // the right form. Recognize that this is happening and treat it as a
    // register without a reset for compatibility.
 ```

It is turning this:

```
    reg x : UInt<4>, clock with :
      reset => (UInt<1>("h0"), x) @[main.scala 10:18]
```
into firrtl.reg instead of firrtl.regreset.

-Chris

lattner, 2021-06-02:
My understanding is that the "0" value on the reset line means that the init value will never be used.

lattner, 2021-06-02:
Ok, I think that is correct as far as I can tell.  This is an imconstprop bug handling the mux.  Will investigate.

seldridge, 2021-06-02:
Your understanding of constructing `reg` out of `regreset` with a self-referential zero-conditioned init is correct. That's fine.

The SFC only has one register IR operation and all fields are mandatory. So, it has to jump through some hoops to represent a reset-less register.

lattner, 2021-06-03:
Ok, I think that 400678d0 should fix it.  Thank you for filing this!

closed 2021-06-03 by lattner