llvm/circt #1204: Unable to build CIRCT

JuanEsco063, 2021-06-04:
Hello,

Once I reach step 4, Build and test CIRCT, of the installation process and run the command "ninja" at one point I get the following error:

[191/221] Building CXX object lib/Dialect/SV/Transforms/CMakeFiles/obj.CIRCTSVTransforms.dir/SVExtractTestCode.cpp.o
FAILED: lib/Dialect/SV/Transforms/CMakeFiles/obj.CIRCTSVTransforms.dir/SVExtractTestCode.cpp.o 
/usr/bin/c++ -DGTEST_HAS_RTTI=0 -D_DEBUG -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -I../llvm/llvm/include -I../llvm/build/include -I../llvm/mlir/include -I../llvm/build/tools/mlir/include -I../include -Iinclude -fno-exceptions -fno-rtti -fPIC -fvisibility-inlines-hidden -Werror=date-time -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wno-missing-field-initializers -Wimplicit-fallthrough -Wno-noexcept-type -Wdelete-non-virtual-dtor -Wno-comment -Wmisleading-indentation -fdiagnostics-color -g  -fno-exceptions -fno-rtti -std=gnu++14 -MD -MT lib/Dialect/SV/Transforms/CMakeFiles/obj.CIRCTSVTransforms.dir/SVExtractTestCode.cpp.o -MF lib/Dialect/SV/Transforms/CMakeFiles/obj.CIRCTSVTransforms.dir/SVExtractTestCode.cpp.o.d -o lib/Dialect/SV/Transforms/CMakeFiles/obj.CIRCTSVTransforms.dir/SVExtractTestCode.cpp.o -c ../lib/Dialect/SV/Transforms/SVExtractTestCode.cpp
../lib/Dialect/SV/Transforms/SVExtractTestCode.cpp: In function ‘void getBackwardSliceSimple(mlir::Operation*, llvm::SmallPtrSetImpl<mlir::Operation*>&, std::function<bool(mlir::Operation*)>)’:
../lib/Dialect/SV/Transforms/SVExtractTestCode.cpp:45:20: error: parse error in template argument list
     if (!op || op->hasTrait<OpTrait::IsIsolatedFromAbove>())
                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../lib/Dialect/SV/Transforms/SVExtractTestCode.cpp:45:59: error: no matching function for call to ‘mlir::Operation::hasTrait<<expression error> >()’
     if (!op || op->hasTrait<OpTrait::IsIsolatedFromAbove>())
                                                           ^
In file included from ../llvm/mlir/include/mlir/IR/OpDefinition.h:23:0,
                 from ../llvm/mlir/include/mlir/IR/FunctionSupport.h:18,
                 from ../llvm/mlir/include/mlir/IR/BuiltinOps.h:16,
                 from ../include/circt/Dialect/HW/HWOps.h:18,
                 from ../lib/Dialect/SV/Transforms/SVPassDetail.h:15,
                 from ../lib/Dialect/SV/Transforms/SVExtractTestCode.cpp:16:
../llvm/mlir/include/mlir/IR/Operation.h:476:53: note: candidate: template<template<class T> class Trait> bool mlir::Operation::hasTrait()
   template <template <typename T> class Trait> bool hasTrait() {
                                                     ^~~~~~~~
../llvm/mlir/include/mlir/IR/Operation.h:476:53: note:   template argument deduction/substitution failed:
../lib/Dialect/SV/Transforms/SVExtractTestCode.cpp:45:59: error: template argument 1 is invalid
     if (!op || op->hasTrait<OpTrait::IsIsolatedFromAbove>())
                                                           ^

I am currently using the latest version of the main epository, including the LLVM submodule.

Thanks in advance.

JuanEsco063, 2021-06-04:
In case it is important, these are the version of the compilers I am using

gcc/g++ version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04) 
clang/llvm version 6.0.0-1ubuntu2 (tags/RELEASE_600/final)

teqdruid, 2021-06-04:
> template<template class Trait> bool mlir::Operation::hasTrait()
> template <template class Trait> bool hasTrait() {

Are you sure you've "git submodule update"d and rebuilt the LLVM submodule?

JuanEsco063, 2021-06-04:
Hey John. Yeah, I even nuked my entire local copy of the repo and followed the installation guide from the very beginning. LLVM compiles correctly but building CIRCT fails

JuanEsco063, 2021-06-04:
And now I feel like I am going crazy. I Just followed the full installation guide again but on a server running native Linux (I have been using a VM) and plenty of RAM as a sanity check but I am getting a different error on top of the previous one. See the attached file for the full error message.

[New error.txt](https://github.com/llvm/circt/files/6601201/New.error.txt)

youngar, 2021-06-04:
The first error in `New error.txt` is showing a that there is a merge conflict in a file.  
```
../lib/Conversion/StandardToHandshake/StandardToHandshake.cpp:1530:1: error: version control conflict marker in file
 <<<<<<< HEAD
 ```
This could have been from a bad `git pull` or something.

JuanEsco063, 2021-06-04:
Hm. I did not do a clean install on the server, just a git pull so that could be the case. Let me try again all the steps from the very beginning on a clean copy of the repo. Thank you @youngar

JuanEsco063, 2021-06-04:
Doing a clean install on a server running native Linux with llvm/clang 11 and gcc 10.2 fixed the problem. Still curious if the issue is related to running on a VM ( with just 8GB of RAM which made the LLVM building process crash a few times) or the compiler versions I was using but I am closing the issue.

closed 2021-06-04 by JuanEsco063

youngar, 2021-06-05:
I think there are some issues with older compilers and we are currently testing on a pretty limited set of compilers.  It seems likely that it could have been the problem.